Ysurac/openmptcprouter #3705: I get a lot of file .openmptcprouter.uci-

Silvanskii, 2024-12-10:
I get a lot of files, over 1000 in 1 day. The device runs out of memory and stops working. To be more precise, the code says memory 0, firewall does not work and all interfaces stop working.
[image]

Ysurac, 2024-12-10:
Please use template

Silvanskii, 2024-12-10:
> Please use template

Server template, version template, question template. Clarify

Ysurac, 2024-12-10:
When you opening a bug issue, there is a template. It's required, as all others bug issues.

Silvanskii, 2024-12-10:
I added a script that removed this garbage in the startup, but I think this is a temporary solution. Thousands of files should not be created.

```
# -*- coding: utf-8 -*-
import os
import logging
logging.basicConfig(filename='/usr/bin/EOled/log/clearlog', level=logging.INFO, format='%(asctime)s - %(message)s')
config_dir = '/etc/config'
deleted_files = 0
for filename in os.listdir(config_dir):
    if filename.startswith('.'):
        try:
            os.remove(os.path.join(config_dir, filename))
            deleted_files += 1
        except OSError as e:
            logging.error(f'Error {filename}: {e}')
logging.info(f'Delete {deleted_files}')
```

Silvanskii, 2024-12-10:
I also checked my old device nanopi r4s with version v0.59.1-5.4 r0 and the story is the same, there are 315 files in the /etc/config folder. They were hidden, I started to see them when I set the settings to show hidden files. These files appear in squashfs system after a quick power off. If you just reboot the system no new files are added. Here is a screenshot and it shows that these files were created half a year ago, but here the size of the flashcard is very large, so there is no problem with lack of space.
[image]

Ysurac, 2024-12-11:
Strange that there is so many files. I added a script to remove old files if there is still some.
I will also check if some commit are not missing...

closed 2025-03-16 by github-actions[bot]

xNagatsu, 2025-08-08:
I have the same problem with 0.61. Furthermore, this corrupted my /etc/config/network file because the /overlay/ partition was full.

Ysurac, 2025-08-08:
You tried latest snapshot ?

Silvanskii, 2025-08-13:
> You tried latest snapshot ?

I've made many attempts to create an image from a snapshot, but I keep getting compilation errors on kernel 6.6.
Maybe I'm doing something wrong.
git checkout develop
OMR_TARGET="EOBonding-orange-6.6" OMR_FEED_SRC="develop" OMR_KERNEL="6.6" ./build.sh
--  ERROR: package/feeds/openmptcprouter/rust [host] failed to build.

git checkout master
OMR_TARGET="EOBonding-orange-master" OMR_FEED_SRC="master" OMR_KERNEL="6.6" ./build.sh

CC [M]  /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock/xt_pknock.o
In file included from ./arch/arm/include/asm/div64.h:107,
                 from ./include/linux/math.h:6,
                 from ./include/linux/math64.h:6,
                 from ./include/linux/time.h:6,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock/xt_pknock.c:10:
/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock/xt_pknock.c: In function 'has_secret':
./include/asm-generic/div64.h:222:35: error: comparison of distinct pointer types lacks a cast [-Werror]
  222 |         (void)(((typeof((n)) *)0) == ((uint64_t *)0));  \
      |                                   ^~
/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock/xt_pknock.c:743:17: note: in expansion of macro 'do_div'
  743 |                 do_div(t, 60);
      |                 ^~~~~~
cc1: all warnings being treated as errors
make[10]: *** [scripts/Makefile.build:243: /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock/xt_pknock.o] Error 1
make[9]: *** [scripts/Makefile.build:480: /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions/pknock] Error 2
make[8]: *** [/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/linux-6.6.86/Makefile:1921: /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions] Error 2
make[7]: *** [Makefile:234: __sub-make] Error 2
make[7]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/linux-6.6.86'
make[6]: *** [Makefile:468: modules] Error 2
make[6]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/extensions'
make[5]: *** [Makefile:622: all-recursive] Error 1
make[5]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28'
make[4]: *** [Makefile:394: all] Error 2
make[4]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28'
make[3]: *** [Makefile:218: /home/debian/openmptcprouter/EOBonding-orange/6.6/source/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-sunxi_cortexa7/xtables-addons-3.28/.built] Error 2
make[3]: Leaving directory '/home/debian/openmptcprouter/feeds/openmptcprouter/xtables-addons'
time: package/feeds/openmptcprouter/xtables-addons/compile#17.28#2.10#20.85
    ERROR: package/feeds/openmptcprouter/xtables-addons failed to build.
make[2]: *** [package/Makefile:179: package/feeds/openmptcprouter/xtables-addons/compile] Error 1
make[2]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source'
make[1]: *** [package/Makefile:173: /home/debian/openmptcprouter/EOBonding-orange/6.6/source/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian/openmptcprouter/EOBonding-orange/6.6/source'
make: *** [/home/debian/openmptcprouter/EOBonding-orange/6.6/source/include/toplevel.mk:233: world] Error 2
debian@debian:~/openmptcprouter/EOBonding-orange/6.6/source$