Ysurac/openmptcprouter #3707: Aggregation and failover doens't seem to work

RodriMora, 2024-12-12:
Hi! Thanks for this great project

## Expected Behavior
Both WAN connections should be used to aggreate the speed.
If one connection goes down, the other wan should still be used and provide internet connection but a lower speed.

## Current Behavior
Omr-test-speed doesn't work for aggregated. 
But when doing individual speed test on each interface both work individually. 
When I manually unplug one of the wan connections (the non-master one) then Omr-test-speed works

## Possible Solution
.

## Steps to Reproduce the Problem
Architecture:
[image]

The router is a bit older Dell laptop with 100Mbit NIC, but 4 cores, 8GB RAM and CPU usage doesn't seem high at any point.

Scenario 1. 

The 5G router is a decent one TP-Link Archer NX200. Reception is good 80/20mbit. 25ms latency
The other router is a 4G one (100mbit nic though) but using it only as a "dumb router" connected to my fiber internet at home 10Gbit fiber.
Wan1 = 5G router
wan2 = 4g router connected to fiber
both wans up
 
`omr-test-speed` **doesn't work**
[image]

`omr-test-speed wan1` works
[image]

`omr-test-speed wan2` works
[image]

Status while doing this test:
[image]

Settings:
[image]
[image]
[image]
[image]

Scenario 2.

Wan1 = 5G router
wan2 = 4g router connected to fiber
Wan 1 disconnected

Unplugging the 5G router (wan1) and leaving only the 4G router connected to fiber:

`omr-test-speed works` (same with `omr-test-speed wan2`)
[image]

Status:
[image]

**Upload test**
I can do upload test but I don't know how to choose which interface to to the upload.
wan1 up+wan2 up = `omr-iperf` works (only at wan1 speeds)
wan1 down+wan2 up = `omr-iperf `doesn't work
wan1 up+wan2 down = `omp-iperf` works

## Context (Environment)
I would like to have both increased download and upload speeds as well as failover in case one connection drops or loses signal. This would be used for livestreaming (so mainly upload) using the SRT protocol using OBS from a remote connection aggregating various internet sources, probably 5G+4G/LTE+Starlink. Currently testing with a mix of ethernet fiber+5G+4G/LTE.

## Specifications

  - OpenMPTCProuter version: 0.61-6.6
  - OpenMPTCProuter VPS version: 0.1031 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Using a local VM with all the ports forwarded (8 core, 8GB RAM, 10Gbit). I have also tested with a OVH VPN (4 core, 8GB RAM, 1Gbit)
  - OpenMPTCProuter platform: x86_64
  - Country: Spain

Ysurac, 2024-12-12:
There is something strange with WAN2: the detected public IP seems to be the same as VPS IP.
Can you give me the result of ```omr-ip-intf wan2``` ?

RodriMora, 2024-12-12:
> There is something strange with WAN2: the detected public IP seems to be the same as VPS IP. Can you give me the result of `omr-ip-intf wan2` ?

That's because the VPN is a VM in my home network, with the ports 6xxxx forwarded. And the wan2 if connected to that same network with a 4G router. Could that be a problem?
I can test with a OVH VPS though if needed. On it right now

RodriMora, 2024-12-12:
Ok! I've done some more test using the OVH VPS instead of a local VM as the VPS. It seems like there is some loop problem in having one of the WAN connections and the VPS with the same external IP. 

I added a USB gigabit dongle in case the integrated 100mbit nic in the router was a problem.

I did a bunch of test and got different results. But after a few hours of testing it seems to be working much better! I also changed Multipath TCP Scheduler to "Red", but im not really sure what it does, should I changed it back for my purpose?

Maybe the integrated NIC was bad either the model for itself or the 100mbit caused a bottleneck. I will do more test with a raspberry pi 5 or another minipc. Closing this for now, thanks!

closed 2024-12-12 by RodriMora