Ysurac/openmptcprouter #3709: Only use 1 WAN connection instead of all 3

zimboman, 2024-12-13:
## Expected Behavior
I have configured the Router with its standard Confiurations. No Spefic changes. 
The Router should use 3 WAN Connections (DSL, STARLINK & LTE)

## Current Behavior
Actually the Connection to the VPS is fine. In the Status TAB the Connection is established an all WAN Connections are Working. 
[image]

When I start a download of a file, via different ways such as downloading directly via the browser, speed tests, or via the JDownloader, then only the “1UND1” line is used, which is the DSL line. 

**Some Infos about my used Enviroment:** 
- OpenMPTCP Router runs on a Microsoft HyperV
- The VPS has Debian12
- The 1UND1 Line have a capacity of 100DOWN/40UP
- The STARLINK Line have a regular capacity of around 300DOWN/20UP 
-> I have the 40GB Priority Service with an public IPv4
- The LTE Line (4G) have arround 100DOWM/20 UP

The IP addresses of the “WAN interfaces” of the OpenMPTC router, which are routed to the various internet lines:

**1UND1**
Interface: eth1 - 10.20.20.254 - Gateway is: 10.20.20.1
Multipath TCP: Master 

**STARLINK** 
Interface eth2- 10.30.30.254 - Gateway is: 10.30.30.1
Multipath TCP: enabled

**LTE** 
Interface eth3 - 10.40.40.254 - Gateway is: 10.40.40.1
Multipath TCP: enabled
Only Interface with activated "SQM autorate"

I have already carried out various tests and I can't really see any faults here. 
I have already made the following adjustments & tests:

**Done a Speedtest on the CLI on die OpenMPTCP Router:**
[image]

**I have checked the established Connections:**
[image]
What I notice here is that it only makes an established connection via the network 10.20.20.254, which is the network for the 1UND1 line. It seems to me that a connection is not established at all via the other lines.

**Ping via the individual interfaces to the Internet works. Each of the router's “WAN interfaces” can reach 8.8.8.8:**
[image]

**The Fullmesh also seems to be in Order:**
[image]

**Changes i have made in Network > MPTCP are:** 
Congestion Control: cubic
Max add address: 3 (Because 1 Master + 2 enabled Lines)

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version:  6.1.0-28-cloud-amd64 0.1031 - Build Date Tue Oct 22 10:46:06 UTC 2024
  - OpenMPTCProuter VPS provider: IONOS (Germany)
  - OpenMPTCProuter platform: Microsoft HyperV x86_64

zimboman, 2024-12-13:
I actually check this in two ways. Once via the web interface you mentioned, yes. But also on my router that hangs in front of it and provides the 3 networks and terminates the dial-up of the Internet connections. There I can also see that the traffic is only increasing on one line:

[image]

In the picture you can see the diagram of 1UND1 at the top left, STARLINK at the bottom left and LTE at the top right. 
It is important to note the information on the left about the landscape format of kBit/s and MBit/s. The one deflection at the top left, the blue one, for example, was a test download where approx. 86 MBit/s was achieved. So I also check outside the OpenMPTC router which line is being utilized here.

Ysurac, 2024-12-13:
Did you try putting master on another interface ?
Did you try with 2 connections ?
Did you try another VPS ?

zimboman, 2024-12-13:
Did you try putting master on another interface ?
_-> If STARLINK is the Master it will still use the 1UND1 Connection only_

Did you try with 2 connections ?
-> After i deleted the LTE Connection and the STARLINK still keeps as master i have now Interruptions in the Download.
After a download has been started, something is downloaded for a few seconds and that's it.
This means more stability problems than before. 
[image]

After the Change of the master from STARLINK to 1UND1 it took a while for the 1UND1 line to be used again. But even after the change, only the respective master line is used again.

Did you try another VPS ?
Not yet, no - But definitly a thing i will try

zimboman, 2024-12-13:
Now i changed the VPS and it use at the beginning both Connections, Nice! 
But the Traffic is slow - The CPUs on the VPS is between 20 and 50% and on the Router it is chilling at 10%

The theoretical Speed on Download should be something around 20mb/s but actually its at Zero, so no download possible or if a download is running its only on a speed around 2 MB/s. 
After a little Time it is "switching" to the 1UND1 Connection only and the Speed go up 
The "Master" in this Test was 1UND1

[image]

For explanation i only use 2 Connections still to test it and the Master is the 1UND1 again. 
Change the Master to STARLINK the Download is like in the First test around 2 MB/s and it remains on this Speed.
[image]

Do you have any more suggestions for Testing?

zimboman, 2024-12-16:
I have now found the solution to my problem. 

### SOLUTION ### 
For all those who have the same problem - for me it was enough to change the VPN protocol to the VPS, which not only caused the bandwidth distribution to work on all lines, but also that UDP traffic goes over the connection instead of TCP. 

The solution is as follows:
System > OpenMPTCProuter > “Show advandced Settings” -check the box > “Default Proxy” -set to “V2Ray VLESS”

It also helped to squeeze the last remnants out of the lines by also entering the “contractually guaranteed” speeds in the “Settings Wizard” for the WAN lines. 

Download and upload now work as intended and presumably as the solution was intended ^^ 
[image]

closed 2024-12-16 by zimboman

Makeshift, 2024-12-29:
I have a somewhat similar issue, but I'm not sure if it's worth opening an issue for. 

In my case, I have two WANs:
- WAN1 is a stable ADSL line
- WAN2 is an unstable 4G/5G connection that must use MPTCP over VPN, which rules out using VLESS (I think - it says MPTCP over VPN only works with Shadowsocks as the proxy, so I'm using the 'Shadowsocks' proxy instead of the default Shadowsocks Rust)

If 'master' is set to WAN1, WAN2 will not be utilized at all. If 'master' is set to WAN2, it will utilize both as expected - but this isn't ideal as WAN2 is unstable.

I'm commenting here since the OP has recently had a similar issue, so curious if it's related.

Ysurac, 2024-12-29:
This issue is closed. Open a new one.