Ysurac/openmptcprouter #3713: Aggregating through VPS with wireguard

SpencerXZX, 2024-12-17:
## Expected Behavior
WAN1 = Starlink
WAN2 = Cellular 5G

OpenMPTCPRouter runs on NanoPi R5S and bonds the two connections.
I'm wanting to use wireguard to also bond the two connections for UDP traffic and make use of redundant traffic, the goal here is low latency gaming using the best packets from two connections.

I already have a VPS ready to go.

I didn't want to use Glorytun because it's not been updated in 4 years and uses slower encryption.

## Current Behavior

None yet, looking for advice on how to make it work.

## Specifications
None yet
  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

AKHwyJunkie, 2024-12-17:
Are you trying to get the tunnel between the router and VPS to use Wireguard here?  If so, I don't think that's an option.  Wireguard can be used for MPTCP over VPN, but that's only necessary if your carriers don't pass MPTCP traffic.  The tunnel options are defined in the "Default VPN" under the Advanced settings in the settings wizard.  In my testing of the various VPN tunnels, I've found Glorytun to be the fastest and most reliable.

Ysurac, 2024-12-18:
MPTCP can aggregate only TCP traffic.
It's why all available proxy are TCP proxy. Only Glorytun UDP and MLVPN are UDP proxy and have own method to aggregate connections (but doesn't always work correctly...).

ccmks, 2025-01-08:
How about XRAY VMESS, VLESS, Trojan? I was able to aggregate UDP on it as long as I put the check mark on enable UDP for V2RAY/XRAY

xzjq, 2025-01-11:
In general, you probably do not want to aggregate UDP via OMR. I have experienced extreme penalties in performance for doing do, which is the ironic, opposite outcome vs your (and my) goal.

"However, given that the total aggregated [UDP] speed of the three WANs in this case is ~1/3 of the non-aggregated master WAN, this is not a benefit..."
https://github.com/Ysurac/openmptcprouter/issues/3067#issuecomment-1874674322

closed 2025-01-17 by SpencerXZX