Ysurac/openmptcprouter #3715: Keep origin IP Portforwarding

bennizone, 2024-12-18:
I sucessfully created an Port 443 forward described in Wiki (tried both - VPN an v2ray)
I forwarded to an reverse Proxy. All incoming Traffic to this Proxy seems to come from omr LAN IP. 
Is it possible to change this? I wish I could see the origin IP at the reverse Proxy. So I can create rules based on that. 

## Specifications
  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: Hetzner VPS
  - OpenMPTCProuter platform: x86_64

I already tried to find a solution on the web but wasnt successfull.

winkmichael, 2024-12-19:
I'm slightly confused by your question, but I believe the answer is the IP address that is coming from you LAN is being NAT'd through your router. If that is correct, the answer is basically no the IP will always be your router (e.g. .1)

bennizone, 2024-12-19:
Excuse my bad English. I'll try to explain.
In the past I had double NAT: ISP router --> opnSense --> reverse proxy.
I had a rule on the reverse proxy that some sites required basic authentication to access. The reverse proxy always showed whether you were coming from outside or not.
Now with the omr everything seems to come from the omr LAN IP and I can't distinguish between internal access and external access.

closed 2025-03-25 by github-actions[bot]