Ysurac/openmptcprouter #3716: GloryTun VPN disabled and OpenVPN re-enabled any time I use the wizard to change settings

SpencerXZX, 2024-12-20:
## Expected Behavior

I am trying to use GloryTun as my VPN, I expect that setting to stay.

## Current Behavior

GloryTun VPN is disabled and OpenVPN is re-enabled every time I change settings in the wizard. Is OpenVPN a requirement?

## Specifications

  - OpenMPTCProuter version: 0.62-snapshot-6.6
  - OpenMPTCProuter VPS version: 0.1032-test 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: NanoPi R5S

SpencerXZX, 2024-12-20:
Also somewhat related to this issue, I had to manually configure the IP for Glorytun UDP on OMR for it to work, I used these commands:

ip addr add 10.255.254.2/30 dev tun0
ip link set tun0 up

Is this normal?

Ysurac, 2024-12-20:
I would need the result of Status->System log and via SSH on the router, the result of ```uci show openvpn```

SpencerXZX, 2024-12-20:
I think the issue here is that I'm enabling Glorytun through the VPN menu and not through the Wizard, however I don't see any VPN selection in the wizard? I am switching to x86 tomorrow for more compute power and can try again then. But is there a way to change VPN type in the wizard?

Ysurac, 2024-12-20:
yes this must be done via the wizard, click on the "Advanced settings" checkbox.

SpencerXZX, 2024-12-21:
Hi Ysurac, thanks for the help. That solved this issue. I'm very close to getting this setup working. Everything works with the default scheduler, however as soon as I enable the redundant scheduler I lose internet connection. I've tried a variety of different proxies and get the same result with all of them, is this something you've seen before? What logs will help troubleshoot this? 

Thanks,

SpencerXZX, 2024-12-21:
Here's my syslog:

https://pastebin.com/mKX2Z2eT

SpencerXZX, 2024-12-22:
Update: I switched to an x86_64 config with different hardware and have this same problem. This time the whole system became unresponsive and I had to power cycle and quickly change MPTCP scheduler before it became unresponsive again.

Ysurac, 2024-12-22:
I will check scheduler after christmas, I may have to update the patch due to kernel update.

SpencerXZX, 2024-12-28:
Hi Ysurac, does the 6.12 kernel already have this patch? Or are the two codebases the same so the scheduler doesn't work on either? I saw you have done some 6.12 kernel updates so I was wondering if that is the preferred snapshot kernel version?

Ysurac, 2024-12-28:
I didn't tested yet on kernel 6.6.
Snapshot with kernel 6.12 is not yet compiled automatically, I need to tests.

For kernel 6.6, do you have same crash issue when you test on the router directly ? With ```omr-test-speed``` for example ?

SpencerXZX, 2024-12-30:
> I didn't tested yet on kernel 6.6. Snapshot with kernel 6.12 is not yet compiled automatically, I need to tests.
> 
> For kernel 6.6, do you have same crash issue when you test on the router directly ? With `omr-test-speed` for example ?

When I enabled the redundant scheduler, the whole system crashed. The router became unresponsive to ssh, ping, and the web UI. I had to power cycle it and I was able to log in and change the scheduler back before it crashed again. There's about 10 seconds after boot where the web UI is responsive so I had to race to change the setting back.

SpencerXZX, 2025-01-05:
I see the commit for minRTT scheduler, is this the same as redundancy mode?

closed 2025-01-17 by SpencerXZX