Ysurac/openmptcprouter #3719: Unable to install mptcp VPS using scripts on Debian 12.8.0

lkimloong, 2024-12-28:
Hi, 

May I know usually what would be the common issue that unable to install the script on Debian x86 machine running version 12.8.0 on kernel 6.1. Do I need to update Debian 12.8.0 kernel from 6.1 to 6.6 in order to install the VPS scripts?

regards,

Ysurac, 2024-12-28:
Strange to use a GUI on a server...
What is the error ?

lkimloong, 2024-12-28:
Using GUI for easy manage. Here are the error screenshot.
[image]

Ysurac, 2024-12-28:
It's due to wrong/old repo set in /etc/apt/sources.list. It's not directly related to VPS script.
It's a fresh 12.8 install or an update ?

lkimloong, 2024-12-28:
The scripts did not install the mptcprouter. It is a fresh install of 12.8. Currently running on kernel 6.1. Does the 12.8 VPS need kernel 6.6 as well? If GUI is not recommended I can perform a fresh install again. Fyi, the VPS is currently running on a Dell WYSE 7040 machine. The mptcprouter is also running a Dell WYSE 7040 but I can manage to install it.

Ysurac, 2024-12-28:
The problem is only bad repo. You have same error with ```apt-get update```.
It's a personal install ? Not a VPS ?
It doesn't need kernel 6.6, it will install it.
You need to solve entries in ```/etc/apt/sources.list``` and ```/etc/apt/sources.list.d```.

You should have something like this in ```/etc/apt/sources.list```:
```
deb https://deb.debian.org/debian bookworm main contrib non-free
deb-src https://deb.debian.org/debian bookworm main contrib non-free

deb https://debian.mirror.constant.com bookworm main contrib non-free
deb-src https://debian.mirror.constant.com bookworm main contrib non-free

deb https://deb.debian.org/debian-security/ bookworm-security main contrib non-free
deb-src https://deb.debian.org/debian-security/ bookworm-security main contrib non-free

deb https://deb.debian.org/debian bookworm-updates main contrib non-free
deb-src https://deb.debian.org/debian bookworm-updates main contrib non-free
```

lkimloong, 2024-12-29:
Yes, is a personal install and not a VPS. Will this make a different? I will try to reinstall again and check the sources entry again.

Ysurac, 2024-12-29:
It's different because I can't test or adapt to the provider.
Also make sure VPS is not at home, else this will be difficult to aggregate anything.

lkimloong, 2025-01-02:
I try to install to cloud VPS and it is working fine. I am using OVH provider as per recommended but I found that the download speed that i have after bond 3 carrier provider is less than 10mbps and my upload speed is 130mbps which is good. Not sure what is happen to the download speed. When I try on single carrier I can get download about 50mbps. 

Also is there any chances the VPS work if install at home with the DDNS?

closed 2025-04-08 by github-actions[bot]