Ysurac/openmptcprouter #3720: No USB Ethernet after upgrade to 6.0-6.6 on Raspberry PI 4B

murateri75, 2024-12-28:
## Expected Behavior
Have eth1 and eth2 (USB network adapters : Realtek 0bda:8153 Realtek USB 10/100/1000 LAN)

## Current Behavior

No eth1 and eth2 
lsusb gives 
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.50 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux 6.6.50 xhci-hcd xHCI Host Controller
Bus 002 Device 002: ID 0bda:8153 Realtek USB 10/100/1000 LAN
Bus 002 Device 003: ID 0bda:8153 Realtek USB 10/100/1000 LAN
no r8152 module in the kernel

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5 / LuCI Master 24.212.79282~65b8002
  - OpenMPTCProuter VPS version: 6.1.0-22-amd64 0.1030
  - OpenMPTCProuter VPS provider: IguaneSolutions
  - OpenMPTCProuter platform: RPI4 :Raspberry Pi 4 Modèle B (4 Go) Quad...

murateri75, 2024-12-29:
I used r8152.ko from rpi5 packages
kmod-usb-net-rtl8152_6.6.50-r1_aarch64_cortex-a72.ipk
Untar and copy files in /etc/modules.d and /lib/modules
Everything find now I got back eth1 and eth2 
Sorry for creating this issue

closed 2024-12-29 by murateri75

Ysurac, 2024-12-29:
It should also be possible via System->Software menu