Ysurac/openmptcprouter #3723: The aggregation router has enabled vps but cannot access the network

lhc1993, 2024-12-29:
Wǒ de jùhé lùyóuqì yǐjīng chénggōng liánjiē dào vps, dànshì fǎngwèn wǎngluò, huì fǎnhuí ERR_CONNECTION_REFUSED, zhège shì shénme yuányīn a,
60 / 5,000
My aggregation router has been successfully connected to the vps, but when accessing the network, it returns ERR_CONNECTION_REFUSED. What is the reason?
[image]

lhc1993, 2024-12-29:
I found that the network accessed by the application is also possible, but the access in Chrome cannot connect to the network

Ysurac, 2024-12-29:
Please use template.
Try with less interfaces where multipath is nebaled.

lhc1993, 2024-12-30:
After the vps was installed, there was no change. The client router was changed to use a pcie sim card 5G module and local network broadband, ipv6 was enabled, and everything else was default.

lhc1993, 2024-12-30:
Try with less interfaces where multipath is nebaled.
-- After turning off multipath, it is OK

Ysurac, 2024-12-30:
Try with one interface set as master and if it's not working, check Status->System log

lhc1993, 2024-12-31:
There are also problems with using one interface. For example, www.baidu.com is accessed differently, but if you turn on the local VPN proxy on your computer, you can access it. Can you give me some ideas or suggestions for troubleshooting?

lhc1993, 2025-01-01:
Jan  1 13:54:21 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 13:54:45 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 13:54:57 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 13:54:57 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 13:54:57 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 13:55:20 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 14:03:52 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address
Jan  1 14:04:11 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address

Ysurac, 2025-01-01:
The DSCP is not a problem, only an address without IPv6 or IPv6 not enabled.
Check if you can access baidu.com from the VPS: ```curl -L https://www.baidu.com```

lhc1993, 2025-01-02:
It is accessible in VPS.
Another phenomenon is that my local client opens a VPN, such as clash, can access normally, so I do not know where the DNS is faulty.
Is the DNS resolution obtained when the package is unpacked locally, or is the DNS obtained when the package is combined on the VPS?

Ysurac, 2025-01-02:
To test if there is no DNS issue, check that you can access http://1.1.1.1
I'm not sur to understand what DNS package you are speaking about

lhc1993, 2025-01-03:
Clear the phone again in U-Boot mode and then restore it. Thank you, big guy.

closed 2025-01-03 by lhc1993