Ysurac/openmptcprouter #3726: Netgear R7800 flash fail

vgnx, 2024-12-30:
## Expected Behavior

Hello,

I'm trying to install openmptcprouter on my Netgear R7800 (Nighthawk X4S) but the firmware provided on [the download page](https://www.openmptcprouter.com/download) is to large.

## Current Behavior

As I understand, this router has a 32Mo limit and the provided firmware' size it 48Mo.
I'm using [`nmrpflash`](https://github.com/jclehner/nmrpflash) to do it and I'm facing to the [tftp-block-rollover-upload](https://github.com/jclehner/nmrpflash?tab=readme-ov-file#tftp-block-rollover-upload-might-fail) error.

As the firmware is available on the download page, I wonder if there is any trick to make it works.

Thank you for your help

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-01-02:
The device should have 128MB NAND Flash and doesn't seems to have some partition limitations.
You can use: https://openwrt.org/toh/netgear/r7800 as flash doc. If you try OpenWRT, you can upgrade to OpenMPTCProuter if you don't keep the settings.

vgnx, 2025-01-02:
Thank you for your answer.
I'll give it a try and get back to you

Lore239, 2025-01-02:
Hello, I'm experiencing the same issue with my R7800 router. Initially, I installed the OpenWRT stock firmware from [this page](https://openwrt.org/toh/netgear/r7800), and everything worked fine. However, after upgrading with the OpenMPTC firmware sysupgrade, the router became unreachable. When I connect my PC to the router, it fails to obtain an IP address. The router appears to boot normally, and the port light blinks when I plug in the cable, but my PC still cannot get an IP address and I dont know how to reach the router web page.

Ysurac, 2025-01-02:
@Lore239 OpenMPTCProuter change ports order: WAN port is used as LAN and LANs port are used for WAN.

Lore239, 2025-01-02:
[image]
Without modifying any configurations, these are the current interfaces. I'm unsure if this error is expected, but none of the ports appear to be functioning.

Ysurac, 2025-01-02:
A driver seems to be missing here.
Can you give me, via ssh, the result of ```lsmod``` ?

Lore239, 2025-01-02:
root@OpenMPTCProuter:~# lsmod
3c59x                  32768  0 
6lowpan                20480  0 
8139cp                 20480  0 
8139too                20480  0 
8250_pci               40960  0 
8390                   16384  1 ne2k_pci
act_csum               16384  0 
act_gact               12288  0 
act_mirred             12288  0 
act_pedit              16384  0 
act_simple             12288  0 
act_skbedit            12288  0 
act_skbmod             12288  0 
af_key                 28672  0 
ah4                    12288  0 
ah6                    16384  0 
ahci                   20480  0 
ahci_platform          12288  0 
ark3116                12288  0 
arp_tables             16384  1 arptable_filter
arpt_mangle            12288  0 
arptable_filter        12288  0 
asix                   24576  0 
asn1_decoder           12288  1 nf_nat_snmp_basic
ath                    20480  3 ath9k_htc,ath9k_common,ath9k_hw
ath9k_common           20480  1 ath9k_htc
ath9k_htc              53248  0 
ath9k_hw              323584  2 ath9k_htc,ath9k_common
atl1                   28672  0 
atl1c                  32768  0 
atl1e                  28672  0 
atl2                   24576  0 
atm                    32768  2 solos_pci,br2684
authenc                12288  1 authencesn
authencesn             12288  0 
autofs4                24576  1 
ax88179_178a           20480  0 
ax88796b               12288  0 
b44                    24576  0 
bcm_phy_lib            16384  1 broadcom
belkin_sa              12288  0 
bnx2                   65536  0 
bonding               102400  0 
br2684                 16384  0 
broadcom               16384  0 
cbc                    12288  0 
cdc_acm                20480  0 
cdc_eem                12288  0 
cdc_ether              12288  2 rndis_host,cdc_ncm
cdc_mbim               12288  0 
cdc_ncm                20480  2 cdc_mbim,huawei_cdc_ncm
cdc_subset             12288  0 
cdc_wdm                16384  3 cdc_mbim,qmi_wwan,huawei_cdc_ncm
cfg80211              241664  5 ath9k_htc,ath9k_common,ath,mt7601u,mac80211
ch341                  12288  0 
chacha_neon            28672  1 libchacha20poly1305
cls_basic              12288  0 
cls_flow               16384  0 
cls_fw                 12288  0 
cls_matchall           12288  0 
cls_route              12288  0 
cls_u32                20480  0 
cmac                   12288  0 
compat                 12288  5 ath9k_htc,ath9k_common,mt7601u,mac80211,cfg80211
configs                49152  0 
cp210x                 20480  0 
crc_ccitt              12288  1 ppp_async
crc_itu_t              12288  1 mmc_spi
crc32c_generic         12288  1 
[... 244 lines cut ...]
xt_CT                  12288  0 
xt_DSCP                12288  0 
xt_HL                  12288  0 
xt_LOG                 12288  0 
xt_MASQUERADE          12288  0 
xt_REDIRECT            12288  0 
xt_TCPMSS              12288  0 
xt_comment             12288  0 
xt_conntrack           12288  0 
xt_dscp                12288  0 
xt_ecn                 12288  0 
xt_hl                  12288  0 
xt_length              12288  0 
xt_limit               12288  0 
xt_mac                 12288  0 
xt_mark                12288  0 
xt_multiport           12288  0 
xt_nat                 12288  0 
xt_state               12288  0 
xt_statistic           12288  0 
xt_tcpmss              12288  0 
xt_tcpudp              12288  0 
xt_time                12288  0 
zram                   16384  0 
zsmalloc               16384  1 zram

Ysurac, 2025-01-02:
There is ath9k but not ath10k that should be here. I will check.

Ysurac, 2025-01-02:
I added ath10k for wifi and qca8327 for switch driver. It's compiling...

Lore239, 2025-01-03:
I will see the new version [there](https://www.openmptcprouter.com/download)?

Lore239, 2025-01-04:
I installed **openmptcprouter-v0.62-snapshot-6.6-r0+28194-cc69be0c13-ipq806x-generic-netgear_r7800-squashfs-sysupgrade.bin** but nothing changed

Ysurac, 2025-01-04:
The image compiled yesterday ? Should be ok... I will test but this will take more time.

Lore239, 2025-01-04:
yes, I installed the sysupgrade file from [here](https://snapshots.openmptcprouter.com/6.6/r7800/targets/ipq806x/generic/) but still the same error, the upgrade. Then I installed the standard OpenWRT [23.05.5](https://openwrt.org/releases/23.05.5) and everything works fine, I sent you the list of drivers in the hope that it can help to find maybe other missing drivers.

Lore239, 2025-01-18:
Hi, is there anything new? I tried every snapshot but nothing works

closed 2025-04-23 by github-actions[bot]