Ysurac/openmptcprouter #3727: Installing on Google Wifi with Openwrt os

NightWalker77V, 2024-12-30:
## Expected Behavior

Hello there, I have a Google Wi-Fi that has OpenWrt on it, and I wanted to use OpenMPTCProuter to bond multiple internet sites to get higher speed and bandwidth, but I don't know which image I should get from the Download page. I have it installed on my VPS. Please guide me on which one I should download and flash on my router.
## Current Behavior

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2024-12-30:
OpenMPTCProuter is not compiled for this platform. You can compile it yourself but it's not easy to install on this and the CPU is slow.

NightWalker77V, 2024-12-30:
> OpenMPTCProuter is not compiled for this platform. You can compile it yourself but it's not easy to install on this and the CPU is slow.

Oh... So like there's no way for me to combine multiple connections with this router? Any other substitutes?
This is the Status of it:

[image]

closed 2025-04-04 by github-actions[bot]