Ysurac/openmptcprouter #3728: bug install vps

esquilo444, 2024-12-30:
hello,

I just reinstalled my VPS and nothing works. Via putty, as I have done dozens of times, but the script blocks ... weird ... am I the only one?

Thanks

Jérôme

Ysurac, 2024-12-30:
What is the error ?

esquilo444, 2024-12-30:
[image]

esquilo444, 2024-12-30:
quelque soit le logiciel ou la version du script , toujours le même retour d'info ..

esquilo444, 2024-12-30:
VPS sur Hostinger 
debian 11 et 12 idem

Ysurac, 2024-12-30:
Wait and try again. Can be a CDN issue...
Check if ```curl https://www.openmptcprouter.com``` give you an output. Check also ```curl -L https://1.1.1.1```.

esquilo444, 2024-12-30:
retour des commandes ... "curl https://www.openmptcprouter.com give you an output. Check also curl -L https://1.1.1.1 "

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Dec 30 15:26:36 2024 from 82.67.123.2
***@***.***:~# curl https://www.openmptcprouter.com
<!doctype html>
<html lang="en">
        <head>
                <meta charset="utf-8">
                <meta name="viewport" content="width=device-width, initial-scale                                                                                                                                   =1, shrink-to-fit=no">
                <meta name="description" content="OpenMPTCProuter permit to aggr                                                                                                                                   egate multiple Internet connections with the help of Multipath TCP (MPTCP) and s                                                                                                                                   hadowsocks">
                <meta name="author" content="Ycarus">
                <meta name="flattr:id" content="8o9lkk">
                <meta property="og:title" content="OpenMPTCProuter">
                <meta property="og:site_name" content="OpenMPTCProuter">
                <meta property="og:url" content="https://www.openmptcprouter.com                                                                                                                                   /">
                <meta property="og:image" content="https://www.openmptcprouter.c                                                                                                                                   om/img/statuspage.png">
                <meta property="twitter:image" content="https://www.openmptcprou                                                                                                                                   ter.com/img/statuspage.png">
                <link rel="icon" type="image/png" href="img/favicon.png">
                <link rel="apple-touch-icon" href="img/favicon.png" />
                <title>OpenMPTCProuter - Internet connection bonding - Home</tit                                                                                                                                   le>
                <style>#donate a{background:#000;color:#fff;text-decoration:none                                                                                                                                   ;font-family:arial,sans-serif;text-align:center;font-weight:bold;padding:5px 40p                                                                                                                                   x;font-size:1rem;line-height:2rem;position:relative;transition:0.5s;}#donate a:h                                                                                                                                   over{background:#c11;color:#fff;}#donate a::before,#donate a::after{content:"";w                                                                                                                                   idth:100%;display:block;position:absolute;top:1px;left:0;height:1px;background:#                                                                                                                                   fff;}#donate ***@***.*** screen and (min-width:1200px){                                                                                                                                   #donate{position:fixed;display:block;top:0;right:0;width:200px;overflow:hidden;h                                                                                                                                   eight:200px;z-index:9999;}#donate a{width:200px;position:absolute;top:30px;right                                                                                                                                   :-60px;transform:rotate(45deg);-webkit-transform:rotate(45deg);-ms-transform:rot                                                                                                                                   ate(45deg);-moz-transform:rotate(45deg);-o-transform:rotate(45deg);box-shadow:4p                                                                                                                                   x 4px 10px rgba(0,0,0,0.8);}}</style><span id="donate"><a href="https://www.payp                                                                                                                                   al.com/donate?business=HPXWV8QZE7KG6&no_recurring=0&currency_code=EUR">Donate</a                                                                                                                                   ></span>
                
                <link rel="stylesheet" href="css/bootstrap.min.css">
                <link href="css/openmptcprouter.css" rel="stylesheet">
        </head>
        <body>
                <nav class="navbar navbar-expand-xl navbar-dark fixed-top bg-omr                                                                                                                                   ">
[... cut ...]
<li>Shadowsocks-Rust support on router side and Shadowsocks-Go support on server                                                                                                                                    side</li>
<li>Automatic speedtest option to set download/upload speed for SQM and Glorytun                                                                                                                                    UDP</li>
<li>OMR-ByPass can be used without nDPI for services bypass</li>
<li>OMR-ByPass can bypass via a VPN set on server side</

esquilo444, 2024-12-30:
j'ai essayé avec une autre connexion internet ,via mon téléphone ... idem 
je ne sais pas si vous avez le même problème ..

Ysurac, 2024-12-30:
As you can reach https://www.openmptcprouter.com, try only ```wget https://www.openmptcprouter.com/server/debian-x86_64.sh```. There is no reason this doesn't work. Check also that you have free space with ```df -h``` and that you can write file.

esquilo444, 2024-12-30:
[image]
pas beaucoup mieux . Le seul point positif, c'est je ne suis pas obliger de relancer putty après avoir utilisé la derniere commande .

Ysurac, 2024-12-30:
It's working...
So complete procedure:
```
rm debian-x86*
wget https://www.openmptcprouter.com/server/debian-x86_64.sh
KERNEL="6.6" sh debian-x86_64.sh
```

kpma1985, 2024-12-30:
hmm  not working for me

kpma1985, 2024-12-30:
struggle at wget -4 -qO- -T 2 http://hostname.openmptcprouter.com

cloudflare dns server

esquilo444, 2024-12-30:
I'm almost reassured that I'm not the only one. 😅

esquilo444, 2024-12-30:
i don't understand what happened ...

esquilo444, 2024-12-30:
I tried with a VPS under ubuntu ... ditto ...
maybe it's the vps host that blocks "hostinger" ... yet so far no problem

Ysurac, 2024-12-31:
It's fixed. The VM was crashed. I will migrate it to another server, but it's not the best time...

esquilo444, 2024-12-31:
Thanks a lot .
I will try this morning ..

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

darkman1983, 2025-01-04:
It's back again...

Ysurac, 2025-01-04:
Seems to be working.
Else there is a timeout on commands and you can also specify domain via VPS_DOMAIN and IP via VPS_PUBLIC_IP the same way as KERNEL.