Ysurac/openmptcprouter #3729: Flint 2 - can it work?

speegs, 2024-12-30:
Seeking advice on this router, does anyone have it working with openmptcprouter?  Seems like good hardware specs.

https://store.gl-inet.com/products/flint-2-gl-mt6000-wi-fi-6-high-performance-home-router

Since I have on, got tempted by a good price, figured if anyone has a path to follow to get it to run I'm happy to attempt to follow it.

Or if a complete dead end, let me know and I'll just move on and use it for what it is with Gl-Inet's firmware.

q2go, 2024-12-30:
In fact, the [GL.inet GL-MT6000](https://openwrt.org/toh/gl.inet/gl-mt6000) is currently one of the most popular devices for OpenWrt because of its good performance and features. It is also readily available at a reasonable price point.

But I suppose you can only support device which you got access to @Ysurac, right?

kpma1985, 2024-12-31:
i built and run it successfully
[image]

kpma1985, 2024-12-31:
[image]

speegs, 2025-01-01:
That's awesome, never built anything for an ARM CPU before, could you point me in the right direction to replicate your success?

speegs, 2025-01-01:
after googling the cpu, seems very similar to a https://wiki.banana-pi.org/Banana_Pi_BPI-R3

Ysurac, 2025-01-01:
I can compile it for Flint 2, but I can't test it as I don't have the router.
I tried for BPI-R3/R4 without having the box, and it's not working... So you will see...

kpma1985, 2025-01-01:
I can also Upload for you

Ysurac, 2025-01-01:
@kpma1985 you made no network or package changes ?
If this is working, I would add it as an "official" support, so a platform available in all snapshots and next releases.

speegs, 2025-01-01:
Seems like I'm best to wait a little while for official support/download, as people with way skills are working on it :)

kpma1985, 2025-01-01:
> @kpma1985 you made no network or package changes ? If this is working, I would add it as an "official" support, so a platform available in all snapshots and next releases.

i didnt change any files. work out of the box

https://mega.nz/folder/zfpEiLjb#6yE54UFTA9Ccy13bdoP46g

Ysurac, 2025-01-03:
Snapshot, not tested image is available here: https://snapshots.openmptcprouter.com/6.6/gl-mt6000/targets/mediatek/filogic/

speegs, 2025-01-05:
Didn't have an opportunity to test it yet, but hopefully in about a week I'll be in a position to try it.  I'll report back when I get the chance.

kpma1985, 2025-01-05:
@Ysurac  can you also build for mt3000 ? would be nice if that is supported too#

KaWiSo, 2025-01-06:
I also have an MT6000, i was using with openmptcprouter up until kernel 6.6.
Even without retaining the config, any 6.6 builds I have tried including the prebuilt snapshots have not worked for me.
The interfaces come up on my switch but there is not a single packet coming in.
It seems as if it was not able to properly start even though the led turns white.

With 6.1-based builds (most recent one I built is from 02.12) I never had any problems booting them.

Houshasei, 2025-01-07:
Looking forward for Flint 2 support, would be nice so I can ditch my MiniPC for OMR.

Ysurac, 2025-01-08:
The snapshot image doesn't seems to work, as I don't have the device it's difficult to solve.

speegs, 2025-01-14:
I had a go with both firmware files listed in this thread.

Got this error:

The firmware is not compatible with this device.

[image]

_

speegs, 2025-01-14:
Also tried uboot method, but also failed.  Must not like something in the files.

[image]

Ysurac, 2025-01-14:
Can you test the sysupgrade file instead ?

speegs, 2025-01-17:
sysupgrade image looks more promising:

[image]

speegs, 2025-01-17:
After that got to 100% the router rebooted.

Then solid white light like it's booted.

Then can't get DHCP on any of the ethernet ports.

Setting my PC to 192.168.100.10 and trying 192.168.100.1 didn't get a result.

So I held the reset button while powering up (got the firmware recovery mode) and recovered the firmware back to Gl-Inet's firmware.

Houshasei, 2025-01-17:
@speegs can you try the build by @kpma1985 on his mega download link?

speegs, 2025-01-17:
This is a good sign

[image]

Ysurac, 2025-01-17:
I would need the result of ```uci show network``` and ```ip a```, to check network configuration. Else a network module may be missing in my image...

speegs, 2025-01-17:
This was using the openmptcprouter-v0.62-snapshot-master-2313389e-r0+24843-acf40c022e-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin

On mega.

I have a single 1gbit uplink and that's going via a VPN Server.  Connected straight away.

Be warned the eth1 on the Flint2 is the 2.5GBit WAN port.  After the firmware.  So the ports might be a bit weird and need remapping.  I'm still working on that.  Trying to add a second wan soon and test iPhone via USB port.

speegs, 2025-01-17:
Outputs for Ysurac.

https://drive.google.com/drive/folders/1Q63-WEOcAjahxjoQVLQSf1iDS-WbyY-n?usp=sharing

Ysurac, 2025-01-17:
The config from ```uci show network``` give really strange results... I will check and try to fix all.

speegs, 2025-01-17:
Port Mapping physical - software:

Initially WAN 2.5G will be your 192.168.100.1 interface, you can change that.

Ports:

WAN 2.5G = eth1
WAN/LAN1 2.5G = Lan1
LAN2 = Lan2 
LAN3 = Lan3
LAN4 = Lan4
LAN5 = Lan5 

I plugged an iphone in it wanted to call that eth2, but I might have crashed something at that point.  Will try that again.

speegs, 2025-01-17:
With the openmptcprouter Wizard, mapping LAN1-5  as a WAN works fine.  As long as you leave yourself with 1 LAN of course.

The Flint2 also has a USB 3 port.  It did seem to detect an iphone plugged in and an eth2 was available.  I'll try and test it later.  I managed to crash the router, messing around with that.  It could have been me doing something wrong.  A power cycle and it was all good again.

Plenty more to test, but the main functionality of it using the ethernet ports and bonding them seems to work and it seems to be able to push 900+ mbit via the default shadowsocks setup.

Excited: 

The Flint2 *if stable* eventually looks like a little powerhouse of a router, has lots of ports, wifi built in and the CPU is fast for the price of the router.

kpma1985, 2025-01-17:
I had a lot of stuggle with the Ports. I can Share the configs of wanted.

KaWiSo, 2025-01-17:
> i didnt change any files. work out of the box
>
> https://mega.nz/folder/zfpEiLjb#6yE54UFTA9Ccy13bdoP46g

This firmware works for me, but that is to be expected as its based on the 6.1 kernel. 
I was able to build on 6.1 and even 5.4 before that. I have not been able to boot a 6.6 kernel version yet.

Ysurac, 2025-01-17:
Can somebody try OpenWRT image https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin ?
If it works, I will need also ```ip a``` and ```uci show network``` result.

kpma1985, 2025-01-17:
> Can somebody try OpenWRT image https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin ? If it works, I will need also `ip a` and `uci show network` result.

`1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9683:c4ff:fea2:1cc7/64 scope link
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN qlen 1000
    link/ether 94:83:c4:a2:1c:c5 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9683:c4ff:fea2:1cc5/64 scope link
       valid_lft forever preferred_lft forever
4: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
6: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
7: lan5@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
8: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
10: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
    inet 10.10.90.254/24 brd 10.10.90.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fdf7:46bf:c45c::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::9683:c4ff:fea2:1cc7/64 scope link
       valid_lft forever preferred_lft forever

    ~  ssh root@10.10.90.254 uci show network                                                                                                                                                                                                                                                                   ✔  base   at 20:38:43 
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fdf7:46bf:c45c::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4' 'lan5'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ip6assign='60'
network.lan.ipaddr='10.10.90.254/24'
network.lan.gateway='10.10.90.253'
network.lan.dns='1.1.1.1'
network.wan=interface
network.wan.device='eth1'
network.wan.proto='dhcp'
network.wan6=interface
network.wan6.device='eth1'
network.wan6.proto='dhcpv6'`

kpma1985, 2025-01-17:
https://fw.gl-inet.com/firmware/mt6000/release/openwrt-mt6000-4.7.0-1205-1733363900.bin

`root@192.168.8.1's password:
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.globals=globals
network.globals.ula_prefix='fdb6:d564:a3d9::/48'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='lan1' 'lan2' 'lan3' 'lan4' 'lan5'
network.@device[0].macaddr='94:83:c4:a2:1c:c7'
network.@device[1]=device
network.@device[1].name='lan1'
network.@device[1].macaddr='94:83:c4:a2:1c:c7'
network.@device[2]=device
network.@device[2].name='lan2'
network.@device[2].macaddr='94:83:c4:a2:1c:c7'
network.@device[3]=device
network.@device[3].name='lan3'
network.@device[3].macaddr='94:83:c4:a2:1c:c7'
network.@device[4]=device
network.@device[4].name='lan4'
network.@device[4].macaddr='94:83:c4:a2:1c:c7'
network.@device[5]=device
network.@device[5].name='lan5'
network.@device[5].macaddr='94:83:c4:a2:1c:c7'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.8.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.isolate='0'
network.@device[6]=device
network.@device[6].name='eth1'
network.@device[6].macaddr='94:83:c4:a2:1c:c5'
network.wan=interface
network.wan.device='eth1'
network.wan.proto='dhcp'
network.wan.force_link='0'
network.wan.ipv6='0'
network.wan.classlessroute='0'
network.wan.metric='10'
network.wan6=interface
network.wan6.proto='dhcpv6'
network.wan6.device='@wan'
network.wan6.disabled='1'
network.guest=interface
network.guest.force_link='1'
network.guest.type='bridge'
network.guest.proto='static'
network.guest.ipaddr='192.168.9.1'
network.guest.netmask='255.255.255.0'
network.guest.ip6assign='60'
network.guest.multicast_querier='1'
network.guest.igmp_snooping='0'
network.guest.isolate='0'
network.guest.bridge_empty='1'
network.guest.disabled='1'
network.policy_relay_lo_rt_lan=rule
network.policy_relay_lo_rt_lan.lookup='16800'
network.policy_relay_lo_rt_lan.in='loopback'
network.policy_relay_lo_rt_lan.priority='1'
network.tethering6=interface
network.tethering6.device='@tethering'
network.tethering6.proto='dhcpv6'
[... 60 lines cut ...]
5: lan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
6: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
7: lan5@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
8: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
9: ra0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
    link/ether 2a:b0:ab:40:40:30 brd ff:ff:ff:ff:ff:ff
10: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 94:83:c4:a2:1c:c7 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.1/24 brd 192.168.8.255 scope global br-lan
       valid_lft forever preferred_lft forever
11: ra1: <BROADCAST,MULTICAST> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 96:83:c4:a2:1c:c8 brd ff:ff:ff:ff:ff:ff
12: rax0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UNKNOWN group default qlen 1000
    link/ether 92:44:5a:cc:e6:e3 brd ff:ff:ff:ff:ff:ff
13: rax1: <BROADCAST,MULTICAST> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 96:83:c4:a2:1c:c9 brd ff:ff:ff:ff:ff:ff
14: apcli0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 92:83:c4:a2:1c:c8 brd ff:ff:ff:ff:ff:ff
15: apclix0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 9e:83:c4:a2:1c:c8 brd ff:ff:ff:ff:ff:ff
`

speegs, 2025-01-19:
>>Can somebody try OpenWRT image https://downloads.openwrt.org/releases/24.10.0-rc5/targets/mediatek/filogic/openwrt-24.10.0-rc5-mediatek-filogic-glinet_gl-mt6000-squashfs-sysupgrade.bin

I gave that firmware a run, noticed it was cutdown and went back to the firmware on Mega.

It's really fast to boot that firmware, but didn't seem to have openmptcprouter, just the minimum to boot.  I think I like the fallback theme better than the normal theme.  Has the newer kernel.  I didn't bother getting the outputs as kpma1985 already did that.

I'd been having troubles with OMR-Bypass on the "mega" firmware.  I suspect that's just old bugs when I googled the error about AS Number Bypass said it was fixed in newer firmware version.  Used MAC address bypass to get netflix going on my TV.

speegs, 2025-01-21:
Let me know if a newer firmware is compiled and I'll try it.  The one that works from the Mega Download site is an older kernel so not really compatible I think with the newer VPS I'm running.

Thanks for the work by all getting closer to supporting this router, seems like it will get there soon.

Ysurac, 2025-01-22:
A new firmware is compiling, not sure it will be working yet...

kpma1985, 2025-01-22:
I can also try tonight GMT+1

speegs, 2025-01-24:
How did the new firmware go?  Do you need someone to try it?  Send link if ready.

speegs, 2025-01-24:
There is a flint 3 coming soon, might explain some of the discounts on the flint 2.  Significant changes is Wifi 7 and 2.5Gbit for all ports I saw on reddit.  Didn't see anything on CPU specs on reddit yet.

kpma1985, 2025-01-24:
just installed.. working

speegs, 2025-01-24:
where is the link on the official website?

kpma1985, 2025-01-24:
> where is the link on the official website?

https://www.gl-inet.com/products/gl-be9300/?utm_source=website&utm_medium=menubar

if u talk about the firmware?
https://snapshots.openmptcprouter.com/6.6/gl-mt6000/targets/mediatek/filogic/

speegs, 2025-01-24:
@kpma1985 

Thanks, the firmware snapshot, thanks.  That's up and running now.  Will try it out some.  Since it's the newer kernel should work better in theory.

speegs, 2025-01-24:
@kpma1985 

Out of interest which default proxy do you find best?  On the prior image, I switched to XRay VLESS and Glorytun TCP worked well for me.

Will try shadowsocks 2022 (I think that was the default).  See how it goes.

kpma1985, 2025-01-24:
i rolled back.. wifi doesnt work correctly for me

speegs, 2025-01-25:
I had wifi switched, off, because I have several unifi access points figured it would only cause trouble.

However I enabled it and even without any password, wifi indeed doesn't work on the mega firmware or this latest one.  It's set to use lan, but it never seems to hand out an IP address.

You can see it connect, but maybe the wifi is somehow not binding to the right network (that's a guess).  I only tested from my iPhone.

Otherwise this firmware does seem good.  

A Warning: Restoring a backup config didn't go well, kept having troubles getting in touch with the VPN Server.  However putting all the information back in seems ok.  Excluding the wifi, which I normally have off where I'm testing it.  

I have a ceiling mounted access point in the same room, so seems silly to turn the wifi on in that use case.

Now running Shadowsocks-Rust2022 and OpenVPN TCP (the defaults), seems fine.

Physical connections:

NBN: 100mbit/40mbit
5G: 180mbit/6mbit

Bonded:

[image]

Ysurac, 2025-01-25:
What is the problem with wifi ? It's not available at all ? Not working ?
Can I have the result of ```dmesg``` ? A firmware may be missing...

speegs, 2025-01-25:
dmesg output:

root@OpenWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.6.72 (runner@fv-az1384-774) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0+28381-ad98c322cc) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Jan 21 10:03:13 2025
[    0.000000] Machine model: GL.iNet GL-MT6000
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004302ffff (192 KiB) nomap non-reusable secmon@43000000
[    0.000000] OF: reserved mem: 0x000000004fc00000..0x000000004fcfffff (1024 KiB) nomap non-reusable wmcpu-reserved@4fc00000
[    0.000000] OF: reserved mem: 0x000000004fd00000..0x000000004fd3ffff (256 KiB) nomap non-reusable wo-emi@4fd00000
[    0.000000] OF: reserved mem: 0x000000004fd40000..0x000000004fd7ffff (256 KiB) nomap non-reusable wo-emi@4fd40000
[    0.000000] OF: reserved mem: 0x000000004fd80000..0x000000004ffbffff (2304 KiB) nomap non-reusable wo-data@4fd80000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 28 pages/cpu s74792 r8192 d31704 u114688
[    0.000000] pcpu-alloc: s74792 r8192 d31704 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line:  root=PARTLABEL=rootfs rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: SWIOTLB bounce buffer size adjusted to 1MB
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x000000007fae0000-0x000000007fbe0000] (1MB)
[    0.000000] Memory: 1003884K/1048576K available (9728K kernel code, 1568K rwdata, 6752K rodata, 512K init, 318K bss, 44692K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 640 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000247] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000254] pid_max: default: 32768 minimum: 301
[    0.002295] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002303] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.003893] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.004288] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.004401] rcu: Hierarchical SRCU implementation.
[    0.004402] rcu:     Max phase no-delay instances is 1000.
[    0.004815] smp: Bringing up secondary CPUs ...
[    0.005089] Detected VIPT I-cache on CPU1
[... 303 lines cut ...]
[   14.649571] usbserial: USB Serial support registered for suunto
[   14.655502] usbserial: USB Serial support registered for vivopay
[   14.661520] usbserial: USB Serial support registered for zio
[   14.671627] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[   14.679447] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[   14.701721] usbcore: registered new interface driver cdc_mbim
[... cut ...]
[   19.192483] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[   19.194490] mt7530-mdio mdio-bus:1f lan1: configuring for phy/sgmii link mode
[   19.247753] mt7530-mdio mdio-bus:1f lan2: configuring for phy/gmii link mode
[   21.067197] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.214535] mt7530-mdio mdio-bus:1f lan2: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.428290] mtk_soc_eth 15100000.ethernet eth1: Link is Up - 1Gbps/Full - flow control rx/tx

speegs, 2025-01-25:
logread -f output when I enable the two wifi networks (default settings, no password or SSID change).

There is some activity from dhcp as I rebooted the router a few minutes before doing this.  This is without attempting to connect a client.

Jan 25 20:54:25 OpenWrt daemon.notice netifd: wan1 (6209): udhcpc: sending renew to server 125.253.63.161
Jan 25 20:54:25 OpenWrt daemon.notice netifd: wan1 (6209): udhcpc: lease of 203.123.116.76 obtained from 125.253.63.161, lease time 600
Jan 25 20:54:26 OpenWrt daemon.notice netifd: radio1 (27756): WARNING: Variable 'data' does not exist or is not an array/object
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Set new config for phy phy1:
Jan 25 20:54:26 OpenWrt daemon.notice wpa_supplicant[3826]: Set new config for phy phy1
Jan 25 20:54:26 OpenWrt daemon.notice wpa_supplicant[3826]: Set new config for phy phy0
Jan 25 20:54:26 OpenWrt daemon.notice wpa_supplicant[3826]: Set new config for phy phy1
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Set new config for phy phy0: /var/run/hostapd-phy0.conf
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Reloaded settings for phy phy0
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Set new config for phy phy1: /var/run/hostapd-phy1.conf
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Restart interface for phy phy1
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: Configuration file: data: driver=nl80211 logger_syslog=127 logger_syslog_level=2 logger_stdout=127 logger_stdout_level=2 hw_mode=a beacon_int=100 stationary_ap=1 chanlist=36 tx_queue_data2_burst=2.0 #num_global_macaddr=1 #maca
Jan 25 20:54:26 OpenWrt daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->HT_SCAN
Jan 25 20:54:26 OpenWrt user.notice NET: hotplug (iface): action='add' interface='phy1-ap0'
Jan 25 20:54:26 OpenWrt daemon.notice netifd: Wireless device 'radio0' is now up
Jan 25 20:54:26 OpenWrt daemon.notice netifd: Wireless device 'radio1' is now up
Jan 25 20:54:26 OpenWrt daemon.info ModemManager[28107]: hotplug: add network interface phy1-ap0: event processed
Jan 25 20:54:26 OpenWrt daemon.warn [28945]: <wrn> [1737838466.845933] [base-manager] additional port phy1-ap0 in device '/sys/devices/platform/soc/18000000.wifi' added after device probing has already finished
Jan 25 20:54:26 OpenWrt daemon.info [28945]: <inf> [1737838466.845954] [base-manager] last device '/sys/devices/platform/soc/18000000.wifi' probing had failed, will retry
Jan 25 20:54:27 OpenWrt daemon.notice netifd: Network device 'phy1-ap0' link is up
Jan 25 20:54:27 OpenWrt daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->ENABLED
Jan 25 20:54:27 OpenWrt daemon.notice hostapd: phy1-ap0: AP-ENABLED
Jan 25 20:54:29 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.0.221 78:04:73:f9:e8:76
Jan 25 20:54:29 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPNAK(eth1) 192.168.0.221 78:04:73:f9:e8:76 address in use
Jan 25 20:54:30 OpenWrt daemon.notice [28945]: <msg> [1737838470.848537] [base-manager] couldn't check support for device '/sys/devices/platform/soc/18000000.wifi': not supported by any plugin
Jan 25 20:54:34 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.0.221 78:04:73:f9:e8:76
Jan 25 20:54:34 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPNAK(eth1) 192.168.0.221 78:04:73:f9:e8:76 address in use
Jan 25 20:54:38 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(eth1) 192.168.0.221 78:04:73:f9:e8:76
Jan 25 20:54:38 OpenWrt daemon.info dnsmasq-dhcp[1]: DHCPNAK(eth1) 192.168.0.221 78:04:73:f9:e8:76 address in use

speegs, 2025-01-25:
What happens after I connect wifi to OpenWrt network is basically, I don't see a DHCP request from the new wifi connection and no ip address seems to get allocated to the wifi adapter.

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Intel(R) Wireless-AC 9560 160MHz
   Physical Address. . . . . . . . . : 34-CF-F6-80-7C-4E
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::af91:9a6b:f51a:8fe%17(Preferred)
   Autoconfiguration IPv4 Address. . : 169.254.122.92(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.0.0
   Default Gateway . . . . . . . . . :
   DHCPv6 IAID . . . . . . . . . . . : 238342134
   DHCPv6 Client DUID. . . . . . . . : 00-01-00-01-26-BE-2F-C2-A4-BB-6D-BA-09-2D
   NetBIOS over Tcpip. . . . . . . . : Enabled

root@OpenWrt:~# logread -f
Jan 25 20:59:51 OpenWrt daemon.info hostapd: phy1-ap0: STA 34:cf:f6:80:7c:4e IEEE 802.11: authenticated
Jan 25 20:59:51 OpenWrt daemon.info hostapd: phy1-ap0: STA 34:cf:f6:80:7c:4e IEEE 802.11: associated (aid 1)
Jan 25 20:59:51 OpenWrt daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 34:cf:f6:80:7c:4e auth_alg=open
Jan 25 20:59:51 OpenWrt daemon.info hostapd: phy1-ap0: STA 34:cf:f6:80:7c:4e RADIUS: starting accounting session 1452DA3E55D1BB7C

However in the web interface I see this, which is confusing.  Looks like it's ready to work, but doesn't work and Windows (or an iPhone) doesn't seem to pickup an IP address.

[image]

speegs, 2025-01-25:
I could be doing something wrong, but figure it should at least connect and get an ip address, if working normally.

[image]

speegs, 2025-01-25:
One client connected, but many details missing, maybe a hint as to what is happening here:

[image]

speegs, 2025-01-25:
What windows is showing, I'm right next to the router on a dell pc, so heaps of signal.

[image]

KaWiSo, 2025-01-25:
The newest snapshot based on 6.6 works for me now.
Wireless is working as expected contrary to what @speegs is experiencing, which makes it seem like its just a config, not firmware issue.

I did retain the config I was using with 6.1-based builds and wireless instantly came back online.
Releasing and renewing on a wireless client instantly got me a dhcp response as well.

Custom interface configs I did:
 - WAN1 and WAN2 on separate VLANs on eth1
 - lan bridge device for bridge ports lan1-lan5
 - br-lan as lan device in openmptcprouter config and lan-zone as network in wireless
 - separate ssids for 2.4 and 5ghz

ip a:
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group defaul                                                                                                                                                             t qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc mq state UP group defa                                                                                                                                                             ult qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9683:c4ff:fea2:1259/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group defa                                                                                                                                                             ult qlen 1000
    link/ether 94:83:c4:a2:12:57 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9683:c4ff:fea2:1257/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
4: lan2@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master                                                                                                                                                              br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master                                                                                                                                                              br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
6: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master                                                                                                                                                              br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
7: lan5@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master                                                                                                                                                              br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
8: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br                                                                                                                                                             -lan state UP group default qlen 1000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr a642:5491:6800::
10: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
12: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group def                                                                                                                                                             ault qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group def                                                                                                                                                             ault qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 6b0:2e07:be05::
15: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
16: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP gr                                                                                                                                                             oup default qlen 2000
    link/ether 94:83:c4:a2:12:59 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fe80::9683:c4ff:fea2:1259/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
[... cut ...]
network.eth1_7_fw_rule.priority='1'
network.eth1_7_fw_rule.mark='0x45393'
network.eth1_7_fw_rule.lookup='3'
network.eth1_10_fw_rule=rule
network.eth1_10_fw_rule.priority='1'
network.eth1_10_fw_rule.mark='0x45394'
network.eth1_10_fw_rule.lookup='4'
network.tun0_fw_rule=rule
network.tun0_fw_rule.priority='1'
network.tun0_fw_rule.mark='0x45391500'
network.tun0_fw_rule.lookup='1500'
network.all_fw_rule=rule
network.all_fw_rule.priority='1'
network.all_fw_rule.mark='0x4539'
network.all_fw_rule.lookup='991337'

speegs, 2025-01-26:
I got the wifi working.  It involved messing around with the bridging device.

Device: br-lan

It said something about luci had an issue and wanted to upgrade the config to a new format.  So I figured why not.

Then 2 of them appeared with the same name, so I removed the first one, keeping the 2nd one.

Added phy0-ap0 and phy1-ap1 (back in, I had them in there before).

Which are the two Wifi Interfaces.  The instantly wifi picked up an IP and it's working.  

So I'm leaving it alone, not sure if I encountered some kind of bug, but messing around with the bridge interface corrected it.  I believe I just rebuilt it with the same settings via the gui.

Sorry I'm not 100% certain what fixed it or caused it, if you encounter this problem, maybe just destroying the bridge and making a new one might help.

My bridge ports are:

eth1 (which is my LAN), which is actually physically marked with WAN (a 2.5Gbit port)
phy1-ap0 (2.4ghz)
phy1-ap1 (5ghz)

I'm using and this gets confusing:

LAN1 = WAN1 (which is marked as WAN/LAN, a 2.5Gbit port)
LAN2  = WAN2 (1Gbit port)

I did test with an extra WAN3 which was operating via LAN3 for a while, but I removed that.  2 Internet connections is enough for what I'm doing at the moment.

closed 2025-01-26 by speegs

kpma1985, 2025-02-04:
the latest snapshot works fine for me now

JIN7260, 2026-10-23:
Можно ссылку на рабочую прошивку на GL-MT6000