Ysurac/openmptcprouter #3731: Confirming Single NAT Setup (OMR → Ubiquiti) – Is this approach correct?

TipsTricksMore, 2024-12-31:
Hi @Ysurac,

First of all, thanks for all your work on OpenMPTCProuter. OMR is working great for years now!

I’d like to confirm whether the following approach for avoiding dual NAT, while letting OMR handle the multi-WAN bonding, is sound. Could you please let me know if these steps are correct or if there’s anything to watch out for?

That is what I'm aiming for while the Ubiquiti Router should handle all firewall related stuff including port forwarding etc.:

```
   WAN1 -----\
              \
   WAN2 -------+---- [ OMR ] ---- [ Ubiquiti Router (Firewall, main Router)] ---- ( Home LAN Network)
              /
   WAN3 -----/
```

----------------------------------------------------------------------------------------------------

**1. On OMR:**
• Disable NAT/masquerading so it only does multi-WAN aggregation but not firewalling.
(Could you provide more detailed instructions on exactly which OMR settings to change (e.g. which checkboxes or config lines) for turning off NAT and masquerading?)

• Assign OMR a static LAN IP (e.g., 192.168.10.1/24) and turn off DHCP on OMR’s LAN.

• Allow traffic inbound from OMR’s VPN zone to pass directly to the new router without NAT or additional port forwarding.
(Again, any further detail about how to configure the inbound rules or firewall so the traffic is “passed” rather than NATed would be really helpful.)
	
	
**2. On the Ubiquiti Router:**
• Plug the Ubiquiti WAN interface into OMR’s LAN.

• Set the Ubiquiti WAN IP to something like 192.168.10.2/24, default gateway 192.168.10.1.

• Enable NAT and firewall rules here.

• Configure DHCP and all LAN subnets/VLANs behind the Ubiquiti.

• Set up any port forwards and firewall rules on the Ubiquiti, not on OMR.

----------------------------------------------------------------------------------------------------

The idea here is to have a single NAT environment (only on the Ubiquiti), letting OMR handle MPTCP bonding but otherwise staying out of the way. Does that logic hold up in practice, or are there special considerations (e.g., certain ports or advanced traffic rules)?

Also, I’d love more step-by-step detail on how to properly disable NAT/masquerading and allow inbound VPN traffic on OMR. Thanks so much for your time!

Leo

AKHwyJunkie, 2024-12-31:
I have a somewhat similar configuration myself, with a pfSense instance sitting in front of OpenMPTCPRouter.

You do need the NAT/Masquerading to occur on the device actually terminating/managing your public internet connections.  The reason for this is that NAT will translate the RFC1918 (private IP space) into public IP space.  This is also true if your carrier(s) are using RFC1918 space since they'll have no routes to your internal OMR/LAN networks.  (And they will not build them for you, that's what BGP is for.)

So, in your design, you want NAT to occur on the OMR's WAN interface.  You can't actually NAT on your Ubiquiti since it doesn't "have" the public IP addresses (or necessary NAT addresses) that your carriers are giving you.

However, you can establish a "routed" network between your Ubiquiti router and the OMR instance.  This will be "optimal" and will ensure only a single layer of NAT.  (At OMR)  This is what I do and I'm quite happy with it, since OMR now sees all my internal IP's and allows me to use OMR's native inbound NAT/PAT abilities.

The gist (not platform specific) of what you'll need to do to achieve this?

- Have a route for all your internal LAN networks on OMR
- Disable NAT/Masquerading at the Ubiquiti
- The other stuff about DHCP/IP Assignment/etc is also likely necessary, but will depend on your specific integration

"Technically" speaking, you can move inbound NAT/PAT to your Ubiquiti router.  To do this, you use the "Redirect all ports from server to this router" option in System->OpenMPTCPRouter->Advanced Settings.  This "should" still work with the above configurations, it's just that inbound NAT/PAT will now occur at your Ubiquiti.

So, you need to make a few small adjustments to your plan, but that should get you where you want to be.

TipsTricksMore, 2025-01-01:
Hi @AKHwyJunkie,

Thank you so much for your detailed explanation. That’s exactly what I was looking for! 
I’ll test out this setup in a few days as soon as my Ubiquiti gear arrives, and I’ll report back with how it goes.

Thanks again!

AKHwyJunkie, 2025-01-02:
You bet.  Oh, I thought of one other thing you'll need if you go with a routed setup.  (It's been a minute since I did my routed setup with OMR.)

You'll also very likely want to allow any/any on your Ubiquiti WAN interface.  In this design, OMR is functionally the firewall.  Since your Ubiquiti will be a router (not a firewall) at that point, everything will be a lot easier on you if you do this.  Otherwise, you'll have know exactly what kinds of traffic to allow from OMR to your UI setup.  Routers are functionally quite different from firewalls, but really all you want your UI device to do is route your LAN subnets to the OMR subnets.

TipsTricksMore, 2025-01-04:
I just received my Ubiquiti Cloud Gateway Max (CGM), and it’s absolutely amazing!
In a previous post and video, I explained how I’m aggregating to a 2 Gbps downlink. The CGM, with its 5x 2.5 Gbps ports, handles this bandwidth flawlessly. 🤩

**Now for the crucial part—solving the dual NAT issue.**
I’ll break this down step-by-step for anyone who, like me, would have appreciated a detailed guide before diving into this setup.

**Basic Setup**
The outgoing configuration involves a functioning OpenMPTCPRouter (OMR) with its default settings—nothing unusual here.

**Configure the CGM’s WAN Internet Connection:**

**Physical Connection:**
Connect the CGM’s WAN port to an OMR LAN port. Power on the CGM and set it up via the Ubiquiti app.

**Initial Setup:**
At this point, you should already have an active internet connection. However, services requiring forwarded ports, such as a WireGuard server, won’t work yet. Don’t worry—we’ll resolve this shortly.

**Assign a Static IP:**
Set the CGM WAN interface to use a static IP provided by the OMR. Find the Current IP: Access the CGM dashboard to identify the IP assigned by OMR:

[image]

**Configure the Static IP:**
Go to Network > Internet > Primary (WAN 1) and set up a static IPv4 address. Use the IP from the dashboard along with the appropriate subnet mask and OMR gateway IP.
This ensures OMR always assigns the same local IP to CGM, which is crucial for the next step.

**Configure Firewall Port Forwarding Rules:**
On the OMR, create a new firewall rule to forward ports to the CGM:

[image]

**Add a Rule on OMR:**
Forward ports from 0-64999 to the static IP of the CGM (set earlier). Note: Ports above 64999 are reserved for OMR tasks.
This allows the CGM firewall to handle all traffic, enabling further port-specific forwarding for devices behind the CGM via its dashboard.

Keep in mind to set "Source Zone" to **"vpn"** not "wan" on OMR like this when setting up the port forwarding:

[image]

This setup routes all necessary traffic to CGM, where you can manage it further. It also enables services like VPNs or web servers hosted behind the CGM.

**Enjoy Your Setup!**

I can’t really explain how, but that was everything I needed to do—it seemed quite easy!

I was able to use the integrated WireGuard VPN of the CGM immediately, and manual port forwarding for web services, such as a Bitcoin Node exposed to the World Wide Web, worked without any issues.
My hosted Bitcoin Node is also assigned to a separate and isolated VLAN on the CGM for security reasons, which was straightforward to configure.

I also checked for double NAT using traceroute, and it is definitely present:
[image]

This is fascinating to me because, to my knowledge, double NAT can typically cause issues with services like VPNs, VoIP, hosted web servers, video calls, and more. Yet, all of these work flawlessly for me with the described setup. The only downside is the additional ~1 ms latency from the second NAT hop on OMR, which is negligible.

For now, I’m very happy, and everything seems to be functioning perfectly. If any issues arise, I’ll update this post.

Best,
Leo

closed 2025-01-04 by TipsTricksMore