Ysurac/openmptcprouter #3732: VLAN on Raspberry PI 5 are not working

kolchagov, 2024-12-31:
## Expected Behavior

After adding device with VLAN ID on my RPI it doesn't send any packets. 
[image]

Expected: ping 192.168.4.1 (gw)
and ping 192.168.4.2 (mikrotik router) should work

## Current Behavior

I've checked it with ping on my mikrotik gw, no packets are coming from RPI (address 192.168.4.3)
[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. Use Raspberry pi 5
  2. Go to Wizard
  3. Setup VLAN using eth0 and tag ID

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.61-6-6
  - OpenMPTCProuter VPS version: 0.1030

 PACKAGE:     openmptcprouter
 VERSION:     v0.61
 TARGET:      rpi5
 ARCH:        rpi5

 BUILD REPO:  https://github.com/ysurac/openmptcprouter

Ysurac, 2024-12-31:
Check that you have no interface set as MacVLAN.

kolchagov, 2024-12-31:
I don't have any MacVLAN interfaces configured, I'm attaching screenshots of my wan2 (not used at the moment) and all data in /Network/Interfaces section
[image]
[image]

Ysurac, 2024-12-31:
I would need the result of ```ip a```, ```ip r``` and ```dmesg``` via SSH on the router.

kolchagov, 2024-12-31:
Here are the results:
`root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 2c:cf:67:01:05:30 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::2ecf:67ff:fe01:530/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
3: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr f69b:8300:f1f8::
4: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 8e92:8a48:b771::
9: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
10: phy0-sta0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN group default qlen 1000
    link/ether 2c:cf:67:01:05:31 brd ff:ff:ff:ff:ff:ff
11: eth0.100@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 2c:cf:67:01:05:30 brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.3/29 brd 192.168.4.7 scope global eth0.100
       valid_lft forever preferred_lft forever
    inet6 fe80::2ecf:67ff:fe01:530/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever

root@OpenMPTCProuter:~# ip r
default via 192.168.4.1 dev eth0.100 metric 9 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.4.0/29 dev eth0.100 scope link metric 9 

root@OpenMPTCProuter:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x414fd0b1]
[    0.000000] Linux version 6.6.50 (runner@fv-az1022-312) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0+27346-c7ba5574f5) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Sep 10 22:37:34 2024
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 5 Model B Rev 1.0
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000001800000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x0000000001800000..0x00000000057fffff (65536 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x0000000000000000..0x000000000007ffff (512 KiB) nomap non-reusable atf@0
[    0.000000] OF: reserved mem: 0x000000003fd16320..0x000000003fd16357 (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000007ffff]
[    0.000000]   node   0: [mem 0x0000000000080000-0x000000003fbfffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] On node 0, zone DMA: 1024 pages in unavailable ranges
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 28 pages/cpu s76392 r8192 d30104 u114688
[    0.000000] pcpu-alloc: s76392 r8192 d30104 u114688 alloc=28*4096
[... 572 lines cut ...]
[ 3156.509273] hwmon hwmon0: Undervoltage detected!
[ 3158.589270] hwmon hwmon0: Voltage normalised
[ 3160.669262] hwmon hwmon0: Undervoltage detected!
[ 3162.749265] hwmon hwmon0: Voltage normalised
[ 3164.829286] hwmon hwmon0: Undervoltage detected!
[ 3200.189273] hwmon hwmon0: Voltage normalised
[ 3202.272747] hwmon hwmon0: Undervoltage detected!
[ 3214.749271] hwmon hwmon0: Voltage normalised
[ 3216.829273] hwmon hwmon0: Undervoltage detected!
[ 3256.349278] hwmon hwmon0: Voltage normalised
[ 3260.509268] hwmon hwmon0: Undervoltage detected!
[ 3264.669276] hwmon hwmon0: Voltage normalised
[ 3266.749278] hwmon hwmon0: Undervoltage detected!
[ 3270.909272] hwmon hwmon0: Voltage normalised
[ 3272.989274] hwmon hwmon0: Undervoltage detected!
[ 3289.629276] hwmon hwmon0: Voltage normalised
[ 3291.709278] hwmon hwmon0: Undervoltage detected!
[ 3310.429279] hwmon hwmon0: Voltage normalised
[ 3312.509265] hwmon hwmon0: Undervoltage detected!
[ 3314.589267] hwmon hwmon0: Voltage normalised
[ 3316.679270] hwmon hwmon0: Undervoltage detected!
[ 3318.749271] hwmon hwmon0: Voltage normalised
[ 3322.909277] hwmon hwmon0: Undervoltage detected!
[ 3324.429397] macb 1f00100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
`

Ysurac, 2024-12-31:
ip r and ip a is normal.
The undervoltage alert in dmesg is a problem. Check power supply, remove USB key or use them with a powered USB hub.

Then check, via SSH on the router, with ```ping -B -I eth0.100 192.168.4.1``` that it's working or not.

kolchagov, 2024-12-31:
It's not power related - the board is stable.
Ignore these messages

kolchagov, 2024-12-31:
The ping is not working - that;s why I started the bug request. 

```
root@OpenMPTCProuter:~# ping -B -I eth0.100 192.168.4.1
PING 192.168.4.1 (192.168.4.1) from 192.168.4.3 eth0.100: 56(84) bytes of data.

^C
--- 192.168.4.1 ping statistics ---
47 packets transmitted, 0 received, 100% packet loss, time 47333ms
pipe 4

```

kolchagov, 2025-01-01:
I've managed to use an external USB LAN and it works, so I'm closing this thread

closed 2025-01-01 by kolchagov