Ysurac/openmptcprouter #3733: Can't contact Server Admin Script (No token yet available)

0midkr, 2025-01-01:
## Expected Behavior
Can't contact Server Admin Script (No token yet available)

everthing is ok but OMR client can not cantact with server admin script, i open FW ports but problem persist

## Current Behavior

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.61-6.6 
  - OpenMPTCProuter VPS version: Ubuntu 20.04 
  - OpenMPTCProuter VPS provider: Abrarvan Cloud - Iran Provider
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-01:
What is the result of ```journalctl -u omr-admin``` on the VPS ?

0midkr, 2025-01-02:
hello
The following command that you mentioned has the following output.

-- Logs begin at Wed 2025-01-01 13:23:04 UTC, end at Thu 2025-01-02 08:16:19 UTC. --
Jan 01 13:23:04 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:04 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at >
Jan 01 13:23:04 ubuntu-eco-small1-bamdad-1 systemd[1]: Stopped OMR-Admin.
Jan 01 13:23:04 ubuntu-eco-small1-bamdad-1 systemd[1]: Started OMR-Admin.
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 omr-admin.py[63637]: ERROR:    [Errno -9] Address family for hostname not su>
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FA>
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at >
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 systemd[1]: Stopped OMR-Admin.
Jan 01 13:23:05 ubuntu-eco-small1-bamdad-1 systemd[1]: Started OMR-Admin.
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 omr-admin.py[63639]: ERROR:    [Errno -9] Address family for hostname not su>
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FA>
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at >
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: Stopped OMR-Admin.
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: Started OMR-Admin.
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 omr-admin.py[63641]: ERROR:    [Errno -9] Address family for hostname not su>
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FA>
Jan 01 13:23:06 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:07 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at >
Jan 01 13:23:07 ubuntu-eco-small1-bamdad-1 systemd[1]: Stopped OMR-Admin.
Jan 01 13:23:07 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Start request repeated too quickly.
Jan 01 13:23:07 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:07 ubuntu-eco-small1-bamdad-1 systemd[1]: Failed to start OMR-Admin.
Jan 01 13:23:13 ubuntu-eco-small1-bamdad-1 systemd[1]: Started OMR-Admin.
Jan 01 13:23:14 ubuntu-eco-small1-bamdad-1 omr-admin.py[63973]: ERROR:    [Errno -9] Address family for hostname not su>
Jan 01 13:23:14 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Main process exited, code=exited, status=1/FA>
Jan 01 13:23:14 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Failed with result 'exit-code'.
Jan 01 13:23:15 ubuntu-eco-small1-bamdad-1 systemd[1]: omr-admin.service: Scheduled restart job, restart counter is at >
Jan 01 13:23:15 ubuntu-eco-small1-bamdad-1 systemd[1]: Stopped OMR-Admin.

Ysurac, 2025-01-02:
I think I solved that in develop branch. Can you try ```wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh``` ?

0midkr, 2025-01-02:
OK, no problem and I will do it as soon as possible and let you know

0midkr, 2025-01-02:
To start, what should be the version of Debian in VPS?

Ysurac, 2025-01-02:
Debian 12, but script will update to Debian to 12 from at least Debian 9

0midkr, 2025-01-02:
Ok I started with Debian 12

0midkr, 2025-01-02:
uname -a
Linux debian-eco-small1-bamdad-1 6.6.36-x64v3-xanmod1 #0~20240628.g36640c1 SMP PREEMPT_DYNAMIC Fri Jun 28 11:42:59 UTC x86_64 GNU/Linux

0midkr, 2025-01-02:
Thank you very much, my problem is solved
Other problems that I had before were temporarily fixed with this address https://github.com/Ysurac/openmptcprouter/issues/764, they were also solved in this update and even the download speed has increased a lot.

closed 2025-01-11 by 0midkr