Ysurac/openmptcprouter #3734: multipath master is set on two intefaces resulting in error.

femrichem, 2025-01-02:
## Expected Behavior

The interface with the lowest ping should be set as multipath master

[image]
there should only be one multipath master. 
## Current Behavior
[image]
it doesn't seem to affect the connection but I am not sure. decreased speed is observed when it occurs.

## Possible Solution

## Steps to Reproduce the Problem

  1. set the multipath to dynamic in advanced settings tab
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: [r v0.61-6.6 r0+27346-c7ba5574f5](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2025-01-02:
I will remove Master interface selection as only balancing mode is now tested and should work in all cases.

femrichem, 2025-01-02:
so dynamic selection is deprecated ?

Ysurac, 2025-01-02:
Yes, since now. It will not be available in next release as there is no real uses cases.
You need to select a master interface as it's in fact the default gateway (so initial connections use it). If the master interface is not available, another interface is used.
The master interface, in balancing mode, have a higher priority and is used as default output by OMR-ByPass when no interface are selected.

closed 2025-01-13 by femrichem