Ysurac/openmptcprouter #3738: Plex is indirectly streaming and also not using aggregated upload speed

BATSO123, 2025-01-04:
## Expected Behavior
Remote users should be able to directly connect to plex server and should have aggregated upload speed from both WAN's

## Current Behavior
Plex shows an indirect streaming notification which suggests that port isn't forwarded properly. Also when the indirect connection isn't an issue the upload speed is not aggregated.
[image]

## Specifications

  - OpenMPTCProuter version: 0.62-snapshot 6.6
  - OpenMPTCProuter VPS version: 0.1031.6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Binary Lane
  - OpenMPTCProuter platform: x86_64

Current port forwarding setup
[image]

BATSO123, 2025-01-05:
[image]
This is while using remote streams. It seems that the TCP streaming is only getting used by the master WAN

sieade245, 2025-01-05:
I've had similar issues. In my case, not only could I not get secure connection. I couldn't get ANY connection on the LAN to the plex server sometimes. Worse, when it did work (insecurely), even though I was on the LAN, all the plex video traffic would go external out onto the internet and back via the Plex relay service which caps out at 2Mb/s. I've played around with the settings and not sure exactly which one sorts it out, here's a list:

1. I signed out of the plex app on my TV and added the plex server manually. This allowed plex to work as it bypasses all of the nonsense Plex have added in the past 10 years. You can rock Plex like its 2014 again.

2. I turned off DNS rebind protection in NETWORK>>DHCP AND DNS>>GENERAL SETTINGS. The plex support guide hints than rebind protection can cause the issues where it goes out onto the internet instead of staying in the LAN. I tried some troubleshooting but no matter what DNS settings I added, I couldn't get Plex to work with rebind protection left on.

4. I gave the plex server a static IP address in NETWORK>>DHCP AND DNS>>STATIC LEASES

5. I used the Plex interface remote access configuration page  (server>>settings>>remote access) to manually specify an external port for port forwarding. UPNP doesn't seem to work properly at all in OMR. I used port 24080. 

6. I used the Plex interface network configuration page  (server>>settings>>network) to manually specify an ip range of addresses and networks that are allowed without auth (this was mainly so No 1. above would work but I left it on). My OMR uses 192.168.12.1 so in this box I put 192.168.12.0/24

7. In NETWORK>>FIREWALL>>PORT FORWARDS, I added 2 port forwards. 
1. Incoming IPV4 from WAN to "this device" on Port 24080, forward to the static IP of the plex server on port 32400.
2. Incoming IPV5 from VPN to "this device" on Port 24080, forward to the static IP of the plex server on port 32400.

I added both the WAN and VPN as some of my other port forwards need this (for example StorJ) to work properly. Not sure if plex needs the VPN one

8. NETWORK>>FIREWALL>>TRAFFIC RULES, I added 2 rules for plex
1. Incoming IPV4 TCP/ UDP from WAN to LAN on Port 32400, accept forward.
2. Incoming IPV4 TCP/ UDP from VPN yo LAN on Port 32400, accept forward.

Again, unsure if the VPN rule is needed. Now I have it working I dare not touch any of the settings as I've noticed there are several bugs with the firewall settings.

For example, if you have multiple VPS in the wizard, the updated firewall settings seem to only be sent to the VPS top of the list, even if this isn't your current VPS in use. After I discovered, this I removed all my secondary VPSs and just change them manually now in the wizard.

Next issue, if you make a firewall change, the settings don't seem to apply, even after a OMR reboot, unless you also make some kind of significant change in the wizard. I've been changing my proxy from Xray to V2ray and back again. This seems to do the trick but even this doesn't also work. This also affects OMR bypass too in the same way. The reason I mention this is this was the main problem which stopped plex working for me. I thought I had it correct in the firewall, but it didn't work so I made change after change after change before realising that none of the changes were actually being implemented by OMR server due to these bugs. Hence the reason I dare not touch it again!!

The good news is that after doing all of this, Plex does work properly with secure connection and be fully accessible outside of the LAN for internet streaming. I don't need to use No.1 above anymore, but to be honest I prefer using it this way as it removes all the Plex bloat they added.

Hope this helps.

andrewmackrodt, 2025-01-11:
> I turned off DNS rebind protection in NETWORK>>DHCP AND DNS>>GENERAL SETTINGS. The plex support guide hints than rebind protection can cause the issues where it goes out onto the internet instead of [...]

@sieade245 I came across this recently too, if you want to get this working:
1. If `Rebind Protection` is enabled under `Network > DHCP and DNS` ensure that `plex.direct` is added to `Domain whitelist`
2. Go to `Services > Recursive DNS > Unbound > Advanced` and set `Filter Private Rebind` to `No Filter`

sieade245, 2025-01-18:
> [@sieade245](https://github.com/sieade245) I came across this recently too, if you want to get this working:
>
> 1. If `Rebind Protection` is enabled under `Network > DHCP and DNS` ensure that [...]

Thanks for this info. I'll give it a shot!

sieade245, 2025-01-19:
@andrewmackrodt You Absolute Legend!. That is working. I spent literally hours on that. Thank you!

closed 2025-04-25 by github-actions[bot]