Ysurac/openmptcprouter #3739: Impossible write on ‘-‘ - VPS SCRIPT

VisionLabGroup, 2025-01-04:
I tried to launch the script to update the VPS but it always returns the error "impossible write on '-'. "

I also tried the snapshot script, the result is the same.

I tried three other VPS and several scripts, always the same result.

Can you help me?

Ysurac, 2025-01-04:
As in the doc (https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS):
```
wget https://www.openmptcprouter.com/server/debian-x86_64.sh
KERNEL="6.6" sh debian-x86_64.sh
```

VisionLabGroup, 2025-01-05:
Yes I used that script, and I have the error

Ysurac, 2025-01-05:
Can you take a screenshot ?

VisionLabGroup, 2025-01-05:
[image]

AndreaLeidi, 2025-01-05:
Why you add "-O -" to the command? Try without.

VisionLabGroup, 2025-01-05:
I don t know. I copied it directly from website.
I try without it

closed 2025-04-10 by github-actions[bot]