Ysurac/openmptcprouter #3740: How to get wanX interface WAN IP address in OMR?

h45h3ncryp73d, 2025-01-04:
I guess this should be quite easy and since this is just a quick info I'll be omitting all the OMR info.

Is there a command to get wanX (i.e. wan2) WAN/external IP address assigned to designated router like shown on infographic System -> OpenMPTCProuter -> Status -> wan ip: from OMR terminal? I could do a little code digging but I guess I'm to lazy for that so I decided to ask politely instead ;-)
Need to update dyndns with just one router ip address but unfortunately I cannot get it from my router not it does have dyndns update feature so I need to do it via cron job.

Thank you in advance.

vempire-ghost, 2025-01-05:
You can execute this command via SSH `curl --interface eth1 "http://ip-api.com/json/"` and filter the result to see only the IP or any other data you want. You can use another geolocation provider instead of ip-api.com if you prefer.

h45h3ncryp73d, 2025-01-05:
Thank you but unfortunately this does not seem to be working with MAC VLAN setup - curl --interface wan2 https://ipinfo.io/ip - does not return anything and curl --interface <IP_ADDRESS> https://ipinfo.io/ip returns VPS IP address.

Ysurac, 2025-01-06:
You can use ```omr-ip-intf <interface>```, where <interface> is the real interface that you can have via ```ip a```

h45h3ncryp73d, 2025-01-06:
Thank you! It is working like a charm!

closed 2025-01-06 by h45h3ncryp73d