Ysurac/openmptcprouter #3741: How to add the rtw88 kernel driver to the 5.4 kernel

fdyf520, 2025-01-05:
How to add the rtw88 kernel driver to the 5.4 kernel
When compiling version 0.58.5, I selected rtw88 and 8822ce, but after the compilation, the Wi-Fi could not be used normally.

AndreaLeidi, 2025-01-05:
Before baking your own kernel, did you try to search in system > software for the following packages?

kmod-rtw88

plus one of the three:

kmod-rtw88-pci
kmod-rtw88-usb
kmod-rtw88-sdio

(depending on your hardware: PCI card, USB dongle or SDIO Secure Digital form factor card)

Install, reboot and check if it works.

fdyf520, 2025-01-05:
> Did you try to search in system > software for the following packages?
> 
> kmod-rtw88
> 
> plus one of the three:
> 
> kmod-rtw88-pci kmod-rtw88-usb kmod-rtw88-sdio
> 
> (depending on your hardware: PCI card, USB dongle or SDIO Secure Digital form factor card)
> 
> Install, reboot and check if it works.

The channel cannot be found after compilation.

fdyf520, 2025-01-05:
The channel cannot be found after compilation.

darkman1983, 2025-01-05:
Kernel 5.4 is EOL, don't use it anymore!

vempire-ghost, 2025-01-05:
> Kernel 5.4 is EOL, don't use it anymore!

[image]
Its not.

Ysurac, 2025-01-05:
OpenMPTCProuter based on kernel 5.4 is EOL as MPTCP patch for kernel 5.4 is EOL.

closed 2025-04-10 by github-actions[bot]