Ysurac/openmptcprouter #3742: How To use Seperate Ethernet ports.

rustynail1984, 2025-01-06:
Hello guys, i've a question. 

Is it possible to use for each Connection a seperate Ethernet port? Ive have a Barebone with 6x 2,5G Ethernet Port.

If yes, how to do it?

## Expected Behavior
Want to use seperate Ethernet Port for each connection.
- 1 DSL in/out
- 1 Mobile 5G in/out
- LAN only for Bounded DSL/5G Mobile internet.

## Specifications
  - OpenMPTCProuter version: [v0.61-6.6 r0+27346-c7ba5574f5](https://github.com/ysurac/openmptcprouter)

Ysurac, 2025-01-06:
Yes sure. You can set them via Network->Interfaces or via System->OpenMPTCProuter, Wizard tab.
It's possible that all ports are not detected, in this case a driver may be needed (and can be installed via System->Software)

rustynail1984, 2025-01-06:
> Yes sure. You can set them via Network->Interfaces or via System->OpenMPTCProuter, Wizard tab. It's possible that all ports are not detected, in this case a driver may be needed (and can be installed via System->Software)

Ok thanks. i think i got it.

closed 2025-01-06 by rustynail1984