Ysurac/openmptcprouter #3748: Aggregation and failover doens't seem to work

0midkr, 2025-01-08:
## Expected Behavior

Omr is expected to work

## Current Behavior

The configurations are done based on the guides, and everything is set correctly inside the omr. I took a few screenshots later, but the result seems to be as expected, but it doesn't work

[image]

[image]

## Specifications

  - OpenMPTCProuter version:  v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 0.1032-test 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: arvanvps
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-08:
What is not working ?

0midkr, 2025-01-08:
I feel this way, now you can help me how to prove and test, everything works and there is no problem

0midkr, 2025-01-08:
This also happened to one of wan's
[image]

0midkr, 2025-01-08:
I think that no traffic passes over wan2

Ysurac, 2025-01-08:
Via SSH on the router, test with ```omr-test-speed wan1```, ```omr-test-speed wan2``` and ```omr-test-speed``` (Ctrl+C to stop after at least 1 minute). Check also, during last command, what you have in Network->MPTCP, "Bandwidth" tab.
Disable SQM during test in wizard (wan2 download/upload speed set seems to be really low).
I don't see any VPN speed, you disabled it ?

0midkr, 2025-01-08:
I really thank you for your efforts. Unfortunately, I don't have access to my PC right now, tomorrow I will do the solution you mentioned at the first opportunity. And I have a problem with the vpn connection that you mentioned, which I will raise tomorrow, thank you very much.

0midkr, 2025-01-09:
Select best test server...
.
.
.

Best server is http://speedtest.frankfurt.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  141M    0  141M    0     0   939k      0 --:--:--  0:02:34 --:--:--  887k

root@OpenMPTCProuter:~#

[image]
[image]

0midkr, 2025-01-09:
omr-test-speed wan2
Select best test server...
.
.
.

Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 9536M    1  102M    0     0   560k      0  4:50:28  0:03:06  4:47:22  585k^C

root@OpenMPTCProuter:~#

[image]
[image]
[image]

0midkr, 2025-01-09:
omr-test-speed
Select best test server...
.
.
.

Best server is http://fra.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 9536M    2  268M    0     0  1564k      0  1:44:00  0:02:55  1:41:05  995k^C

root@OpenMPTCProuter:~#

[image]
[image]
[image]
[image]

0midkr, 2025-01-09:
Hello, I did not disable the VPN connection during config OMR, It was disabled by default and when I try to activate the VPN, I get the following error.

[image]

0midkr, 2025-01-09:
or somtimes get this error

[image]

Ysurac, 2025-01-09:
Don't use OpenVPN Bonding. Keep the default OpenVPN TCP if possible or Glorytun TCP when OpenVPN is blocked.
You have no aggregation at all.
What is the proxy used ?
What do you have in Network->MPTCP, "MPTCP fullmesh" tab ?

0midkr, 2025-01-09:
The proxy I use is shadowsock

[image]

0midkr, 2025-01-09:
the fullmesh tab.

[image]

0midkr, 2025-01-09:
After setting openvpn tcp, the problem was solved and there is no error, but the vpn traffic is very low

[image]

0midkr, 2025-01-09:
> Use Shadowsocks-Rust as proxy. It's the default.

OK, thanks

0midkr, 2025-01-09:
VPN traffic is still low

[image]

darkman1983, 2025-01-09:
And another tip:
Check the "Established connections" TAB under Network -> MPTCP
If there is only one connection, if you use Shadowsocks, it's bugged then.
Try XRAY-VLESS then, there should be displayed more then one connection.

0midkr, 2025-01-09:
Thanks buddy, I'll definitely try it.

0midkr, 2025-01-09:
> And another tip: Check the "Established connections" TAB under Network -> MPTCP If there is only one connection, if you use Shadowsocks, it's bugged then. Try XRAY-VLESS then, there should be displayed more then one connection.

Thanks buddy, I'll definitely try it.

closed 2025-01-11 by 0midkr