Ysurac/openmptcprouter #3750: IPv6 as Path, even if it's on the same interface as IPv4 path?

darkman1983, 2025-01-10:
## Expected Behavior
Shouldn't there only one Path be used if it is the same Interface?
Or should each interface have all available Paths?
Why should an interface have both path's?
I thought each route is per interface, if one has two routes, there is no benefit?!

## Current Behavior
I've enabled IPv6 and MPTCP added WAN1 as a new Path for Subflows.
So i've got two Paths for eht1, Ipv4 and IPv6.
But WAN2 has only the IPv4 Path.

## Specifications

  - OpenMPTCProuter version: Version 0.62-snapshot-6.12
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.5-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64

[image]
[image]

Ysurac, 2025-01-10:
All IPs associated with an interface are added.
Wan2 have a Link-local IP, it's why it's not added. It's also strange to have a fd00:* IP on wan1 interface.

darkman1983, 2025-01-10:
I had only an ULA adress for the fritzbox, had to get an fe80 with "ip -6 neigh", because in Webinterface of router it's not displayed xD