Ysurac/openmptcprouter #3751: APK in Snapshot 0.62 doesn't work

darkman1983, 2025-01-10:
## Expected Behavior
The apk command finishes successfully.

## Current Behavior
The apk command gives errors on update.

[image]

## Possible Solution
You tell me.

## Steps to Reproduce the Problem

  1. Login to console
  2. Use apk command: apk update

## Context (Environment)
Can't use apk command, because it can't update.

## Specifications

  - OpenMPTCProuter version: Version 0.62-snapshot-6.12
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.5-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-10:
For now I didn't create apk packages repos.

darkman1983, 2025-01-10:
> For now I didn't create apk packages repos.

😭

closed 2025-04-15 by github-actions[bot]