Ysurac/openmptcprouter #3752: Saving the OpenMPTCPRouter settings result in webinterface error.

darkman1983, 2025-01-10:
## Expected Behavior
I can save the settings and get no error.

## Current Behavior
On saving the settings for OMR under wizard tab, i get an error.
[image]

## Possible Solution
You tell me.

## Steps to Reproduce the Problem

  1. Go to wizard and change settings
  2. Save them

## Context (Environment)
It seems to be only a glitch and no real show stopper, my settings get saved.

## Specifications

  - OpenMPTCProuter version: Version 0.62-snapshot-6.12
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.5-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-10:
It's due to a typo, this will be fixed in next snapshot. Thanks.

darkman1983, 2025-01-11:
The next Snapshot was "January 10", it's not fixed :-D

darkman1983, 2025-01-15:
Saving settings with latest snapshot worked now.

closed 2025-01-17 by darkman1983