Ysurac/openmptcprouter #3753: Systemlog is empty / not loading

darkman1983, 2025-01-10:
## Expected Behavior
If i open the Systemlog in Webinterface, it is displayed

## Current Behavior
Nothing is shown
[image]

## Possible Solution
You tell me

## Steps to Reproduce the Problem

  1. Go to Status -> Systemprotocol

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: Version 0.62-snapshot-6.12
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.5-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-10:
I'm not able to reproduce this issue on my local compilation, I will try with current snapshot.

darkman1983, 2025-01-10:
Yeah, i'm on current Snapshot from 8 January.

darkman1983, 2025-01-11:
Still present in 10 Januaray. Snapshot.

yanko66, 2025-01-11:
Тhe same. 
v0.62-snapshot-6.12 r0+28523-285bdd29cb

darkman1983, 2025-01-12:
In "/etc/config/system":
`config system
        option hostname 'OpenMPTCProuter'
        option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option ttylogin '1'
        option log_size '64'
        option urandom_seed '0'
        option cronloglevel '9'
        option conloglevel '7'
        option zonename 'Europe/Berlin'
        option log_proto 'udp'
        option zram_comp_algo 'lzo'

config timeserver 'ntp'
        option enable_server '1'
        option use_dhcp '0'
        option enabled '1'
        list server '0.pool.ntp.org'
        list server '1.pool.ntp.org'
        list server '2.pool.ntp.org'
        list server '3.pool.ntp.org'

config rngd
        option device '/dev/urandom'
        option enabled '1'`

Shouldn't there be entries like:
option log_file '/tmp/log/messages'
option log_level '7'

The only entry i see is for "option log_size '64'".
The command logread only gives this one line, like in webinterface shown.
`Jan 12 01:37:59 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.37.0`

The path "/var/log/messages" is empty too.
So where should the messages get logged?

Ysurac, 2025-01-12:
Do you have any output via ```logread``` command ?
For https://multipath-tcp.org/ is normal, it's MPTCPv0 that was used in kernel 5.4 by default, now MPTCPv1 is used.

darkman1983, 2025-01-12:
As i already told you: Output for logread is: Jan 12 01:37:59 OpenMPTCProuter syslog.info syslogd started: BusyBox v1.37.0

darkman1983, 2025-01-14:
Still waiting for new Snapshot :-D

darkman1983, 2025-01-14:
OK, installed Snapshot from Januaray 14, Syslog still not working, still empty log.

darkman1983, 2025-01-15:
With the Snapshot mentioned my OMR stopped working at all, ports not forwarded correctly and all other things also doesn't work. I also tried with kernel 6.6 of Snapshot 0.62, but nothing is working anymore. A complete fresh install also had no effect, still nothing works anymore with latest snapshot.

And as is need working internet, im back to stable 0.61 with 6.6 kernel.

yanko66, 2025-01-23:
solution:
I replaced the busybox v.1.37.0-r4 binary with busybox 1.36.1-r2 and everything works now.
The problem may be in busybox-1.37.0-r4.

closed 2025-04-28 by github-actions[bot]