Ysurac/openmptcprouter #3756: some things about master test

fareign, 2025-01-11:
Environment
three interfaces. master mode balance.proxy:v2ray

if I move master card, the vps must get lost,and proxy v2ray get down. but the other two cards also can work. but the public IP can't get response,and a long long time ,the vps can reconnect.
when I pull back master card, VPS connect. v2ray reload. the network must reload. and a few seconds broken.

so when master not work, another interface  quickly get link for VPS. and no need reload proxy?

if master mode set dynamic.
 the master card can change,but the link vps interface also is the original。

## Specifications

  - OpenMPTCProuter version: 0.61
  - OpenMPTCProuter VPS version: 0.1031
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: rpi4
  - Country:

ccmks, 2025-01-11:
Are you implying that if Master WAN is down, no internet? Despite of the other WAN are working?

fareign, 2025-01-11:
> Are you implying that if Master WAN is down, no internet? Despite of the other WAN are working?

YES, I moved all of the ant. when master is down, the VPS get down.

fareign, 2025-01-11:
sometimes   the VPS connect is not the master .

mrkaffeine92, 2025-02-09:
I think this has to do with the one going down on you - being set as master? Try flipping it to enabled and the more "trustworthy" one as master?

closed 2025-05-15 by github-actions[bot]