Ysurac/openmptcprouter #3759: Pull the latest source code and compile the system. The ath9k driver uses the ar9382, but the wireless option does not have a channel

fdyf520, 2025-01-11:
Pull the latest source code and compile the system. The ath9k driver uses the ar9382, but the wireless option does not have a channel
[image]
[image]

fdyf520, 2025-01-11:
[image]
This is what normal looks like

Ysurac, 2025-01-12:
This seems to be an old release/code, kernel 5.4 is not supported anymore. I'm not sure the issue is still here on develop branch.

fdyf520, 2025-01-13:
The latest version of the system still has problems

closed 2025-04-18 by github-actions[bot]