Ysurac/openmptcprouter #3760: "Multipath seems to be blocked on the connection" is back random

darkman1983, 2025-01-11:
## Expected Behavior
The above message is not shown.

## Current Behavior
The message is totally random displayed, sometimes for one WAN, sometimes for both, sometimes not at all.
With the stable release the message wasn't displayed at all.

[image]

MPTCP has connections:
[image]

## Possible Solution
You tell me.
Between Kernel 6.6 to 6.12 there is a new mecanism added in MPTCP, wich detects it self, if a Path has MPTCP blocked and then avoids this path. I dunno if you could get that value from MPTCP, if yes, it could replace the actual detection.

## Steps to Reproduce the Problem

  1. Just installed Snapshot build

## Context (Environment)
MPTCP seems to be working normal, so the detection on mptcprouter might be broken or is still not functioning.
Sometimes if i restart OMR, the message isn't displayed. Sometimes it appears after some use time.

## Specifications

  - OpenMPTCProuter version: Version 0.62-snapshot-6.12
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.5-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: HT-Hosting
  - OpenMPTCProuter platform: x86_64
  - Country: DE

darkman1983, 2025-01-12:
After reboot:
[image]

darkman1983, 2025-01-12:
Now it's all green, it's after a few hours running. Seems the detection is buggy?!
[image]

darkman1983, 2025-01-13:
And back to Blocked again...

darkman1983, 2025-01-13:
Does this block anything? I can't get MPTCP to work correctly on bith WAN's, maybe the "Save Settings" bug is influencing here.

Ysurac, 2025-01-13:
The message doesn't change anything, it's only a test.
Kernel 6.12 based snapshot is still not tested at all.

lhc1993, 2025-01-15:
how to close this test？

lhc1993, 2025-01-15:
let display keep green @Ysurac

closed 2025-06-19 by github-actions[bot]