Ysurac/openmptcprouter #3761: When connecting to USB portable WiFi, device registration failed

lhc1993, 2025-01-12:
When connecting to a portable wifi, the device can be seen through lsusb, but when connecting, wwan2 will be created first, then logged out, and then usb0 will be created and then logged out, and no device will be found after that.

[  827.954147] usb 3-1.5: new high-speed USB device number 9 using xhci-hcd
[  828.205130] qmi_wwan 3-1.5:1.2: cdc-wdm2: USB WDM device
[  828.206129] qmi_wwan 3-1.5:1.2 wwan2: register 'qmi_wwan' at usb-xhci-hcd.2.auto-1.5, WWAN/QMI device, 86:b7:40:41:94:ba
[  861.964699] usb 3-1.5: USB disconnect, device number 9
[  861.965749] qmi_wwan 3-1.5:1.2 wwan2: unregister 'qmi_wwan' usb-xhci-hcd.2.auto-1.5, WWAN/QMI device
[  862.325285] usb 3-1.5: new high-speed USB device number 10 using xhci-hcd
[  862.513116] rndis_host 3-1.5:1.0 usb0: register 'rndis_host' at usb-xhci-hcd.2.auto-1.5, RNDIS device, 02:0c:50:01:33:64
[  862.519216] rndis_host 3-1.5:1.0 usb0: unregister 'rndis_host' usb-xhci-hcd.2.auto-1.5, RNDIS device

Ysurac, 2025-01-12:
Please use template.
The dive seems to change mode from QMI to RNDIS. What is exactly the device and output of lsusb ?

lhc1993, 2025-01-15:
i have found reason，this issue close，thanks @Ysurac

closed 2025-01-15 by lhc1993