Ysurac/openmptcprouter #3762: segfault error 4 in libcrypto.so (likeley on CPU 1)

misterbond007, 2025-01-12:
## Expected Behavior

No erros when logging in webGUI.

## Current Behavior

Error seen on terminal when trying to logging into the webGUI: uhttpd 7830: segfault at xxxxxx ip 00007f36xxxxx sp 00000xxxxx error 4 in librypto.so.3 xxxxxxxx likely on CPU 1 (cocre1, socket 0) Code: 00 00 31 xx xx xx etc

Fresh install as a virtual machine in DSM 7.2 on my Synology NAS DS720+ (Intel Celeron J4125).

## Specifications

  - OpenMPTCProuter version: 0.61.6.6
  - OpenMPTCProuter VPS version: Kernel 6.6
  - OpenMPTCProuter VPS provider: Netcup
  - OpenMPTCProuter platform: Debian 11

Ysurac, 2025-01-13:
Can you try latest snapshot ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

misterbond007, 2025-01-13:
At https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/ I don't see any ova (Open Virtual Appliance ext4 image (EFI)) downloads.

Ysurac, 2025-01-13:
The OVA is generated by a script from the VMDK only for releases. You should be able to directly replace the VMDK.

Ysurac, 2025-01-13:
yes

misterbond007, 2025-01-13:
I have replaced the VMDK file with the latest snapshot VMDK file. I have also installed the developer server script on the VPS.
I can now login to OMR router locally without errors. When I set the options and then open the status tab, the page is loading infinite.
[image]

Ysurac, 2025-01-13:
Strange, you should have something like "dhcp-range=set:lan,192.168.100.1,192.168.1.249,255.255.255.0,12h"
Check IP of LAN interface.

misterbond007, 2025-01-13:
I have changed the line to no-dhcp-interface=eth0
Now I get 
[image]

misterbond007, 2025-01-13:
Will the changes be published in the next official releases?

closed 2025-01-13 by misterbond007

Arachnid, 2025-02-06:
I'm seeing this same issue, with uhttpd segfaulting regularly. I upgraded to the latest snapshot, but the issue persists.

alchool, 2025-02-12:
On a fresh installation I'm facing same issue... Also is very difficult to navigate LuCi's pages and frequently I received "**Error
XHR request aborted by browser**" and on System Log there's a  "_OpenMPTCProuter user.notice omr-schedule-010-services: Can't find uhttpd, restart it..._"
I also tried to upgrade from .58.5 or from .60 version and still facing the same issue. Previous versions was ok for luci but I have to upgrade from .60 cause Network/ModemManager issues

Ysurac, 2025-02-12:
Try latest snapshot.

alchool, 2025-02-14:
So I tried to upgrade from cli to the latest snapshot: 
"sysupgrade openmptcprouter-v0.62-snapshot-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined"
XHR request aborted by browser still appear.
System logs are gone.
DNS configuration gone.
KERNEL LOG: _uhttpd[6824]: segfault at 7f7d8bc29f80 ip 00007f7d8be45620 sp 00007ffd8cb17f18 error 4 in libcrypto.so.3[7f7d8be07000+26a000] likely on CPU 0 (core 0, socket 0)_
I tried also to upgrade snapshot from Luci 0.60. Same result
It looks like to install libustream-wolfssl20201210 seems to fix the problem, after reboot

Ysurac, 2025-02-14:
@alchool What you mean with DNS configuration gone ?
For system log a file is missing, I will add that.
I don't have any segfault, check for hardware problem like with memory or too high CPU temperature.

alchool, 2025-02-21:
I didn't dig too much on that issue but it uses /etc/resolv.conf and as result:
nslookup: write to '127.0.0.1': Connection refused
nslookup: write to '::1': Connection refused
;; connection timed out; no servers could be reached
Connection was up on a Wan, so I just fixed it adding a script to overwrite nameserver 1.1.1.1 every minute

Maverick78de, 2025-03-25:
I have the same issue segfaulting libcrypto.so and sometime crash the whole machine.
I installed it (new installation) in a VM withing proxmox used the vmdk openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined.vmdk

(cpuinfo:
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov                                                                                           pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm c                                                                                          onstant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_t                                                                                          sc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est t                                                                                          m2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand                                                                                           lahf_lm 3dnowprefetch epb pti ibrs ibpb stibp tpr_shadow flexpriority ept vpid                                                                                           tsc_adjust smep erms dtherm ida arat vnmi md_clear
vmx flags       : vnmi preemption_timer invvpid ept_x_only flexpriority tsc_offs                                                                                          et vtpr mtf vapic ept vpid unrestricted_guest
bugs            : cpu_meltdown spectre_v1 spectre_v2 mds msbds_only mmio_unknown
)

I tried different types of virtual CPU for the VM, without any luck. After spending several hours I disabled AES support for the VM and now it is working without error or crash. But, a big but, the throughput over Shadowsocks and other VPN Type is really really poor, I can't get not more then 80Mbps because of high CPU usage. (a little bit more with chacha20 used). My old rpi4 (1st series) is 2 times faster.

So there is an issue with libcrypto.so and AES support in the current stable version for x86_64.

Would be great to get it fixed because I want to relace the rpi4 soon.

Ysurac, 2025-03-25:
This issue is closed.

Arachnid, 2025-03-26:
...yes, but it's not fixed.

Ysurac, 2025-03-26:
Open a new issue. This issue was also far from initial problem, not on current release and closed by author...

Arachnid, 2025-03-26:
It's literally the same error. Opening a new issue would lose the history of the issue. Maintainers have the power to reopen issues.

Ysurac, 2025-03-26:
You seems to not understand that the issue speaked about DHCP so not releted to issue. It was also an install on Synology.
Maintainers have also the right to ask to open a new issue. I don't want to loose my time on sterile discussion, I lock this issue.