Ysurac/openmptcprouter #3764: Download speed is not bonding but upload speed is bonding

lkimloong, 2025-01-13:
## Expected Behavior

Using 3 android mobile as usb tethering to bond 3 WAN. 
## Current Behavior

Single line usb tethering on a direct connection to laptop - average speed for 3 individual line upload 50mbps / download 130mbps
When bonding using MPTCPRouter we get download 10mbps / upload 130mbps
We on both OVHCloud & self build x86_64 VPS - both result are same
Try with speedify using the same usb tethering method on the same simcard the speed is bonding
Is there any setting that we miss out?
## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.61-6.6 
  - OpenMPTCProuter VPS version: Debian 12 Kernel 6.6
  - OpenMPTCProuter VPS provider: OVHCloud & selfbuild VPS on x86_64
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-13:
Can you try latest 6.6 snapshot: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

ovehbe, 2025-01-13:
I have the same issue and I actually tried with the snapshot but the issue persists.

Ysurac, 2025-01-13:
What is the proxy used ? Default is Shadowsocks-Rust.
If it's the case, I would need ```uci show shadowsocks-rust```

ovehbe, 2025-01-13:
Yes its Shadowsocks-Rust, here is the output:

`root@OpenMPTCProuter:~# uci show shadowsocks-rust
shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.keep_alive='15'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='0'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='1'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss1'
shadowsocks-rust.hi2.local_address='::'
shadowsocks-rust.hi2.local_port='1101'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.keep_alive='15'
shadowsocks-rust.hi2.fast_open='0'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='0'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='1'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.redir_tcp='hi1'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.ss_rules.redir_udp='hi2'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='0'
shadowsocks-rust.sss0.server='154.53.180.201'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss0.password='FhsGiBqTnKf2V13MtXodZJbKuaZkT0d0zvSzZp274dg=:NImZLeMXOjyx/4yKd74sWOBH7raB5HOmA/rZKr3cpHw='
shadowsocks-rust.sss1=server
shadowsocks-rust.sss1.disabled='1'
shadowsocks-rust.sss1.server='192.168.1.3'
shadowsocks-rust.sss1.server_port='65280'
shadowsocks-rust.sss1.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss1.obfs='0'
shadowsocks-rust.sss1.password='FhsGiBqTnKf2V13MtXodZJbKuaZkT0d0zvSzZp274dg=:NImZLeMXOjyx/4yKd74sWOBH7raB5HOmA/rZKr3cpHw='
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.forward_address='8.8.8.8'
shadowsocks-rust.dns.forward_port='53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.keep_alive='15'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'
`

ovehbe, 2025-01-13:
Yes, omr-test-speed gives average of 20m in speed test btw I get 55download (normally wan1 is 100mbps and wan2 is 35mbps).

Note: Checked and verified vps limits Download: 200mbps, Upload: 50mbps.

omr-test-speed output:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 14 9536M   14 1398M    0     0  20.3M      0  0:07:47  0:01:08  0:06:39 18.4M
```

Network->MPTCP, "Bandwidth" tab, "all":
[image]

Speedtest resault:
[image]

Ysurac, 2025-01-13:
omr-test-speed give 20.3MB/s, so 162.4Mb/s. Not so bad. It was run on the router or VPS ? If it was on VPS, run it on the router.

ovehbe, 2025-01-13:
omr-test-speed (router):

```
Best server is http://prg.download.datapacket.com/10000mb.bin, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 9536M    4  463M    0     0  7231k      0  0:22:30  0:01:05  0:21:25 8899k
```

Ysurac, 2025-01-13:
Can you run same test and check MPTCP->Bandwidth tab ?
Then can you in System->OpenMPTCProuter, "Wizard" tab, "Advanced settings" checkbox, set "encryption" to "chacha20" save and test again omr-test-speed on the router ? (RPI4 doesn't have hardware AES support)

ovehbe, 2025-01-13:
Also if this is a dead end because of hardware limitations of the raspi4, what do you recommend as good hardware for openmptcprouter?

Ysurac, 2025-01-13:
A RPI4 should be able to deliver about 400Mb/s. A RPI5 can give up to 980Mb/s (limited by the ethernet 1Gb/s port).
Can you give me a screenshot of System->OpenMPTCProuter, Status page ?

ovehbe, 2025-01-13:
I also noticed that wan2 doesnt establish connection:

```
ESTAB 0       0      192.168.1.100:21205 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21215 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49129 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:59737 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49273 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49173 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:64217 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:59767 154.53.180.201:65248
ESTAB 9156160 0      192.168.1.100:23921 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49255 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:23869 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49245 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49127 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49233 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:59739 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21137 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:48993 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21103 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49125 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49143 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49191 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:23935 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49169 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:64303 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21055 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21147 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:59717 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21227 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49229 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:64325 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49177 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:23867 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:48981 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49109 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49261 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49203 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49299 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:49287 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21197 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:21155 154.53.180.201:65248
ESTAB 0       0      192.168.1.100:64315 154.53.180.201:65248
```

Current Config: XRAY/VLESS + Glorytun UDP

Ysurac, 2025-01-13:
The VPN is not related to TCP speed, VPN is only used for anything that is not TCP (UDP, ICMP,...) and the proxy for TCP.
There is a high load on VPS... Really too high for something that is doing near nothing...

ovehbe, 2025-01-13:
Hmm, the specs of the server arent that bad though. Can we debug the reason? If its something like vps specs I'll try with another vps.

Ysurac, 2025-01-13:
On the VPS: ```apt-get install htop``` and run ```htop``` to check usage.

ovehbe, 2025-01-13:
Yeah acutally the server always peaks in cpu, the problem isnt specific as xray was using too much first then I changed it then anything honestly peaks it. So now I upgrade the cores from 2 to 8 and it looks like it working now but still wan2 is not bonding though.

ovehbe, 2025-01-15:
Ok so the speed isnt getting any better like this, I think I'll try another vps but do you have any reccommendations for the server specs?

Ysurac, 2025-01-15:
Not really, but a big problem is with some VPS using shared CPU that is too much shared.
1vCPU and 1024MB RAM is enough in most case (it's what I have on Vultr, Digital Ocean, Scaleway, GTT,...). AWS, GCP and Azure are not always great on low cost VPS.

ovehbe, 2025-01-15:
Ok so, I actually got a server from vultr with no blocked ports and setup the snapshot config and I tested again with each combination and checked omr-test-speed, htop, bandwith and all of the resault are almost exactly the same with the onlu difference in ping which offcourse because of the distance. So my local vps with 8vCPU and new vultr all stats are the same except for cpu which yes like you said my local is really sharing cpu too much but the point is the issue currently is not cpu. And again one thing I notice is the established connection on both are only showing wan1 ip. and I also can tell its not my source wan speeds because I also test speeds by disabling server and one wan interface at a time and the resaults (escpecially shadowsocks (not rust)) when the server is enabled show almost exact speed of the raw wan1. So I really think the wans arent bonding for download.

It would really help if you can continue troubleshouting this with me. Please let me know if I can provide any data on these.

lkimloong, 2025-01-17:
I try updated to the latest version as well and having the same result. It is not really bonding 3 of my line.

Ysurac, 2025-01-17:
@lkimloong can you do same tests : ```omr-test-speed``` via SSH on the router and check at the same time in Network->MPTCP, "Bandwidth" tab what you have ?

mvmaksnet, 2025-02-12:
I have the same problem. Download is 4mbps and upload is good bonded. I use the latest image 6.6 and the latest vps.
In past I didn't have problems like these with older versions.

giuchie, 2025-02-12:
same problem... I tried with 2 identical connections limited to 200mbps each down and 10 up... i can't get more than 180-200mbps down... it seems that the openmptcprouter uses only one wan at most and the second is limited. i tried every combination of settings but nothing... could it be the vps? i'm using aruba

[image]

Ysurac, 2025-02-12:
The VPS need to be able to do more than aggregated speed.
And I still need ```omr-test-speed``` and ```omr-test-speed wan*``` (where wan* is each WAN real interface), for 2 minutes. Stop with Ctrl+C.

In most case, it's router CPU limitation, VPS bandwidth limit or CPU or connection issue (2 4G/5G connection on same provider for example).

You can also test snapshot: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots (I will make a new release quite soon, fixing last issues)

closed 2025-05-19 by github-actions[bot]