Ysurac/openmptcprouter #3766: OMR Failover noting work for me

0midkr, 2025-01-13:
Hi! Thanks for this great project

## Expected Behavior

Both WAN connections should be used to aggreate the speed.
If one connection goes down, the other wan should still be used and provide internet connection 

## Current Behavior

When my first router, which is also the master router, disconnects from the Internet, my aggregated Internet connection is also interrupted, which means that my second modem is no longer able to pass traffic and my access to the Internet is interrupted

## Specifications

  - OpenMPTCProuter version: 0.61 
  - OpenMPTCProuter VPS version:  
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-13:
Try latest router snapshot: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

0midkr, 2025-01-13:
thank very much

0midkr, 2025-01-13:
I can update the system in this way System->Backup/Flash Firmware->Backup->Flash new firmware image

0midkr, 2025-01-15:
What is the Fail2ban feature؟

Ysurac, 2025-01-15:
This install fail2ban with custom configurations for OpenVPN, more will be added later.

closed 2025-01-16 by 0midkr