Ysurac/openmptcprouter #3767: The backup interface does not work when the master interface is down

stt12706, 2025-01-15:
## Expected Behavior

I am using iperf3 as a testing tool on QNAP 301w. First, WAN1 serves as the primary interface responsible for iperf uplink transmission. Then, using the ifdown command to disable WAN1. At this point, the iperf command continued to run without interruption, and WAN2, as the backup interface, take over and continued transmitting the iperf uplink data.
## Current Behavior

When I shut down the master interface WAN1, the upstream TCP traffic of iperf does not get transmitted through the backup interface WAN2 (although it sometimes works successfully).

I'm curious why the switch to the backup interface sometimes succeeds and sometimes doesn't. Is there any configuration I might have overlooked or could adjust? What step can I take to resolve this issue?

Thank you for your help!

## Specifications
  - OpenMPTCProuter version: openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 0.1032-test 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-01-15:
Backup mode is less tested.
To fix this faster, I would need the result of ```ip r``` before and after you disable WAN1.

stt12706, 2025-01-16:
The output of `ip r s` is shown below
before `ifdown wan1`:
```
default via 10.255.254.1 dev tun0
default via 192.168.50.1 dev 10g-1 metric 7
default via 192.168.123.254 dev 10g-2 metric 8
default via 10.255.254.1 dev tun0 metric 1500
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2
10.255.254.2 dev tun0 scope link metric 1500
127.0.0.0/8 dev lo proto static scope link metric 6
192.168.50.0/24 dev 10g-1 scope link metric 7
192.168.123.0/24 dev 10g-2 scope link metric 8
211.23.22.154 via 192.168.50.1 dev 10g-1 metric 1
211.23.22.154 via 192.168.50.1 dev 10g-1 metric 7
211.23.22.154 via 192.168.123.254 dev 10g-2 metric 8
```

after `ifdown wan1`:
```
default via 10.255.254.1 dev tun0
default via 192.168.123.254 dev 10g-2 metric 8
default via 10.255.254.1 dev tun0 metric 1500
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2
10.255.254.2 dev tun0 scope link metric 1500
127.0.0.0/8 dev lo proto static scope link metric 6
192.168.123.0/24 dev 10g-2 scope link metric 8
211.23.22.154 via 192.168.123.254 dev 10g-2 metric 8
```

 the dev `10g-1` is wan1 and dev `10g-2` is wan2

stt12706, 2025-01-16:
I think I’ve found the issue. When I execute the iPerf command while the route `211.23.22.154 via 192.168.50.1 dev 10g-1 metric 1` exists, the system doesn’t switch to WAN2 if WAN1 fails. 
On the other hand, if I run the iPerf command when this route isn’t present (not manually deleted), the system successfully switches to WAN2 when WAN1 fails.

stt12706, 2025-01-16:
The result of `uci show openmptcprouter.settings` is shown below:
```
openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.sipalg='1'
openmptcprouter.settings.restrict_to_lan='0'
openmptcprouter.settings.vpn='glorytun_udp'
openmptcprouter.settings.master='change'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='3'
openmptcprouter.settings.status_getip_timeout='1'
openmptcprouter.settings.enable_nodelay='1'
openmptcprouter.settings.scaling_governor='performance'
openmptcprouter.settings.menu='OpenMPTCProuter'
openmptcprouter.settings.disableintfrename='1'
openmptcprouter.settings.tracebox='0'
openmptcprouter.settings.version='0.61-6.6'
openmptcprouter.settings.allmptcpovervpn='0'
openmptcprouter.settings.apilc='1737016437'
openmptcprouter.settings.mptcpovervpn='wireguard'
openmptcprouter.settings.dns64='0'
openmptcprouter.settings.proxy='xray'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.encryption='aes-256-gcm'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.disable_fastopen='0'
openmptcprouter.settings.disable_6in4='0'
openmptcprouter.settings.disable_modemmanager='0'
openmptcprouter.settings.external_check='1'
openmptcprouter.settings.debug='0'
openmptcprouter.settings.disablegwping='0'
openmptcprouter.settings.disableloopdetection='0'
openmptcprouter.settings.disableserverhttptest='0'
openmptcprouter.settings.defaultgw='1'
openmptcprouter.settings.modemmanager='1'
openmptcprouter.settings.disableserverping='0'
openmptcprouter.settings.shadowsocksudp='0'
openmptcprouter.settings.ndpi='1'
openmptcprouter.settings.scaling_min_freq='1017600'
openmptcprouter.settings.scaling_max_freq='2208000'
openmptcprouter.settings.sfe_enabled='0'
openmptcprouter.settings.sfe_bridge='0'
openmptcprouter.settings.vnstat_backup='0'
```

Ysurac, 2025-01-16:
Change the System->OpenMPTCProuter->Advanced Settings, "Master interface selection" to "Balancing". Other modes are not tested and will not be supported in next version.

stt12706, 2025-01-16:
Ok, I'll test it again later.
Thanks for your assistance!

stt12706, 2025-01-17:
I tested the "Balancing" mode, and when `211.xxx.xxx.xxx via 192.168.50.1 dev 10g-1 metric 1` was present, the traffic still didn't switch to WAN2 during the iPerf test.

Below is the result of `ip r`:
Before disable WAN1:
```
default via 10.255.254.1 dev tun0
default via 192.168.50.1 dev 10g-1 metric 7
default via 192.168.123.254 dev 10g-2 metric 8
default via 192.168.123.254 dev 10g-2 metric 999
default via 10.255.254.1 dev tun0 metric 1500
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2
10.255.254.2 dev tun0 scope link metric 1500
127.0.0.0/8 dev lo proto static scope link metric 6
192.168.50.0/24 dev 10g-1 scope link metric 7
192.168.123.0/24 dev 10g-2 scope link metric 8
211.xxx.xxx.xxx via 192.168.50.1 dev 10g-1 metric 1
211.xxx.xxx.xxx via 192.168.123.254 dev 10g-2 metric 999
```

After disable WAN1:
```
default via 10.255.254.1 dev tun0
default via 192.168.123.254 dev 10g-2 metric 8
default via 192.168.123.254 dev 10g-2 metric 999
default via 10.255.254.1 dev tun0 metric 1500
10.255.254.1 dev tun0 proto kernel scope link src 10.255.254.2
10.255.254.2 dev tun0 scope link metric 1500
127.0.0.0/8 dev lo proto static scope link metric 6
192.168.123.0/24 dev 10g-2 scope link metric 8
211.xxx.xxx.xxx via 192.168.123.254 dev 10g-2 metric 999
```

stt12706, 2025-02-12:
When I delete the route `211.xxx.xxx.xxx via 192.168.50.1 dev 10g-1 metric 1`, the `ip mptcp monitor` command on the VPS shows MPTCP establishment messages:
```
[       CREATED] token=80b2b3d0 remid=0 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:114.xxx.xxx.xxx sport=65280 dport=63603
[   ESTABLISHED] token=80b2b3d0 remid=0 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:114.xxx.xxx.xxx sport=65280 dport=63603
[SF_ESTABLISHED] token=80b2b3d0 remid=1 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:60.xxx.xxx.xxx sport=65280 dport=49270 backup=0
[       SF_PRIO] token=80b2b3d0 remid=0 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:114.xxx.xxx.xxx sport=65280 dport=63603 backup=1
[        CLOSED] token=ed477aab
```

However, when the route `211.xxx.xxx.xxx via 192.168.50.1 dev 10g-1 metric 1` exists, only a standard TCP connection is established on the VPS, and MPTCP does not take effect:
```
[       CREATED] token=0583252b remid=0 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:60.xxx.xxx.xxx sport=65280 dport=27937
[   ESTABLISHED] token=0583252b remid=0 locid=0 saddr6=::ffff:211.xxx.xxx.xxx daddr6=::ffff:60.xxx.xxx.xxx sport=65280 dport=27937
[        CLOSED] token=0583252b
```
It looks like the routing configuration caused the MPTCP backup to fail.

Ysurac, 2025-02-12:
Did you try latest snapshot ?
Can you send me the result of ```ip mptcp endpoint``` ?

stt12706, 2025-02-12:
I haven’t tested the latest snapshot yet.

Of course!
This is the successful case (I kill this routing rule `211.xxx.xxx.xxx via 192.168.50.1 dev 10g-1 metric 1`):
omr:
```
[          CLOSED] token=2f93548b
[         CREATED] token=5e3b74e9 remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=39575 dport=65280
[     ESTABLISHED] token=5e3b74e9 remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=39575 dport=65280
[       ANNOUNCED] token=5e3b74e9 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[         CREATED] token=c7744e74 remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=39577 dport=65280
[     ESTABLISHED] token=c7744e74 remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=39577 dport=65280
[       ANNOUNCED] token=c7744e74 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[          CLOSED] token=c7744e74
[         CREATED] token=eea6b70d remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=28069 dport=65280
[     ESTABLISHED] token=eea6b70d remid=0 locid=0 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=28069 dport=65280
[       ANNOUNCED] token=eea6b70d remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[          CLOSED] token=eea6b70d
[         CREATED] token=2d360487 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=33453 dport=65280
[     ESTABLISHED] token=2d360487 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=33453 dport=65280
[       ANNOUNCED] token=2d360487 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[  SF_ESTABLISHED] token=2d360487 remid=0 locid=1 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=21898 dport=65280 backup=0 ifindex=7
[          CLOSED] token=2d360487
[          CLOSED] token=dd01a095
[         CREATED] token=f027a801 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=15375 dport=65280
[     ESTABLISHED] token=f027a801 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=15375 dport=65280
[       ANNOUNCED] token=f027a801 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[  SF_ESTABLISHED] token=f027a801 remid=0 locid=1 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=14772 dport=65280 backup=0 ifindex=7
[          CLOSED] token=f027a801
[         CREATED] token=96309f7d remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=47281 dport=65280
[     ESTABLISHED] token=96309f7d remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=47281 dport=65280
[       ANNOUNCED] token=96309f7d remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[  SF_ESTABLISHED] token=96309f7d remid=0 locid=1 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=53894 dport=65280 backup=0 ifindex=7
[          CLOSED] token=96309f7d
[         CREATED] token=aaf648f9 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=61269 dport=65280
[     ESTABLISHED] token=aaf648f9 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=61269 dport=65280
[       ANNOUNCED] token=aaf648f9 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[  SF_ESTABLISHED] token=aaf648f9 remid=0 locid=1 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=53108 dport=65280 backup=0 ifindex=7
[          CLOSED] token=aaf648f9
[          CLOSED] token=4cd39810
[         CREATED] token=87ab51a1 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=53783 dport=65280
[     ESTABLISHED] token=87ab51a1 remid=0 locid=0 saddr4=192.168.123.185 daddr4=211.xx.xx.xx sport=53783 dport=65280
[       ANNOUNCED] token=87ab51a1 remid=2 daddr6=fe80::67c:16ff:feb7:70f6 dport=65280
[  SF_ESTABLISHED] token=87ab51a1 remid=0 locid=1 saddr4=192.168.50.66 daddr4=211.xx.xx.xx sport=46020 dport=65280 backup=0 ifindex=7
[          CLOSED] token=87ab51a1
[          CLOSED] token=fad5cabc
[          CLOSED] token=df3e71bf
[          CLOSED] token=113cd464
```

vps:
```
[       CREATED] token=35d021d3 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=39575
[   ESTABLISHED] token=35d021d3 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=39575
[       CREATED] token=47a8a22a remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=39577
[   ESTABLISHED] token=47a8a22a remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=39577
[        CLOSED] token=47a8a22a
[       CREATED] token=29f40762 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=28069
[   ESTABLISHED] token=29f40762 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=28069
[        CLOSED] token=29f40762
[       CREATED] token=b3f042b6 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:114.xx.xx.xx sport=65280 dport=64863
[   ESTABLISHED] token=b3f042b6 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:114.xx.xx.xx sport=65280 dport=64863
[SF_ESTABLISHED] token=b3f042b6 remid=1 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=21898 backup=0
[       SF_PRIO] token=b3f042b6 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:114.xx.xx.xx sport=65280 dport=64863 backup=1
[        CLOSED] token=b3f042b6
[        CLOSED] token=93d90eb7
[       CREATED] token=de7b91db remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:114.xx.xx.xx sport=65280 dport=65145
[   ESTABLISHED] token=de7b91db remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:114.xx.xx.xx sport=65280 dport=65145
[... cut ...]
[       CREATED] token=852db440 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=47775
[   ESTABLISHED] token=852db440 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=47775
[       CREATED] token=50d99d88 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=40487
[   ESTABLISHED] token=50d99d88 remid=0 locid=0 saddr6=::ffff:211.xx.xx.xx daddr6=::ffff:60.xx.xx.xx sport=65280 dport=40487
[        CLOSED] token=50d99d88
```

closed 2025-05-19 by github-actions[bot]