Ysurac/openmptcprouter #3768: Bridge LAN doesnt work on omr vm

max93776, 2025-01-16:
## Expected Behavior
I want to use 2 ethernet ports (eth0, eth1) bridged to access internet. 

## Current Behavior
I'm runnning OMR as a virtual machine in TrueNASScale and passthrought the NICs. I have eth0 and eth1 bridged on lan firewall zone and dhcp server enabled on eth0. Eth0 is working fine and if I plug in the cable in eth1 I get a IP from DHCP but no internet connection. There is also phy0-ap bridged with wifi but I already tested with it turned off. Any idea whats the problem here?

## Specifications

  - OpenMPTCProuter version: v0.61-6.6 r0
  - OpenMPTCProuter VPS version: ?
  - OpenMPTCProuter VPS provider: openstack 
  - OpenMPTCProuter platform: x86_64

My setup: 
[image]
[image]
If I plug in my laptop in eth1:
```
root@OpenMPTCProuter:~# logread -f
Jan 16 12:17:48 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.100.130 00:e0:4c:c6:94:ba
Jan 16 12:17:48 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.100.130 00:e0:4c:c6:94:ba ***-Laptop
```
From both sides ips are not pingabel, tried already with different laptops. Wi-fi is working.

max93776, 2025-01-16:
Router:
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
    link/ether 00:a0:98:73:f7:46 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
    link/ether 00:a0:98:4e:de:61 brd ff:ff:ff:ff:ff:ff
4: eth2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:a0:98:5f:5b:a5 brd ff:ff:ff:ff:ff:ff
5: eth3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:a0:98:66:6d:31 brd ff:ff:ff:ff:ff:ff
    inet 192.168.10.169/24 brd 192.168.10.255 scope global eth3
       valid_lft forever preferred_lft forever
6: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:a0:98:70:bb:9c brd ff:ff:ff:ff:ff:ff
    inet 192.168.11.12/24 brd 192.168.11.255 scope global eth4
       valid_lft forever preferred_lft forever
7: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr aa19:5ab9:a68b::
8: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
9: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
10: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
11: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
12: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr daac:dc80:e1dc::
13: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
14: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:a0:98:73:f7:46 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fe80::2a0:98ff:fe73:f746/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
20: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 2000
    link/ether e8:c8:29:76:c6:f9 brd ff:ff:ff:ff:ff:ff
21: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.255.252.3/24 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::126:3e80:a5fd:ee9c/64 scope link stable-privacy proto kernel_ll
       valid_lft forever preferred_lft forever
root@OpenMPTCProuter:~# ip r
default via 10.255.252.1 dev tun0
default via 192.168.11.2 dev eth4 metric 1
default via 192.168.10.2 dev eth3 metric 7
default via 192.168.11.2 dev eth4 metric 8
default via 192.168.10.2 dev eth3 metric 999
default via 10.255.252.1 dev tun0 metric 1500
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.3
10.255.252.0/24 dev tun0 scope link metric 1500
10.255.252.1 via 10.255.252.1 dev tun0
127.0.0.0/8 dev lo proto static scope link metric 5
192.168.10.0/24 dev eth3 scope link metric 7
192.168.11.0/24 dev eth4 scope link metric 8
212.********** via 192.168.11.2 dev eth4 metric 1
212.********** via 192.168.10.2 dev eth3 metric 999
```
Laptop:
```
>ipconfig
[... 65 lines cut ...]
===========================================================================
Ständige Routen:
  Netzwerkadresse          Netzmaske  Gatewayadresse  Metrik
          0.0.0.0          0.0.0.0      192.168.1.1  Standard
===========================================================================

IPv6-Routentabelle
===========================================================================
Aktive Routen:
 If Metrik Netzwerkziel             Gateway
  1    331 ::1/128                  Auf Verbindung
 19    281 fe80::/64                Auf Verbindung
 19    281 fe80::f90b:91c8:940a:3779/128
                                    Auf Verbindung
  1    331 ff00::/8                 Auf Verbindung
 19    281 ff00::/8                 Auf Verbindung
===========================================================================
Ständige Routen:
  Keine
```
note: I turned off wlan on my laptop for route PRINT because its better to read

if I use the working eth0, the routes in OMR and my laptop stay the same.

Maybe it has something to do with TrueNASscale?

max93776, 2025-01-21:
Hi!! Have you ever had similar experiences with such an error? If not then I'll close this issue :)

Ysurac, 2025-01-22:
No, I use bridge on one test platform without issues.
I would look in ```uci show network```, ```uci show firewall``` and ```uci show dhcp```.

max93776, 2025-04-16:
It was a NIC passthrought error in the hypervisor os. thx

closed 2025-04-16 by max93776

Ysurac, 2025-04-16:
Thanks for the feedback.