Ysurac/openmptcprouter #3769: Redundant Scheduler does not work

SpencerXZX, 2025-01-16:
## Expected Behavior

Redundant scheduler should send duplicated packets on all interfaces and use the fastest response.

## Current Behavior

Internet is not accessible when using redundant scheduler, however I can ping IPs and resolve DNS, so I am not sure where the problem lies. The CPU usage of my router goes all the way to max, showing a load of 20 despite being a quad core router.

[image]

## Possible Solution

Not sure. But all of the other schedulers seem to work, so it seems to be just this one.

## Steps to Reproduce the Problem

  1. Turn on mptcp redundancy scheduler
  2. Try to browse the internet.
 
## Context (Environment)

Trying to minimize latency for online gaming using two WANs.

Redundant scheduler causes high cpu usage and no internet access.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.62-snapshot-6.6 r0+28308-c06d4df974
  - OpenMPTCProuter VPS version: 0.1032-test 6.6.36-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Local provider
  - OpenMPTCProuter platform: R5S
  - Country: USA

Ysurac, 2025-01-17:
I don't have issues with Shadowsocks-Rust and XRay, but OpenVPN doesn't work with red scheduler.
I will check that.

SpencerXZX, 2025-01-17:
I am using XRay VLess with Glorytun TCP and it does not work. I've tried a bunch of different proxies but never swapped from Glorytun, maybe that's it?

Ysurac, 2025-01-17:
You can disable VPN to test. Glorytun TCP doesn't always work well with 6.6

SpencerXZX, 2025-01-17:
Ok I will test when I get home, could this have been causing my issue?

SpencerXZX, 2025-01-17:
I updated my router to the latest snapshot after the glorytun tcp update, however I am still seeing the same results. This time, I tried disabling the VPN entirely. Now the status screen sometimes shows direct output, and for a few seconds will show connected to the VPS, in fact, I was able to get a webpage to half load by refreshing a bunch, but it wouldn't fully load. I tested shadowsocks rust 2022 and xray vless, both with vpn disabled and couldn't get it to work. Please let me know if I can provide any other info to help:

[image]

[image]

Ysurac, 2025-01-17:
The new snapshot is not yet build, and It's not directly related I think.
I'm making tests on 6.12, I will test on 6.6.

SpencerXZX, 2025-01-19:
Thanks, I tried compiling 6.12 but it doesn't appear to be compatible with R5S yet, or I did something wrong. I'm not very familiar with compiling from source so it's likely I made a mistake.

Ysurac, 2025-01-19:
It's only tested on x86_64 for now.

SpencerXZX, 2025-01-27:
Have you had a chance to test this on 6.6 yet? Or can anyone else test on 6.6? Trying to figure out if it's an issue on my side or not. I don't have access to my x86 test device anymore so I can't test 6.12 on x86_64.

aydinj, 2025-04-05:
I’m experiencing the same issue with v0.62.
The redundant mode has consistently had problems — it’s by far the most confusing part of the system. I’m not an expert, but I rely on this feature desperately. I’ve tried many versions, and the problem has always persisted.
Please address this. For some of us who are far from stable connections and depend on this for mission-critical situations, it’s absolutely essential.

SpencerXZX, 2025-04-05:
At this point I'm just waiting for the 6.12 image for R5S to be available and I'm hoping it's fixed there @aydinj .

closed 2025-07-10 by github-actions[bot]