Ysurac/openmptcprouter #3770: OMR-Bypass Not working

MarcelHService, 2025-01-17:
## Expected Behavior
OMR-Bypass as Domain 
Domain: 'https://promods.net' Interface eth0.10 

## Current Behavior
OMR-Bypass Not working. Same isp on the mobile phone it work the Domain. In Ethernet its Not Working. 
Ip got blocked by Promods.net
I have try to Chance the Interface but Same result. The Networksettings on the PC are:
IPv4: 192.168.100.255
Gateway: 192.168.100.1
DNS: 192.168.100.1

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1031
  - OpenMPTCProuter VPS provider: Ionos Germany
  - OpenMPTCProuter platform: x64

Ysurac, 2025-01-17:
Can you try latest snapshot kernel 6.6 ? https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
I fixed some issues in omr-bypass

MarcelHService, 2025-01-17:
Still the Same Problem.

on https://promods.net/

> Request Blocked
Your request to access promods.net has been blocked because your IP address appears to be abusive

> 

Versions

Server: 6.6.36-x64v3-xanmod1 0.1032-test
Router: openmptcprouter v0.62-snapshot-6.6 r0+28308-c06d4df974
Default Proxy: XRay VMESS
Default VPN: A Dead Simple VPN

Server Log

> Jan 17 20:32:14 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address

>

Ysurac, 2025-01-17:
I would need, via SSH on the router, the result of ```nft list ruleset```

MarcelHService, 2025-01-17:
Here there are.

https://pastebin.com/1S3i4aSu

Ysurac, 2025-01-17:
I find ips for promods.net, but not in good section. I would also need the result of ```uci show omr-bypass```

MarcelHService, 2025-01-18:
sure here there are https://pastebin.com/PgcQiQMZ

Ysurac, 2025-01-18:
The domain must be without https://, so only promods.net

MarcelHService, 2025-01-18:
i have test it but same result.

when i do as domain www.whatismyip.com or whatismyip.com then i get still the VPS Server IP not the WAN Ip

MarcelHService, 2025-01-18:
i don't know what i do but now its work.

Thanks a lot for your Support :)

closed 2025-01-18 by MarcelHService

MarcelHService, 2025-11-13:
Hi,
I have Version v0.63-6.6 on the Router and Version 6.12.41-x64v3-xanmod1 0.1039 on the VPS installed and it works fine for me.