Ysurac/openmptcprouter #3772: Question about OMR kernel 5.4 now that it is deprecated.

vempire-ghost, 2025-01-18:
For my use case I need to use OMR with Kernel 5.4, which works perfectly, my question now is how the VPS installation will be handled since it is no longer supported. Do you plan to completely eliminate the option to install this version in the future, or will it just no longer be updated?

I can save the image of OMR router, but the VPS script depends on your repository to download the packages, so if you deactivate it, it would be impossible to install it in the future. If you intend to deactivate it, could you inform me how I can download the necessary files for the installation script to work locally on the VPS without needing your repository?

I plan to migrate to upstream as soon as the official kernel has satisfactory support for the redundant scheduler.

Ysurac, 2025-01-19:
For now VPS script will stay compatible with OMR kernel 5.4, but not tested.
Everything is on Github: https://github.com/Ysurac/openmptcprouter-vps and URL used for scripts can be modified via VPSURL parameter (currently https://www.openmptcprouter.com)

vempire-ghost, 2025-01-20:
Thanks Ysurac. I cloned the deb repository and forked it from GitHub to ensure a frozen version in case any updates break something in 5.4. Thank you very much for your work.

closed 2025-01-20 by vempire-ghost