Ysurac/openmptcprouter #3776: Merge both 2xOMRRouter for One VPS

GermanAizek, 2025-01-22:
Hi @Ysurac, thanks for huge contribution to project.
Since I have OMRrouter on old x86-64 PCs, number PCI Express lines is limited for network cards, I lack several ports, is it possible to scale by connecting another OMR router?

```

| OMRrouter |-------------\
                           \
                             \
                              |---------------> | VPS |
                             /
                           /
| OMRrouter |-------------/

```

GermanAizek, 2025-01-22:
> 1 Router : 1 VPS

This is very bad, then there's only one way out, I will have to install server motherboard that has more PCI-E lines with 2 CPUs, but it also consumes more electricity.

GermanAizek, 2025-01-22:
> 1 Router : 1 VPS

@darkman1983, @Ysurac 
I wonder if anyone has tried using VPS out-merged-connection as WAN for another OMR router?
Is it possible not to update motherboard in this way, and achieve full channel aggregation?

Ysurac, 2025-01-23:
No, it's not possible. How to connect VPS 0 to second router ?
It's possible, if you have 1Gb/s ports, to connect a switch to one card and by using static IPs (or vlan with a manageable switch).

Ysurac, 2025-01-23:
It's another solution for your issue...

GermanAizek, 2025-01-23:
> It's another solution for your issue...

And now I understand what it's about, really as solution to problem.