Ysurac/openmptcprouter #3778: Doesn't show WAN ADDRESS and WHO IS in "Status Tab"

rogerio77, 2025-01-22:
- OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 6.10.2-x64v3-xanmod1 0.1032-test
  - OpenMPTCProuter VPS provider: Contabo
  - OpenMPTCProuter platform: Virtual Box
  - Country: Argentina

Hi all
I can't see wan ip in the second wan interface, Also I can't see "who is". Look the picture attached to understand better what I mean

[image]

[image]

I tried "cubic" and neither do it

Ysurac, 2025-01-23:
There is high latencies here, there is a 1s timeout on whois command and also a 1s timeout on public IP test.
For the public IP test, you can change setting in System->OpenMPTCProuter, "Advanced settings" tab, "WAN IPs retrieve timeout".
I will also add a setting for whois.

rogerio77, 2025-01-23:
Thanks for your answer!!

closed 2025-01-29 by rogerio77