Ysurac/openmptcprouter #3781: OMR-Bypass for Netflix

speegs, 2025-01-25:
I'm expecting Netflix to bypass the VPN/Bonding.  Which does work if I add the device to OMR-Bypass via Mac address (yay).

However I've seen in various posts it should also work via AS Number.  Am I doing it wrong or is this part just not working?

[image]

Effectively this seems to have no effect.  I'm pretty new to all this, but can ssh into and check things out if I know where to look.

Using: openmptcprouter v0.62-snapshot-6.6 r0+28381-ad98c322cc
On a Flint 2 router, so certainty running experimental software, but I think maybe I'm just doing it wrong and would appreciate some pointers.

I'd rather not add just about every mobile phone and ipad to the Mac address bypass so Netflix works.  Seems better to just make Netflix only bypass.

The domain name bypass is working great for reddit.com and yes all the clients are using the router as the DNS server.  As far as I can tell, the DNS bypass method is not helpful for netflix.  If I don't add the MAC of the client, I get the netflix vpn/proxy error every time.

Ysurac, 2025-01-26:
What is the result of ```nft list ruleset``` and of ```uci show omr-bypass``` ?
Netflix can be bypassed using service section.

speegs, 2025-01-26:
I didn't know about the services section, still so much to learn.  I just googled it an AS Number was coming up as the way to do it, probably old information.

Here is the output, I've changed the last few digits on the MAC addresses.

root@OpenWrt:~# nft list ruleset
table ip mangle {
        chain PREROUTING {
                type filter hook prerouting priority mangle; policy accept;
        }

        chain INPUT {
                type filter hook input priority mangle; policy accept;
                counter packets 38015117 bytes 6675390823 jump omr-bypass-dpi
        }

        chain FORWARD {
                type filter hook forward priority mangle; policy accept;
                counter packets 10820469 bytes 19819026294 jump omr-bypass-dpi
        }

        chain OUTPUT {
                type route hook output priority mangle; policy accept;
        }

        chain POSTROUTING {
                type filter hook postrouting priority mangle; policy accept;
        }

        chain omr-bypass-dpi {
        }
}
table inet fw4 {
        ct helper amanda {
                type "amanda" protocol udp
                l3proto inet
        }

        ct helper ftp {
                type "ftp" protocol tcp
                l3proto inet
        }

        ct helper RAS {
                type "RAS" protocol udp
                l3proto inet
        }

        ct helper Q.931 {
                type "Q.931" protocol tcp
                l3proto inet
        }

        ct helper irc {
                type "irc" protocol tcp
                l3proto ip
        }

        ct helper netbios-ns {
                type "netbios-ns" protocol udp
                l3proto ip
        }

        ct helper pptp {
                type "pptp" protocol tcp
                l3proto ip
        }

        ct helper sane {
                type "sane" protocol tcp
                l3proto inet
[... 689 lines cut ...]
omr-bypass.tun0=interface
omr-bypass.tun0.id='1500'
omr-bypass.@asns[0]=asns
omr-bypass.@asns[0].asn='2906'
omr-bypass.@asns[0].note='Netflix1'
omr-bypass.@asns[1]=asns
omr-bypass.@asns[1].asn='40027'
omr-bypass.@asns[1].note='Netflix2'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='reddit.com'
omr-bypass.@domains[0].family='ipv4ipv6'
omr-bypass.@macs[0]=macs
omr-bypass.@macs[0].mac='38:E1:3D:A6:00:00'
omr-bypass.@macs[1]=macs
omr-bypass.@macs[1].mac='7A:85:29:C7:00:00'
omr-bypass.@macs[2]=macs
omr-bypass.@macs[2].mac='00:04:4B:B7:00:00'
omr-bypass.@macs[3]=macs
omr-bypass.@macs[3].mac='9A:89:7E:62:00:00'
omr-bypass.@macs[4]=macs
omr-bypass.@macs[4].mac='9C:95:61:3A:00:00'
omr-bypass.@macs[5]=macs
omr-bypass.@macs[5].mac='AC:5A:F0:0B:00:00'
omr-bypass.bonding_omrvpn=interface
omr-bypass.bonding_omrvpn.id='1500'

speegs, 2025-01-26:
I've added the services for netflix and reddit.  Will see how it goes doing it that way.  Sorry I didn't know about that way of doing it.  I think it needs time to reset to see if it works out.  I know it took time to be effective on the dns omr rule for reddit.

The mac rules seemed rather quick to become active.

speegs, 2025-01-27:
Thanks the services mode seems to work fine.  Don't know why the ASN mode didn't work out.  Seems services is superior and I overlooked that, thinking it was just for protocols.  Not for predefined streaming services and websites of which there seems to be a good number of popular ones built in.

closed 2025-01-27 by speegs