Ysurac/openmptcprouter #3783: OMR running on Mini PC from Amazon suddenly crashing & rebooting after years of great performance

peterxxl, 2025-01-29:
## Expected Behavior
to work as it did for 2 years

## Current Behavior
I have 6 port Mini PC from Amazon connected to 5 Starlink WANs
[Micro Firewall Appliance, 6 Intel I225-V 2.5GbE NIC Ports Fanless Mini PC, Intel Core i3 10110U Dual Core, Support AES-NI,TPM2.0 1xDP 1xHDMI 1xCOM 6xUSB, 16GB DDR4, 128GB M.2 SSD]

LAN port is supplying bonded internet to UDM-SE thru 2.5Gbps  
Initially (2 years ago when I first set it up) there was a problem with the interfaces, but this was resolved by selecting "Force link" on each interface

Since then, this has been working pretty much flawlessly. 
Lately I started getting lots of internmitent outages on the Unifi graph... and was not sure why.
And today, the unit became pretty much unusable... it reboots/crashes every minute or less. One can tell by the uptime counter constantly restarting. And the internet not working of course.

[image]

There was no change in configuration. It only happens when I set in Unifi to use this as primary WAN. 
If I set it as failover, it doesn't crash. I thought it might have something to do with the load. But when I set it as failover, SSH into the terminal and do omr-test-speed, it pulls over 500Mbps sustained for multiple minutes without hiccup. The moment I switch the WANs in Unifi and set it as primary, it starts crashing again. And the load is between 50 and 80 Mbps, so nowhere near the speed test.

here is system log: https://pastebin.com/raw/cML4sH8r
here is kernel log: https://pastebin.com/raw/mygpEXV1

## Specifications

  - OpenMPTCProuter version: v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 6.6.36-x64v2-xanmod1 0.1031
  - OpenMPTCProuter VPS provider: Grupo ZGH
  - OpenMPTCProuter platform: x86_64

darkman1983, 2025-01-30:
So, maybe you should check the CPU + RAM usage on that load.
It doesn't depend on the throughput only.
If your traffic is coming from the UDM, maybe there're much more connections and open ports, then a simple speedtest on OMR.
Maybe you're overloading the Dual-Core CPU?!

peterxxl, 2025-02-04:
Hello @darkman1983 and thank you for your response.

as far as I can tell, CPU load seems to be fine... looking at `top` output, it's usually 90%+ idle
same thing with memory... it has 16GB and seems to be mostly free... less than 1GB used

web interface is quite sluggish thou... 

how can I determine if the storage is dying? I believe it's a SSD memory and I can easily replace that if that's the case, but how I can determine that?

also I am getting lots of these messages in the log:
`
Feb  4 05:27:44 OpenMPTCProuter daemon.err /usr/bin/ss-redir[24745]: send: Connection reset by peer
Feb  4 05:28:25 OpenMPTCProuter daemon.err /usr/bin/ss-redir[24745]: remote recv: Connection reset by peer
`

 I have modified one of the Starlinks to use public IP and set it directly to the interface based on your recommendation... is that the correct way to do it?

peterxxl, 2025-02-04:
now the mini PC running OMR seems to be crashing constantly

it reboots... runs for few seconds and reboots again

```
PING 10.2.0.1 (10.2.0.1): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
Request timeout for icmp_seq 5
Request timeout for icmp_seq 6
Request timeout for icmp_seq 7
Request timeout for icmp_seq 8
Request timeout for icmp_seq 9
Request timeout for icmp_seq 10
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12
Request timeout for icmp_seq 13
Request timeout for icmp_seq 14
Request timeout for icmp_seq 15
Request timeout for icmp_seq 16
Request timeout for icmp_seq 17
Request timeout for icmp_seq 18
Request timeout for icmp_seq 19
Request timeout for icmp_seq 20
Request timeout for icmp_seq 21
Request timeout for icmp_seq 22
Request timeout for icmp_seq 23
Request timeout for icmp_seq 24
Request timeout for icmp_seq 25
Request timeout for icmp_seq 26
Request timeout for icmp_seq 27
Request timeout for icmp_seq 28
Request timeout for icmp_seq 29
Request timeout for icmp_seq 30
Request timeout for icmp_seq 31
Request timeout for icmp_seq 32
Request timeout for icmp_seq 33
Request timeout for icmp_seq 34
Request timeout for icmp_seq 35
Request timeout for icmp_seq 36
Request timeout for icmp_seq 37
Request timeout for icmp_seq 38
Request timeout for icmp_seq 39
Request timeout for icmp_seq 40
Request timeout for icmp_seq 41
Request timeout for icmp_seq 42
Request timeout for icmp_seq 43
Request timeout for icmp_seq 44
Request timeout for icmp_seq 45
Request timeout for icmp_seq 46
Request timeout for icmp_seq 47
Request timeout for icmp_seq 48
Request timeout for icmp_seq 49
Request timeout for icmp_seq 50
Request timeout for icmp_seq 51
Request timeout for icmp_seq 52
Request timeout for icmp_seq 53
Request timeout for icmp_seq 54
Request timeout for icmp_seq 55
Request timeout for icmp_seq 56
Request timeout for icmp_seq 57
Request timeout for icmp_seq 58
Request timeout for icmp_seq 59
Request timeout for icmp_seq 60
Request timeout for icmp_seq 61
Request timeout for icmp_seq 62
Request timeout for icmp_seq 63
64 bytes from 10.2.0.1: icmp_seq=64 ttl=62 time=0.714 ms
64 bytes from 10.2.0.1: icmp_seq=65 ttl=62 time=0.779 ms
64 bytes from 10.2.0.1: icmp_seq=66 ttl=62 time=0.619 ms
64 bytes from 10.2.0.1: icmp_seq=67 ttl=62 time=0.994 ms
Request timeout for icmp_seq 68
Request timeout for icmp_seq 69
Request timeout for icmp_seq 70
Request timeout for icmp_seq 71
Request timeout for icmp_seq 72
Request timeout for icmp_seq 73
Request timeout for icmp_seq 74
Request timeout for icmp_seq 75
Request timeout for icmp_seq 76
Request timeout for icmp_seq 77
Request timeout for icmp_seq 78
Request timeout for icmp_seq 79
Request timeout for icmp_seq 80
Request timeout for icmp_seq 81
Request timeout for icmp_seq 82
Request timeout for icmp_seq 83
Request timeout for icmp_seq 84
Request timeout for icmp_seq 85
Request timeout for icmp_seq 86
Request timeout for icmp_seq 87
Request timeout for icmp_seq 88
Request timeout for icmp_seq 89
Request timeout for icmp_seq 90
Request timeout for icmp_seq 91
Request timeout for icmp_seq 92
Request timeout for icmp_seq 93
Request timeout for icmp_seq 94
Request timeout for icmp_seq 95
Request timeout for icmp_seq 96
Request timeout for icmp_seq 97
Request timeout for icmp_seq 98
Request timeout for icmp_seq 99
Request timeout for icmp_seq 100
Request timeout for icmp_seq 101
Request timeout for icmp_seq 102
Request timeout for icmp_seq 103
Request timeout for icmp_seq 104
Request timeout for icmp_seq 105
Request timeout for icmp_seq 106
Request timeout for icmp_seq 107
Request timeout for icmp_seq 108
Request timeout for icmp_seq 109
Request timeout for icmp_seq 110
Request timeout for icmp_seq 111
Request timeout for icmp_seq 112
Request timeout for icmp_seq 113
Request timeout for icmp_seq 114
Request timeout for icmp_seq 115
Request timeout for icmp_seq 116
Request timeout for icmp_seq 117
Request timeout for icmp_seq 118
Request timeout for icmp_seq 119
Request timeout for icmp_seq 120
Request timeout for icmp_seq 121
Request timeout for icmp_seq 122
Request timeout for icmp_seq 123
Request timeout for icmp_seq 124
Request timeout for icmp_seq 125
Request timeout for icmp_seq 126
Request timeout for icmp_seq 127
Request timeout for icmp_seq 128
Request timeout for icmp_seq 129
Request timeout for icmp_seq 130
Request timeout for icmp_seq 131
64 bytes from 10.2.0.1: icmp_seq=132 ttl=62 time=0.645 ms
64 bytes from 10.2.0.1: icmp_seq=133 ttl=62 time=0.752 ms
64 bytes from 10.2.0.1: icmp_seq=134 ttl=62 time=0.717 ms
64 bytes from 10.2.0.1: icmp_seq=135 ttl=62 time=0.603 ms
64 bytes from 10.2.0.1: icmp_seq=136 ttl=62 time=0.579 ms
Request timeout for icmp_seq 137
Request timeout for icmp_seq 138
Request timeout for icmp_seq 139
Request timeout for icmp_seq 140
Request timeout for icmp_seq 141
Request timeout for icmp_seq 142
Request timeout for icmp_seq 143
Request timeout for icmp_seq 144
Request timeout for icmp_seq 145
Request timeout for icmp_seq 146
Request timeout for icmp_seq 147
```

Ysurac, 2025-02-04:
If there is no updates of OpenMPTCProuter, backup configuration first.
Then:
* Put a screen if possible and check what can be displayed.
* Clean all fans and everything inside.
* Try to connect a computer directly to it with another RJ45 cable and check if there is same problem.
* Try to install smartmontools to check smart info of SSD/HDD (opkg update && opkg install smartmontool && smartcl -a /dev/sda)

It's working correctly without unifi ?

peterxxl, 2025-02-04:
now I have it on WAN2 as failover and works fine... 

```
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149711] [eth0]   subsystems: net, pci
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149719] [eth0]   drivers: igc
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149728] [eth0]   vendor: 8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149736] [eth0]   product: 15f3
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149745] [eth0]   subsystem vendor: 8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149753] [eth0]   revision: 0003
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149761] [eth0]   ID_VENDOR_ID: 0x8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149771] [eth0]   ID_MODEL_ID: 0x15f3
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149781] [eth0]   ID_REVISION: 0x0003
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149843] [eth0] property added: ID_MM_CANDIDATE=1
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149872] [base-manager] adding port eth0 at sysfs path: /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/net/eth0
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149887] [filter] (net/eth0) port allowed: device is allowlisted by plugin (vid)
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.149897] [base-manager] port eth0 is first in device '/sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0'
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.150000] [eth0/probe] port type hints loaded: AT no, QMI no, MBIM no, QCDM no, XMMRPC no, AUDIO no, GPS no
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.150021] [plugin-manager] task 0: new support task for device: /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.150036] [plugin-manager] task 0: port added: eth0
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.150048] [plugin-manager] task 0,eth0: new support task for port
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.150057] [plugin-manager] task 0,eth0: deferred until min wait time elapsed
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170572] [base-manager] kernel event reported:
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170592] [base-manager]   action:    add
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170604] [base-manager]   subsystem: net
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170615] [base-manager]   name:      eth1
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170627] [base-manager]   uid:       n/a
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170646] [eth1] preloading contents and properties...
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170687] [eth1] sysfs path: /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/net/eth1
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170850] [eth1] port contents loaded:
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170860] [eth1]   bus: pci
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170872] [eth1]   device: /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170885] [eth1]   subsystems: net, pci
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170896] [eth1]   drivers: igc
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170908] [eth1]   vendor: 8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170919] [eth1]   product: 15f3
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170930] [eth1]   subsystem vendor: 8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170942] [eth1]   revision: 0003
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170953] [eth1]   ID_VENDOR_ID: 0x8086
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170967] [eth1]   ID_MODEL_ID: 0x15f3
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.170979] [eth1]   ID_REVISION: 0x0003
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.171044] [eth1] property added: ID_MM_CANDIDATE=1
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.171076] [base-manager] adding port eth1 at sysfs path: /sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0/net/eth1
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.171093] [filter] (net/eth1) port allowed: device is allowlisted by plugin (vid)
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.171107] [base-manager] port eth1 is first in device '/sys/devices/pci0000:00/0000:00:1c.5/0000:02:00.0'
Feb  4 16:24:11 OpenMPTCProuter daemon.debug [29202]: <dbg> [1738686251.171136] [eth1/probe] port type hints loaded: AT no, QMI no, MBIM no, QCDM no, XMMRPC no, AUDIO no, GPS no
[... cut ...]
Feb  4 16:24:42 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Feb  4 16:24:42 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Feb  4 16:25:31 OpenMPTCProuter daemon.info unbound: [1243:0] info: generate keytag query _ta-4f66. NULL IN
Feb  4 16:38:29 OpenMPTCProuter authpriv.info dropbear[15051]: Child connection from 10.2.0.2:51092
Feb  4 16:38:32 OpenMPTCProuter authpriv.notice dropbear[15051]: Password auth succeeded for 'root' from 10.2.0.2:51092
```

peterxxl, 2025-02-04:
in the afternoon I will get screen and go there and try to capture whats happening when it's connected as main source of the internet for the network and crashes

closed 2025-05-10 by github-actions[bot]