Ysurac/openmptcprouter #3789: NanoPi R5C Hardwaresupport

chhofi, 2025-02-02:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

This request is for adding support for the NanoPi R5C, a single-board computer (SBC) based on the Rockchip RK3568 SoC ink. WIFI. The board features ARM Cortex-A55 cores, dual 2.5GbE Ethernet, and other peripherals that could benefit from openmptcrouter support.

**Describe why you are asking for this support?**
Currently, openmptcrouter has no official support for that router.

**Additional context**
https://www.friendlyelec.com/index.php?route=product/product&path=69&product_id=290

chhofi, 2025-02-04:
I see there is already a config for the R5C available. How can I contribute and compile an image that can than be downloaded from the website?

Ysurac, 2025-02-04:
I just launched a r5c image build, will be available in a few hours (max 11h) on https://snapshots.openmptcprouter.com/

chhofi, 2025-02-04:
Wow xD that was quick 😇Thanks so much

chhofi, 2025-02-05:
I tested the install multiple times. The good news is that the led blinking indication is working. But I can not reach the web interface on 192.168.100.1. I have set the ip of my pc to 192.168.100.2

chhofi, 2025-02-06:
Ok, I made a stupid mistake. When I plug into the WAN port I get dhcp and 192.168.100.1 is the ui. Do not know if this is the expected behavior or if the ports are switched in the config but like this it works :)