Ysurac/openmptcprouter #3790: Nanopi R5C build - Interface eth1 constantly loosing connection

chhofi, 2025-02-07:
Interface eth1 is experiencing intermittent connectivity. The interface is going down (losing link connectivity) and coming back up.

`Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5268] N: rops_handle_POLLIN_netlink: DELADDR
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5270] N: [wsisrv|7|adopted]: _lws_route_check_wsi: dest route gone
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5272] N: WS closed from 192.168.100.2, clients: 0
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5273] N: killing process, pid: 22969
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5276] N: __lws_lc_untag:  -- [wsisrv|7|adopted] (0) 16.008s
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5282] N: process killed with signal 1, pid: 22969
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5365] N: rops_handle_POLLIN_netlink: DELADDR
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:5368] N: rops_handle_POLLIN_netlink: DELADDR
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is disabled
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Network device 'eth1' link is down
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' has link connectivity loss
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now down
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:9776] N: rops_handle_POLLIN_netlink: DELADDR
Jan 21 10:06:36 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is disabled
Jan 21 10:06:36 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:36:9884] N: rops_handle_POLLIN_netlink: DELADDR
Jan 21 10:06:37 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is enabled
Jan 21 10:06:37 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is setting up now
Jan 21 10:06:37 OpenMPTCProuter daemon.notice netifd: Interface 'lan' is now up
Jan 21 10:06:37 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of lan (eth1)
Jan 21 10:06:38 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 21 10:06:39 OpenMPTCProuter daemon.notice netifd: Network device 'eth1' link is up
Jan 21 10:06:39 OpenMPTCProuter daemon.notice netifd: Interface 'lan' has link connectivity 
Jan 21 10:06:40 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 21 10:06:41 OpenMPTCProuter user.notice omr-bypass: OMR-ByPass is running
Jan 21 10:06:41 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:41:6397] N: __lws_lc_tag:  ++ [wsisrv|8|adopted] (1)
Jan 21 10:06:41 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:41:6404] N: HTTP /token - 192.168.100.2
Jan 21 10:06:41 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:41:6462] N: __lws_lc_tag:  ++ [wsisrv|9|adopted] (2)
Jan 21 10:06:41 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:41:6472] N: WS   /ws - 192.168.100.2, clients: 1
Jan 21 10:06:41 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:41:6527] N: started process, pid: 28138
Jan 21 10:06:43 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Jan 21 10:06:43 OpenMPTCProuter auth.info login[28138]: root login on 'pts/0'
Jan 21 10:06:44 OpenMPTCProuter user.notice unbound: default memory configuration
Jan 21 10:06:46 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:46:6433] N: __lws_lc_untag:  -- [wsisrv|8|adopted] (1) 5.003s
Jan 21 10:06:46 OpenMPTCProuter user.notice MPTCP: Set wan1 to master from wan1 is deactivated
Jan 21 10:06:46 OpenMPTCProuter user.notice MPTCP: Set wan2 to on from wan2 is deactivated
Jan 21 10:06:47 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:47:2014] N: WS closed from 192.168.100.2, clients: 0
Jan 21 10:06:47 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:47:2016] N: killing process, pid: 28138
Jan 21 10:06:47 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:47:2018] N: __lws_lc_untag:  -- [wsisrv|9|adopted] (0) 5.555s
Jan 21 10:06:47 OpenMPTCProuter daemon.notice ttyd[21119]: [2025/01/21 10:06:47:2026] N: process killed with signal 1, pid: 28138
Jan 21 10:06:48 OpenMPTCProuter daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Jan 21 10:06:48 OpenMPTCProuter daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Jan 21 10:06:48 OpenMPTCProuter daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses`

This happens on a fresh install.

Ysurac, 2025-02-07:
What do you have in ```dmesg``` or Status->Kernel log ?

chhofi, 2025-02-07:
The output of the kernel log is the following:
```

[  534.333261] r8169 0002:01:00.0 eth1: Link is Down
[  534.385122] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  534.665242] r8169 0002:01:00.0 eth1: Link is Down
[  534.855010] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  535.155152] r8169 0002:01:00.0 eth1: Link is Down
[  537.751505] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  553.242306] r8169 0002:01:00.0 eth1: Link is Down
[  553.295031] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  553.575239] r8169 0002:01:00.0 eth1: Link is Down
[  553.755024] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  554.035187] r8169 0002:01:00.0 eth1: Link is Down
[  556.658126] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  572.241734] r8169 0002:01:00.0 eth1: Link is Down
[  572.295087] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  572.575112] r8169 0002:01:00.0 eth1: Link is Down
[  572.755104] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  573.035217] r8169 0002:01:00.0 eth1: Link is Down
[  575.755010] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  591.233309] r8169 0002:01:00.0 eth1: Link is Down
[  591.285040] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  591.555305] r8169 0002:01:00.0 eth1: Link is Down
[  591.745074] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  592.015218] r8169 0002:01:00.0 eth1: Link is Down
[  594.545505] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  610.436286] r8169 0002:01:00.0 eth1: Link is Down
[  610.485040] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  610.755230] r8169 0002:01:00.0 eth1: Link is Down
[  610.935018] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  611.215232] r8169 0002:01:00.0 eth1: Link is Down
[  613.866780] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  629.499702] r8169 0002:01:00.0 eth1: Link is Down
[  629.545104] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  629.815232] r8169 0002:01:00.0 eth1: Link is Down
[  630.005040] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  630.285246] r8169 0002:01:00.0 eth1: Link is Down
[  633.037604] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  648.520460] r8169 0002:01:00.0 eth1: Link is Down
[  648.565029] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  648.845208] r8169 0002:01:00.0 eth1: Link is Down
[  649.025020] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  649.295239] r8169 0002:01:00.0 eth1: Link is Down
[  651.819364] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  667.035521] r8169 0002:01:00.0 eth1: Link is Down
[  667.085031] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  667.365233] r8169 0002:01:00.0 eth1: Link is Down
[  667.545448] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  667.825231] r8169 0002:01:00.0 eth1: Link is Down
[  670.480548] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  686.006905] r8169 0002:01:00.0 eth1: Link is Down
[  686.055030] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  686.325247] r8169 0002:01:00.0 eth1: Link is Down
[  686.505022] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  686.805248] r8169 0002:01:00.0 eth1: Link is Down
[  689.507713] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  705.003154] r8169 0002:01:00.0 eth1: Link is Down
[  705.055124] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  705.325187] r8169 0002:01:00.0 eth1: Link is Down
[  705.505035] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  705.775230] r8169 0002:01:00.0 eth1: Link is Down
[  708.389756] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  723.729192] r8169 0002:01:00.0 eth1: Link is Down
[  723.775026] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  724.055227] r8169 0002:01:00.0 eth1: Link is Down
[  724.225041] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  724.515211] r8169 0002:01:00.0 eth1: Link is Down
[  727.046600] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[  742.646522] r8169 0002:01:00.0 eth1: Link is Down
[  742.705040] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  742.975217] r8169 0002:01:00.0 eth1: Link is Down
[  743.165028] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[  743.435228] r8169 0002:01:00.0 eth1: Link is Down
[  746.078362] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
```

chhofi, 2025-02-07:
I do not know if that contributes to solving the issue. But the default configuration puts the LAN assignment to the WAN labeled port on the Device itself. Think should not matter just a confusing part when first starting. But have not reassigned it to test the initial install setup.

chhofi, 2025-02-07:
And here the complete log from the start of the device:
```

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.6.72 (runner@fv-az1118-846) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r0+28381-ad98c322cc) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue Jan 21 10:03:13 2025
[    0.000000] Machine model: FriendlyElec NanoPi R5C
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x00000001f0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] cma: Reserved 16 MiB at 0x00000000ef000000 on node -1
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 28 pages/cpu s77416 r8192 d29080 u114688
[    0.000000] pcpu-alloc: s77416 r8192 d29080 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xfe660000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1031688
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000e8ba7000-0x00000000ecba7000] (64MB)
[    0.000000] Memory: 4003756K/4192256K available (10560K kernel code, 3074K rwdata, 7528K rodata, 2944K init, 750K bss, 172116K reserved, 16384K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] GICv3: MBI range [296:319]
[    0.000000] GICv3: Using MBI frame 0x00000000fd410000
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS: No ITS available, not enabling LPIs
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001813] Console: colour dummy device 80x25
[    0.002322] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.003339] pid_max: default: 32768 minimum: 301
[    0.008696] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.009447] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.014994] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.016190] rcu: Hierarchical SRCU implementation.
[    0.016666] rcu: 	Max phase no-delay instances is 1000.
[... 897 lines cut ...]
[ 1255.000466] r8169 0002:01:00.0 eth1: Link is Down
[ 1255.045090] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[ 1255.315151] r8169 0002:01:00.0 eth1: Link is Down
[ 1255.485051] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[ 1255.765262] r8169 0002:01:00.0 eth1: Link is Down
[ 1258.523084] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[ 1273.882240] r8169 0002:01:00.0 eth1: Link is Down
[ 1273.935065] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[... cut ...]
[ 1296.037060] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
[ 1311.589570] r8169 0002:01:00.0 eth1: Link is Down
[ 1311.655027] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[ 1311.935223] r8169 0002:01:00.0 eth1: Link is Down
[ 1312.115021] Realtek Internal NBASE-T PHY r8169-2-100:00: attached PHY driver (mii_bus:phy_addr=r8169-2-100:00, irq=MAC)
[ 1312.395221] r8169 0002:01:00.0 eth1: Link is Down
[ 1315.028177] r8169 0002:01:00.0 eth1: Link is Up - 1Gbps/Full - flow control off
```

Ysurac, 2025-02-07:
By default, OMR put WAN as LAN and LANs as WAN. Some platform have multiples LANs ports or WAN port with a better connection. It's not the case on Nanopi R5C, to solve this can you give me the result of ```/etc/os-release``` and ```/etc/board.json``` via SSH ?

For the network issue, I don't know. How do you connect to it ? via a switch, directly,... ?

chhofi, 2025-02-07:
sure :)
```

root@OpenMPTCProuter:~# cat /etc/os-release
NAME="openmptcprouter"
VERSION="v0.62-snapshot-6.6"
ID="openmptcprouter"
ID_LIKE="lede openwrt"
PRETTY_NAME="openmptcprouter v0.62-snapshot-6.6"
VERSION_ID="v0.62-snapshot-6.6"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0+28381-ad98c322cc"
OPENWRT_BOARD="rockchip/armv8"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="busybox override"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="openmptcprouter v0.62-snapshot-6.6 r0+28381-ad98c322cc"
OPENWRT_BUILD_DATE="1737453793"
root@OpenMPTCProuter:~# cat /etc/board.json
{
        "model": {
                "id": "friendlyarm,nanopi-r5c",
                "name": "FriendlyElec NanoPi R5C"
        },
        "led": {
                "wan": {
                        "name": "WAN",
                        "sysfs": "green:wan",
                        "type": "netdev",
                        "device": "eth1",
                        "mode": "link tx rx"
                },
                "lan": {
                        "name": "LAN",
                        "sysfs": "green:lan",
                        "type": "netdev",
                        "device": "eth0",
                        "mode": "link tx rx"
                },
                "wlan": {
                        "name": "WLAN",
                        "sysfs": "green:wlan",
                        "type": "netdev",
                        "device": "phy0-ap0",
                        "mode": "link tx rx"
                }
        },
        "network": {
                "lan": {
                        "device": "eth0",
                        "protocol": "static",
                        "macaddr": "6e:b5:b8:48:f3:8a"
                },
                "wan": {
                        "device": "eth1",
                        "protocol": "dhcp",
                        "macaddr": "6e:b5:b8:48:f3:89"
                }
        },
        "wlan": {
                "phy0": {
                        "path": "3c0000000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0",
                        "info": {
                                "antenna_rx": 3,
                                "antenna_tx": 3,
                                "bands": {
                                        "2G": {
                                                "ht": true,
                                                "max_width": 40,
                                                "modes": [
                                                        "NOHT",
                                                        "HT20",
                                                        "HT40"
                                                ],
                                                "default_channel": 1
                                        },
                                        "5G": {
                                                "ht": true,
                                                "vht": true,
                                                "max_width": 80,
                                                "modes": [
                                                        "NOHT",
                                                        "HT20",
                                                        "VHT20",
                                                        "HT40",
                                                        "VHT40",
                                                        "VHT80"
                                                ],
                                                "default_channel": 36
                                        }
                                },
                                "radios": [
                                ]
                        }
                }
        }
}
root@OpenMPTCProuter:~# 
```

regarding the disconnect: I am directly connect to the nanopi via ethernet cable to my Macbook. No Switch or router in-between. Really strange behavior that I do not see on the FriendlyElec OpenWRT

chhofi, 2025-02-07:
[image]

When I look at the Devices list two ETH1 interfaces are listed. Maybe that causes something?

Ysurac, 2025-02-07:
There is something strange...
What is the result of ```ip a``` and ```uci show network``` ?

Ysurac, 2025-02-07:
There is a bridge and the interface declared 2 times in config, this can be an issue. I will solve that.

chhofi, 2025-02-07:
All right :) Thank you again!

chhofi, 2025-02-08:
[image]
I have tested the new version: openmptcprouter-v0.62-snapshot-6.6-r0+28431-92e020b50f-rockchip-armv8-friendlyarm_nanopi-r5c-ext4-sysupgrade

And now the Interfaces look right. Also no disconnect anymore.

 But now the ETH1 port does not get detected anymore. In order to make sure it is not a hardware issue I installed the original friendlywrt and the port shows up and is working.

```
root@OpenMPTCProuter:~# ip a
 ip a 
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 2a5b:8ff1:e5f4::
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
5: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr ee16:59b:6cb8::
8: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
9: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 2000
    link/ether 6e:b5:b8:48:f3:8a brd ff:ff:ff:ff:ff:ff permaddr 02:49:9d:ee:2b:ab
    inet 192.168.100.1/24 brd 192.168.100.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd57:f855:acfd::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::6cb5:b8ff:fe48:f38a/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever

root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.metric='5'
network.loopback.ip4table='5'
network.loopback.ip6table='65'
network.globals=globals
network.globals.ula_prefix='fd57:f855:acfd::/48'
network.globals.packet_steering='0'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='8'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
[... 133 lines cut ...]

[   11.612052] usbcore: registered new interface driver hso
[   11.616485] usbcore: registered new interface driver ipheth
[   11.621050] usbcore: registered new interface driver kaweth
[   11.716026] pegasus: Pegasus/Pegasus II USB Ethernet driver
[   11.716663] usbcore: registered new interface driver pegasus
[   11.719988] usbcore: registered new device driver r8152-cfgselector
[   11.720615] usbcore: registered new interface driver r8152
[   11.723392] r8169 0001:01:00.0: enabling device (0000 -> 0003)
[   11.821498] r8169 0001:01:00.0 eth0: RTL8125B, 02:49:9d:ee:2b:ab, XID 641, IRQ 78
[   11.822173] r8169 0001:01:00.0 eth0: jumbo features [frames: 9194 bytes, tx checksumming: ko]
[   11.825781] sis900.c: v1.08.10 Apr. 2 2006
[   11.828814] sky2: driver version 1.30
[   11.833734] Solos PCI Driver Version 1.04
[   11.836888] tulip: tulip: unknown CPU architecture, using default csr0
[   11.839544] usbcore: registered new interface driver usbhid
[   11.840048] usbhid: USB HID core driver
[   11.846266] usbcore: registered new interface driver usbserial_generic
[   11.846909] usbserial: USB Serial support registered for generic
[   11.851069] usbcore: registered new interface driver visor
[   11.851606] usbserial: USB Serial support registered for Handspring Visor / Palm OS
[   11.852306] usbserial: USB Serial support registered for Sony Clie 5.0
[   11.852910] usbserial: USB Serial support registered for Sony Clie 3.5

```

chhofi, 2025-02-09:
Hm... now I reinstalled the original OpenWRT and upgraded the firmware via the web interface to openmptcprouter. Now the eth1 gets detected again... guess I have made some mistake then installing the new firmware. no disconnects and eth1 working. so I would say the issue is solved. thanks! :)

closed 2025-05-15 by github-actions[bot]