Ysurac/openmptcprouter #3791: ZBT-Z8109AX Hardware adaption

fdyf520, 2025-02-09:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Equipment Name: ZBT-Z8109AX
This device is an upgraded version of the ZBT z8102ax NAND 128Mo, supporting dual 4G and dual 5G networks, while the rest remains the same.But their system is not universal.I can get the dts of the device and whether I can make an adaptation to the device

**Describe why you are asking for this support?**

It is already excellent to adapt this system to the hardware of ZBT, z8102ax, NAND 128Mo. Now there are new hardware, and I hope to get support from openmptcprouter. I can provide the hardware

[image]

[image]

Ysurac, 2025-02-09:
With the DTS I can at least check if it's the same hardware or not as z8102ax or need little adaptation.
But I think there is some difference, I can see there is 3 LANs ports instead of 4.

fdyf520, 2025-02-09:
/dts-v1/;
#include "mt7981.dtsi"
/ {
	model = "ZBT-Z8109AX";
	compatible = "zbtlink,z8109ax", "mediatek,mt7981-spim-snand-rfb";
	chosen {
		bootargs = "console=ttyS0,115200n1 loglevel=8  \
				earlycon=uart8250,mmio32,0x11002000";
	};

	memory {
		// fpga ddr2: 128MB*2
		reg = <0 0x40000000 0 0x10000000>;
	};

	gpio-keys {
		compatible = "gpio-keys";
		reset {
			label = "reset";
			linux,code = <KEY_RESTART>;
			gpios = <&pio 1 GPIO_ACTIVE_LOW>;
		};
	};

        gpio-leds {
                compatible = "gpio-leds";
                net {

                        label = "net";
                        gpios = <&pio 11 GPIO_ACTIVE_LOW>;
                };

                4g {

                        label = "4g";
                        gpios = <&pio 8 GPIO_ACTIVE_LOW>;
                };

                4g2 {

                        label = "4g2";
                        gpios = <&pio 13 GPIO_ACTIVE_LOW>;
                };

                4g3 {

                        label = "4g3";
                        gpios = <&pio 9 GPIO_ACTIVE_LOW>;
                };

                4g4 {

                        label = "4g4";
                        gpios = <&pio 10 GPIO_ACTIVE_LOW>;
                };
        };

        gpio-export {

                compatible = "gpio-export";
                #size-cells = <0>;

                wdg {
                        gpio-export,name = "wdg";
                        gpio-export,output = <1>;
                        gpios = <&pio 2 GPIO_ACTIVE_HIGH>;
                };

                4g {
                        gpio-export,name = "4g";
[... 327 lines cut ...]
			groups = "spi1_1";
		};
	};

	uart1_pins: uart1-pins-g1 {
                mux {
                        function = "uart";
                        groups = "uart1_1";
                };
        };

	uart2_pins: uart2-pins-g1 {
		mux {
                        function = "uart";
                        groups = "uart2_1";
                };
        };
};

&xhci {
	mediatek,u3p-dis-msk = <0x0>;
	phys = <&u2port0 PHY_TYPE_USB2>,
	       <&u3port0 PHY_TYPE_USB3>;
	status = "okay";
};

fdyf520, 2025-02-09:
You can compare the dts files, or you can precompile one and I can test it for you. I can send you my hardware when necessary

Ysurac, 2025-02-10:
What version do you have ? With a flash of 512,256,128 or 64MB ?

fdyf520, 2025-02-10:
128MB
 

小徐徐徐徐徐徐
***@***.***

fdyf520, 2025-02-10:
Can I get in touch with you through other means, such as wechat or qq or other instant messaging software? In this way, we can maintain quick communication and contact with you

小徐徐徐徐徐徐
***@***.***

Ysurac, 2025-02-10:
I'm on Telegram or IRC on libera.chat, both with nickname Ycarus.

fdyf520, 2025-02-10:
Ok, thank you. I'll register first and get in touch with you after registration

小徐徐徐徐徐徐
***@***.***

umekzg, 2025-03-10:
What speeds are you getting with that 1.3GHz dual code CPU?

Ysurac, 2025-03-10:
ZBT 810x can reach about 400Mbps

Tema8820, 2025-05-21:
I am planning to buy the ZBT Z8102AX model and flash it on Keenetic.
I accidentally saw this information.
Maybe I should consider this version ZBT-Z8109AX?
What is the difference?

virtualized-human, 2025-09-27:
is it only me or is the device not booting after flashing https://releases.openmptcprouter.com/v0.63-6.12/z8109ax_128m/targets/mediatek/filogic/openmptcprouter-v0.63-6.12-r0+30806-070d8eb4d5-mediatek-filogic-z8109ax-128m-squashfs-sysupgrade.bin ?

virtualized-human, 2025-09-27:
ah it is booted just only wan port was configured :)

kiloon, 2025-09-27:
> ah it is booted just only wan port was configured :)
I had same thing. Maybe it's good to add to the manual. Or add defconfig to the firmware so lan ports will be configured after firmware upgrade.