Ysurac/openmptcprouter #3798: Newly joined networks do not participate in aggregation

lhc1993, 2025-02-15:
## Expected Behavior

After adding a new device, OMR has enabled aggregation, but the newly added network does not pass traffic and is not added to the aggregation. What is the reason?

## Current Behavior

yes

## Possible Solution

Check the status of new devices and add them to the aggregation

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

logs：
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi1' has link connectivity loss
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): udhcpc: received SIGTERM
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): udhcpc: unicasting a release of 10.234.135.135 to 10.234.135.136
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): udhcpc: sending release
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): udhcpc: bind(UDP): Address not available
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): udhcpc: entering released state
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (31808): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "pcie_mhi1" } (Permission denied)
Feb 15 03:07:39 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi1' is now down
Feb 15 03:07:39 OpenMPTCProuter user.notice mwan3-hotplug[21878]: mwan3 hotplug on pcie_mhi1 not called because interface disabled
Feb 15 03:07:43 OpenMPTCProuter user.notice post-tracking-001-initialize: Routes not correctly set for pcie_mhi1 (pcie_mhi1 table 16) with IPs 10.234.135.135 
Feb 15 03:07:46 OpenMPTCProuter user.notice omr-tracker: Launching...
Feb 15 03:08:04 OpenMPTCProuter user.notice omr-tracker: Launched
Feb 15 03:08:04 OpenMPTCProuter user.notice post-tracking-002-error: pcie_mhi1 (pcie_mhi1) switched off because link down
Feb 15 03:08:06 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 212.27.48.10)
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: Network device 'pcie_mhi1' link is up
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi1' has link connectivity 
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi1' is setting up now
Feb 15 03:08:41 OpenMPTCProuter daemon.notice ttyd[13803]: [2025/02/15 11:08:41:4877] N: rops_handle_POLLIN_netlink: DELADDR
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (5827): udhcpc: started, v1.36.1
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (5827): udhcpc: broadcasting discover
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (5827): udhcpc: broadcasting select for 10.192.139.16, server 10.192.139.17
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: pcie_mhi1 (5827): udhcpc: lease of 10.192.139.16 obtained from 10.192.139.17, lease time 7200
Feb 15 03:08:41 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi1' is now up
Feb 15 03:08:42 OpenMPTCProuter user.notice mwan3-hotplug[6012]: mwan3 hotplug on pcie_mhi1 not called because interface disabled
Feb 15 03:08:42 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of pcie_mhi1 (pcie_mhi1)
Feb 15 03:08:43 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Feb 15 03:08:48 OpenMPTCProuter user.notice post-tracking-003-up: pcie_mhi1 (pcie_mhi1) switched up
Feb 15 03:08:50 OpenMPTCProuter user.notice post-tracking-003-up: Local IP changed for pcie_mhi1 (pcie_mhi1) force Multipath
Feb 15 03:09:00 OpenMPTCProuter user.notice omr-schedule-010-services: Set firewall on server vps

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

lhc1993, 2025-02-15:
For example, at the beginning, I checked that multiple network cables and some USB network cards were being aggregated. After a while, I unplugged several of them and only kept one. After about 10 minutes, I plugged in the others, and they would not automatically join the aggregated link. What is the reason?

lhc1993, 2025-02-15:
**Plug in other networks, stop the traffic, and then unplug the device that is running the traffic. The following log will appear**

Feb 15 04:49:13 OpenMPTCProuter daemon.notice netifd: Interface 'pcie_mhi0' is now down
Feb 15 04:49:13 OpenMPTCProuter user.notice mwan3-hotplug[32191]: mwan3 hotplug on pcie_mhi0 not called because interface disabled
Feb 15 04:49:18 OpenMPTCProuter user.notice post-tracking-001-initialize: Routes not correctly set for pcie_mhi0 (pcie_mhi0 table 15) with IPs 10.209.35.125 
Feb 15 04:49:20 OpenMPTCProuter user.notice omr-tracker: Launching...
Feb 15 04:49:38 OpenMPTCProuter user.notice omr-tracker: Launched
Feb 15 04:49:38 OpenMPTCProuter user.notice post-tracking-002-error: pcie_mhi0 (pcie_mhi0) switched off because link down
Feb 15 04:49:38 OpenMPTCProuter user.notice post-tracking-002-error: wwan0 (wwan0) switched off because link down
Feb 15 04:49:40 OpenMPTCProuter daemon.info omr-tracker-xray: xray is up (can contact via http 212.27.48.10)
**Feb 15 04:50:01 OpenMPTCProuter user.notice post-tracking-002-error: omrvpn down because gateway down, interface have no IPv6, interface have no IPv6 gateway
Feb 15 04:50:01 OpenMPTCProuter user.notice post-tracking-002-error: OpenVPN down, restart it
Feb 15 04:50:01 OpenMPTCProuter daemon.err openvpn(omr)[20860]: event_wait : Interrupted system call (fd=-1,code=4)
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[20860]: /usr/libexec/openvpn-hotplug route-pre-down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[20860]: net_addr_v4_del: 10.255.252.2 dev tun0
Feb 15 04:50:01 OpenMPTCProuter daemon.notice ttyd[544]: [2025/02/15 12:50:01:3137] N: rops_handle_POLLIN_netlink: DELADDR
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is down**
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now down
Feb 15 04:50:01 OpenMPTCProuter daemon.notice ttyd[544]: [2025/02/15 12:50:01:3179] N: rops_handle_POLLIN_netlink: DELADDR
Feb 15 04:50:01 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='tun0'
Feb 15 04:50:01 OpenMPTCProuter daemon.info ModemManager[12948]: hotplug: remove network interface tun0: event processed
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[20860]: /usr/libexec/openvpn-hotplug down omr tun0 1420 0 10.255.252.2 255.255.255.0 init
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[20860]: SIGTERM[hard,] received, process exiting
Feb 15 04:50:01 OpenMPTCProuter user.notice mwan3-hotplug[12940]: mwan3 hotplug on omrvpn not called because interface disabled
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: OpenVPN 2.6.10 aarch64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] [DCO]
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: library versions: OpenSSL 3.0.14 4 Jun 2024, LZO 2.10
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: DCO version: N/A
Feb 15 04:50:01 OpenMPTCProuter daemon.warn openvpn(omr)[13110]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Feb 15 04:50:01 OpenMPTCProuter daemon.warn openvpn(omr)[13110]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: TCP/UDP: Preserving recently used remote address: [AF_INET]123.57.0.66:65301
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: Attempting to establish TCP connection with [AF_INET]123.57.0.66:65301
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: TCP connection established with [AF_INET]123.57.0.66:65301
Feb 15 04:50:01 OpenMPTCProuter daemon.warn openvpn(omr)[13110]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: TCPv4_CLIENT link local: (not bound)
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: TCPv4_CLIENT link remote: [AF_INET]123.57.0.66:65301
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is disabled
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: [server] Peer Connection Initiated with [AF_INET]123.57.0.66:65301
Feb 15 04:50:01 OpenMPTCProuter daemon.warn openvpn(omr)[13110]: sitnl_route_save: rtnl: can't get ifname for index 0: No such device or address (errno=6)
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: TUN/TAP device tun0 opened
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: net_iface_mtu_set: mtu 1420 for tun0
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: net_iface_up: set tun0 up
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is enabled
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Network device 'tun0' link is up
Feb 15 04:50:01 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: net_addr_v4_add: 10.255.252.2/24 dev tun0
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' has link connectivity 
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is setting up now
Feb 15 04:50:01 OpenMPTCProuter daemon.notice netifd: Interface 'omrvpn' is now up
[... cut ...]
Feb 15 04:50:02 OpenMPTCProuter user.notice mwan3-hotplug[13550]: mwan3 hotplug on omrvpn not called because interface disabled
Feb 15 04:50:02 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Feb 15 04:50:03 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Feb 15 04:50:04 OpenMPTCProuter user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 192.168.100.1)
Feb 15 04:50:06 OpenMPTCProuter daemon.notice openvpn(omr)[13110]: Initialization Sequence Completed

lhc1993, 2025-02-15:
The phenomenon I see now is that the connection being aggregated does not support hot plugging well. For example, if you unplug and plug it in for a short time, it will recover, but if it takes a long time, it will not recover. Please help me, thank you.

Ysurac, 2025-02-15:
Please complete template (versions)

lhc1993, 2025-02-15:
[openmptcprouter v0.62-snapshot-v0.61rc1-9977ddac r0+27160-b72c4b5386](https://github.com/ysurac/openmptcprouter)

lhc1993, 2025-02-15:
OpenMPTCProuter version: [openmptcprouter v0.62-snapshot-v0.61rc1-9977ddac r0+27160-b72c4b5386](https://github.com/ysurac/openmptcprouter)
OpenMPTCProuter VPS version: 

[image]

OpenMPTCProuter VPS provider:
OpenMPTCProuter platform: Armv8
Country:

lhc1993, 2025-02-15:
If one of the networks in the aggregated connection is disconnected for a short time and then connected again, the network can be restored and re-aggregated. If it is disconnected for a long time, the network will not be able to be aggregated. If a new aggregated connection is created, there will be no problem. I don't know if there is any delay parameter setting that causes this problem. Please help me answer it. Thank you.

lhc1993, 2025-02-15:
@Ysurac please help me，thanks.

Ysurac, 2025-02-15:
Update VPS and router with latest snapshot: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
It's easier to check if issue is still here or not.
I would also need to know what is the proxy used (In System->OpenMPTCProuter->Wizard, "advanced settings" checkbox)

YuMinQ, 2025-04-08:
Hello, any updates on this? I'm encountering the same issue. I'm currently using v0.61 - does this version not support hot-plugging of network interfaces?

lhc1993, 2025-04-08:
It seems that none of them support it. As long as one interface is down, it will not join after it comes back online. I have tested it in v0.62rc1/v0.62/v0.63-snapshot

Ysurac, 2025-04-08:
It's supported and working.
But without answer to requests to find where can be the problem in your config there is no updates on your issue.

closed 2025-07-12 by github-actions[bot]