Ysurac/openmptcprouter #3799: I need a bonding system to make work videocalls and video streaming

lucecreativa, 2025-02-15:
Hi,
thanks in advance.
I need a bonding system for my job; i need a mobile solution for video calls and video conferences.
I own a Raspberry Pi 4 mod B.
Which is the right procedure to build a good bonding hardware with OpenMpTcp?
I bought a debian ver 12  in  VPS Server in AWS Lightsails.
Could you guide me to the right procedure  ?
Thank you very much to all in advance

Ysurac, 2025-02-16:
All is on the doc.
For the RPI: connect it to a switch, set all modems/box to fixed IP in different subnets, on the RPI http://192.168.100.1/ in System->OpenMPTCProuter, Wizard tab, set keys, modems IP as gateway and another IP in same subnet for each WAN.
For the VPS: AWS have a firewall that need to be configured with ports 65000-65535 TCP&UDP open, after script installation and reboot, the VPS is accessible via port 65222

lucecreativa, 2025-02-16:
> All is on the doc. For the RPI: connect it to a switch, set all modems/box to fixed IP in different subnets, on the RPI http://192.168.100.1/ in System->OpenMPTCProuter, Wizard tab, set keys, modems [...]

Thank you so much for your feedback but I have 2 issues:

- After script installation with" wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="6.6" sh" and after Debian Vps Reboot, on Aws Lightsail VPS, i can't login with SSH app on Mac Terminus.
Before script installation all was ok. 

- on Raspberry Pi 4 mod B, after a few hours of use, after having entered most of the configuration data, all the operations waiting to be saved cannot be saved.
Every time I click on the save button the page is refreshed and everything goes back to how it was before and always gives me the operations waiting to be saved.
The only solution is to install the flash on the raspberry again.
I've done it 4 times already.
I also cleared the cache of both chrome and safari but it still happens the same.

Thank you so much for your future support

Ysurac, 2025-02-16:
* Check via a console access that ```/etc/shorewall/params.net``` have correct internet interface
* Check that the SD card is not read only. You use ext4 or squashfs version ?

lucecreativa, 2025-02-16:
> * Check via a console access that `/etc/shorewall/params.net` have correct internet interface

PACKAGE:     openmptcprouter
 VERSION:     v0.61
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Tue Oct 22 10:01:29 UTC 2024
-----------------------------------------------------

I noticed that if I go to OpenMpTcp Router status page i read filesystem is read only.
At start everytime I edit basic parameters all it's ok, then this happens.
I used a lexar 64gb  microsd card

> * Check that the SD card is not read only. You use ext4 or squashfs version ?
ext4

Ysurac, 2025-02-16:
Change SD card and use squashfs. Check also power supply.

lucecreativa, 2025-02-16:
> Change SD card and use squashfs. Check also power supply.

this is second card I changed ad if I'm logged in router why I should have o check power supply ? Raspberry it's powered on.

Ysurac, 2025-02-16:
For the RPI, the power supply need to give enough amperage and to be stable, else there is many strange issues.
Your problem is hardware, in many case it's power supply. Also remove all USB connected device if any.

lucecreativa, 2025-02-16:
I made a mistake, I mooved LAN 1 in wlan firewall and now I can't log in. How can I login? I installed squashfs version and I can't enter to factory reset.
Thanks for your help

lucecreativa, 2025-02-16:
I tried to format and restore image but notthing happen

Ysurac, 2025-02-16:
If you write again image to SD card, then you do a full factory reset, this will work.

lucecreativa, 2025-02-16:
> If you write again image to SD card, then you do a full factory reset, this will work.

I writed 3 times the image but every time i try to ligin with 192.168.1.200 browser says that I can't open site.

lucecreativa, 2025-02-16:
I tried with custom flashing with raspberry Pi mager and wid wifi setting but nothing .... i can't login.

lucecreativa, 2025-02-16:
I tried with original Ip 192.168.100.1 and with with mine but I can't login. I have to buy another micro sd ?

Ysurac, 2025-02-17:
The default IP is 192.168.100.1

lucecreativa, 2025-02-17:
> The default IP is 192.168.100.1

I know but I tried with 192.168.100.1 and with 192.168.1.200 mine; nothing happens.
I tried to format micro sd and r flash 5 times. 
Every time I start raspberry I have the the problem.

Ysurac, 2025-02-17:
Don't need to format again if you can't ping it. Check your computer IP, must be in same subnet.
I'm not sure to understand why you try 192.168.1.200, after a flash it's 192.168.100.1 in all cases.

lucecreativa, 2025-02-17:
As you can see in attached images, my RPI 4 it's available on 192.168.1.200 but I can't login. 

[image]

[image]

Ysurac, 2025-02-17:
You can ping it ? You can SSH on it ?

lucecreativa, 2025-02-17:
> You can ping it ? You can SSH on it ?

I can't ping it and I can't SSH

[image]

[image]

lucecreativa, 2025-02-17:
But I see it in network scan 

[image]

lucecreativa, 2025-02-17:
Is there a way to delete squashfs filesystem ?

Ysurac, 2025-02-17:
Seems strange without any answer of the device. If you flash again the SD card you erase everything. It's why it's not possible on 192.168.1.200 IP.

lucecreativa, 2025-02-17:
I flashed SD card 10 times ...

lucecreativa, 2025-02-17:
If I try to login via ssh i get connection refuse message

lucecreativa, 2025-02-17:
> If you get "connection refused" something is answering, check if you don't have something on your network already using this IP. In any case, try to connect directly the RPI to your computer.

I connected RPI to my pc via lan directly; doesn't work.

lucecreativa, 2025-02-17:
[image]

As you can see RPI it's ready

lucecreativa, 2025-02-17:
As you can see rpi it's ready

[image]

Ysurac, 2025-02-17:
So it seems you have a network configuration issue. If you have also a keyboard connected, you can try to ping your PC.

lucecreativa, 2025-02-17:
If I try to ping RPI from my pc i get this:

Last login: Mon Feb 17 12:28:29 on ttys000
grafica999@Fotografia-Essenziale ~ % ping 192.168.1.200
PING 192.168.1.200 (192.168.1.200): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
ping: sendto: No route to host
Request timeout for icmp_seq 4
ping: sendto: Host is down
Request timeout for icmp_seq 5
ping: sendto: Host is down
Request timeout for icmp_seq 6
ping: sendto: Host is down
Request timeout for icmp_seq 7
ping: sendto: Host is down
Request timeout for icmp_seq 8
ping: sendto: Host is down
Request timeout for icmp_seq 9
ping: sendto: Host is down
Request timeout for icmp_seq 10
ping: sendto: Host is down

lucecreativa, 2025-02-17:
> So it seems you have a network configuration issue. If you have also a keyboard connected, you can try to ping your PC.

I linked Keyboard; can you please guide me to reset LAN permissione and move to LAN firewall so I can login ?

Ysurac, 2025-02-17:
By default all is open on LAN port. Try again ping from RPI, no idea what command it use here.

lucecreativa, 2025-02-17:
> By default all is open on LAN port. Try again ping from RPI, no idea what command it use here.

when yesterday I was inside RPI , for a my mistake, I moved LAN in the same firewal of wans.
is there a command to make a factory reset via console ?

lucecreativa, 2025-02-17:
which are default user and password to admin terminal ?

lucecreativa, 2025-02-17:
I changed destination zone for LAN yesterday

Ysurac, 2025-02-17:
You said that you flashed SD card since, so you erased the last config by flashing the SD card.

lucecreativa, 2025-02-17:
I downloaded squashfs images from openmptcprouter page another time but ... nothing ...

lucecreativa, 2025-02-17:
[image]

lucecreativa, 2025-02-17:
I just re flashed and Now it works ... I don't know why but works. 
Thank you so much for your support.

Now i'm trying to configure openmptcp

closed 2025-05-25 by github-actions[bot]