Ysurac/openmptcprouter #3800: Cannot Install VPS Script using latest snapshot

SpencerXZX, 2025-02-15:
Using:

` wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh`

I get the following:

`
wget -O - https://github.com/Ysurac/openmptcprouter-vps/blob/c942c303843e6c27f008d74e2d8138328a592e7b/debian-x86_64.sh | KERNEL="6.6" sh
--2025-02-15 23:23:29--  https://github.com/Ysurac/openmptcprouter-vps/blob/c942c303843e6c27f008d74e2d8138328a592e7b/debian-x86_64.sh
Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘STDOUT’

-                           [<=>                           ]       0  --.-KB/s               sh: 8: Syntax error: newline unexpected
-                           [ <=>                          ]   8.02K  --.-KB/s    in 0.03s

Cannot write to ‘-’ (Success).
root@omrvps:/home/dspencer# wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.6" sh
--2025-02-15 23:26:01--  https://www.openmptcprouter.com/server-test/debian-x86_64.sh
Resolving www.openmptcprouter.com (www.openmptcprouter.com)... 169.150.236.100, 2400:52e0:1a00::1207:2
Connecting to www.openmptcprouter.com (www.openmptcprouter.com)|169.150.236.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 105286 (103K) [application/octet-stream]
Saving to: ‘STDOUT’

-                         0%[                              ]       0  --.-KB/s               Check user...
Check Linux version...
Check architecture...
Check about broken packages...
Check if OpenMPTCProuter VPS is already installed...
Update mode
Update repo.openmptcprouter.com key
--2025-02-15 23:26:03--  https://repo.openmptcprouter.com/openmptcprouter.gpg.key
Resolving repo.openmptcprouter.com (repo.openmptcprouter.com)... 169.150.236.100, 2400:52e0:1a00::1206:2
Connecting to repo.openmptcprouter.com (repo.openmptcprouter.com)|169.150.236.100|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1241 (1.2K) [application/octet-stream]
Saving to: '/etc/apt/trusted.gpg.d/openmptcprouter.gpg'

/etc/apt/trusted.gpg.d/ 100%[=============================>]   1.21K  --.-KB/s    in 0s

2025-02-15 23:26:03 (920 MB/s) - '/etc/apt/trusted.gpg.d/openmptcprouter.gpg' saved [1241/1241]

Remove lock and update packages list...
Hit:1 http://us.archive.ubuntu.com/ubuntu jammy InRelease
Hit:2 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:3 https://repo.openmptcprouter.com next InRelease
Hit:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease
Hit:5 http://us.archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:6 http://us.archive.ubuntu.com/ubuntu jammy-backports InRelease
Hit:7 https://repo.openmptcprouter.com buster InRelease
Hit:8 http://archive.ubuntu.com/ubuntu bionic InRelease
Hit:9 http://old-releases.ubuntu.com/ubuntu impish InRelease
Reading package lists... Done
W: http://archive.ubuntu.com/ubuntu/dists/bionic-backports/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
W: http://archive.ubuntu.com/ubuntu/dists/bionic/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Install apt-transport-https, gnupg and openssh-server...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
gnupg is already the newest version (2.2.27-3ubuntu2.1).
openssh-server is already the newest version (1:8.9p1-3ubuntu0.10).
apt-transport-https is already the newest version (2.4.13).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Add OpenMPTCProuter repo...
Explanation: Prefer OpenMPTCProuter provided packages over the Debian native ones
Package: *
Pin: origin repo.openmptcprouter.com
[... 109 lines cut ...]

iperf-3.18.tar.gz.1     100%[=============================>] 671.91K  --.-KB/s    in 0.01s

2025-02-15 23:26:53 (69.0 MB/s) - 'iperf-3.18.tar.gz.1' saved [688037/688037]

--2025-02-15 23:26:53--  http://deb.debian.org/debian/pool/main/i/iperf3/iperf3_3.18-1.debian.tar.xz
Resolving deb.debian.org (deb.debian.org)... 151.101.2.132, 151.101.66.132, 151.101.130.132, ...
Connecting to deb.debian.org (deb.debian.org)|151.101.2.132|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13856 (14K) [application/x-xz]
Saving to: 'iperf3_3.18-1.debian.tar.xz'

iperf3_3.18-1.debian.ta 100%[=============================>]  13.53K  --.-KB/s    in 0.001s

2025-02-15 23:26:53 (21.2 MB/s) - 'iperf3_3.18-1.debian.tar.xz' saved [13856/13856]

Install iperf3 dependencies...
-                        93%[===========================>  ]  96.00K  4.43KB/s    in 22s

Cannot write to ‘-’ (Success).
`

Is this related to the recent iperf3 from source update? Is there a workaround?

Thanks

Ysurac, 2025-02-16:
No, here it wasn't able to download a file.
Check on VPS disk space with ```df --h``` if it seems ok, restart script with: ``` wget https://www.openmptcprouter.com/server-test/debian-x86_64.sh && KERNEL="6.6" sh debian-x86_64.sh```

darkman1983, 2025-02-18:
He's using ubuntu, thought only debian is supported?

Ysurac, 2025-02-18:
Only Debian is supported, Ubuntu is not tested (may work or not...).

SpencerXZX, 2025-02-18:
I tried Debian first and got the same error so I swapped to Ubuntu to see if it also has the issue. Disk is 11% full.

Ysurac, 2025-02-18:
And check /tmp free space, the script try to write here.

closed 2025-05-25 by github-actions[bot]