Ysurac/openmptcprouter #3801: VPN speed low , ok after reboot

Majaxx, 2025-02-16:
**Expected Behavior**
The flow rate drops on remote access (example of service: home assistant, emby, nextcloud)

**Current Behavior**
I have to restart the router to get a correct flow rate from the remote access
I use GloryTun TCP, I tried to use openvpn tcp but the ports remain blocked
I want to use another VPN but none of the others work

**Spécifications**
OpenMPTCProuter version: v0.61-6.6 r0
OpenMPTCProuter VPS version: 0.10.31 6.6.36
OpenMPTCProuter VPS provider: PulseHeberg
Country: FR
OpenMPTCProuter platform: Proxmox

Ysurac, 2025-02-16:
It's why Glorytun TCP is not the default anymore.
To debug with others VPN, check https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

darkman1983, 2025-02-16:
But there is no alternative, if you need like stable pings, all the other VPN's fail at that.

Ysurac, 2025-02-16:
I have quite stable ICMP ping with OpenVPN TCP.
Don't forget that ICMP ping doesn't mean connection latency as TCP is using a Proxy.

Ysurac, 2025-02-17:
OpenVPN is patched to support MPTCP. All available VPNs/Proxy support multipath.
UDP with Shadowsocks doesn't aggregate but with *Ray it can aggregate as UDP over TCP.

darkman1983, 2025-02-17:
> OpenVPN is patched to support MPTCP. All available VPNs/Proxy support multipath. UDP with Shadowsocks doesn't aggregate but with *Ray it can aggregate as UDP over TCP.

Do you mean "OpenVPN Bonding"? That one never worked for me. And the normal OpenVPN over TCP has bad latency.

Ysurac, 2025-02-17:
No, I mean "OpenVPN TCP", the default VPN on latest releases.

With latest snapshots, using OpenVPN TCP as VPN and on 2 FTTH + 1 4G:
```
root@OpenMPTCProuter:~# ping -c 4 9.9.9.9
PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=58 time=18.9 ms
64 bytes from 9.9.9.9: icmp_seq=2 ttl=58 time=18.3 ms
64 bytes from 9.9.9.9: icmp_seq=3 ttl=58 time=19.0 ms
64 bytes from 9.9.9.9: icmp_seq=4 ttl=58 time=22.7 ms

--- 9.9.9.9 ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3001ms
rtt min/avg/max/mdev = 18.253/19.703/22.674/1.738 ms
```

Not so bad. It's less good with 2 4G modems...

darkman1983, 2025-02-17:
Not all have and use FTTH; this example is probably more of an exception. I use cable and DSL, and my ping times are significantly worse with OpenVPN. FTTH is the best option, but you can't assume that applies to everyone.

> PING google.com (216.58.206.78) 56(84) bytes of data.
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=1 ttl=116 time=136 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=2 ttl=116 time=34.3 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=3 ttl=116 time=23.7 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=4 ttl=116 time=52.6 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=5 ttl=116 time=20.2 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=6 ttl=116 time=36.0 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=7 ttl=116 time=36.4 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=8 ttl=116 time=21.3 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=9 ttl=116 time=32.9 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=10 ttl=116 time=61.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=11 ttl=116 time=20.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=12 ttl=116 time=23.6 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=13 ttl=116 time=93.5 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=14 ttl=116 time=87.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=15 ttl=116 time=46.0 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=16 ttl=116 time=23.6 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=17 ttl=116 time=90.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=18 ttl=116 time=91.3 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=19 ttl=116 time=51.3 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=20 ttl=116 time=37.2 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=21 ttl=116 time=33.5 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=22 ttl=116 time=63.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=23 ttl=116 time=83.8 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=24 ttl=116 time=81.5 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=25 ttl=116 time=78.9 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=26 ttl=116 time=29.5 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=27 ttl=116 time=24.9 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=28 ttl=116 time=22.3 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=29 ttl=116 time=26.9 ms
64 bytes from tzfraa-aa-in-f14.1e100.net (216.58.206.78): icmp_seq=30 ttl=116 time=22.9 ms
--- google.com ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29045ms
rtt min/avg/max/mdev = 20.237/49.635/136.370/29.759 ms

Ysurac, 2025-02-17:
It was an example to say that OpenVPN doesn't always have a bad latency, you can't assume that applies to everyone, and has I said it's not as good with 2 4g modems...

Majaxx, 2025-02-17:
Hello 
when i lose 4G : 
 netstat -an | grep 65001
tcp        0      0 192.168.1.230:29824     VPS:65001   ESTABLISHED

after placing the order

root@OpenMPTCProuter:~# netstat -an | grep 65001
tcp        0      0 192.168.1.230:50528     IP_VPS:65001   TIME_WAIT
tcp        0      0 192.168.1.230:29824     IP_VPS:65001   TIME_WAIT
tcp        0      0 192.168.1.230:26872     IP_VPS:65001   ESTABLISHED
tcp        0      0 192.168.2.230:43827     IP_VPS:65001   TIME_WAIT
tcp        0      0 192.168.2.230:60993     IP_VPS:65001   ESTABLISHED

Majaxx, 2025-02-17:
Testing and fixing , add crontab
`
#!/bin/bash

# Adresse IP locale à vérifier
IP="192.168.2.230"
# Fichier de log
LOG_FILE="/var/log/glorytun_check.log"
# Date et heure
DATE=$(date '+%Y-%m-%d %H:%M:%S')

# Vérifier si une connexion ESTABLISHED existe pour l'IP
RESULT=$(netstat -an | grep "$IP" | grep ESTABLISHED)

if [ -n "$RESULT" ]; then
    echo "$DATE - La connexion pour $IP est ESTABLISHED." | tee -a "$LOG_FILE"
else
    echo "$DATE - Aucune connexion ESTABLISHED trouvée pour $IP. Redémarrage du service glorytun..." | tee -a "$LOG_FILE"
    /etc/init.d/glorytun restart
    echo "$DATE - Service glorytun redémarré." | tee -a "$LOG_FILE"
fi
`

darkman1983, 2025-02-19:
> It was an example to say that OpenVPN doesn't always have a bad latency, you can't assume that applies to everyone, and has I said it's not as good with 2 4g modems...

OpenVPN does not foward ports, i dunno whats the problem, but when i switch from Glorytun TCP to OpenVPN, ports no more forwarded. So it's unusable, i restarted VPS + Router several times, makes no difference.

Ysurac, 2025-02-19:
Not related to this issue, open a new one with results of https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug